WouterJD/FortiusANT #393: iFlow Power Reading vs Target Power

mikeherriman, 2022-08-24:
Hey guys,
It has been a long while since I’ve been on here…been creeping in the background once in a while, but I need to send huge cudos to the continued development of this project!!!

I last version I had running I believe was 5.1 on Pi 4.0 and little 7” touch screen display….worked like a charm using ANT+ to old Windows 10 PC running Zwift.

Then Zwift had their little forced upgrade and I had to get a new PC.  Still windows 10, but it has a compatible video adapter now.

I thought since I am getting back into the swing of things, I should upgrade PI to Bullseye OS and FortiusAnt to the latest version 6.5….well none of that worked and the PI and display are in a box.

back to Windows 10 laptop to run FortisAnt and ANT+ dongle to the Zwift PC.

I just tried to go for a ride, and the power readings were way too high.  Zwift and RGT used, and there was no change in resistance when incline changed.  The %incline appeared on the Fortius display and varied along the course, but the resistance did not change.  The target power varied with the %incline, but my level of resistance did not change.  The power reading also appeared to be maxed out around 800-900watts….my max ftp back in the day was only about 250.

varying the gear ratio with the + and - also did not change the resistance.

The older version of FortusAnt I was using, the only option I used was a -c12.

This new version with the json file, etc maybe I’m missing something?  Also, I think that the old version displayed the resistance value?  When I was riding before, the resistance display would go from low of 1900 up to about 4200-4300.

Long winded and lots of things covered here.

hope someone can point me in the right direction.

happy riding!
Mike H.

WouterJD, 2022-08-24:
Hi @mikeherriman it's a pitty that things do not work the way you would like them to.

For this issue, following the title, I will focus on the power reading.

FIRST: check **all** cabling and connections, including the cadence sensor first.

What I would do is use FortiusAnt in manual mode, so that you are not dependant of Zwift and/or ANT.
In this case you use the USB cable, the up/down on the headunit and the FortiusAnt display.
Check whether this all works.

If not, please attach a logfile and I will see whether I see something odd in there.

WouterJD, 2022-08-24:
Oh yeah. See [wiki](https://github.com/WouterJD/FortiusANT/wiki) I think that nothing changed on the power calculation.

mikeherriman, 2022-08-26:
Update:
I checked all cabling between the win 10 laptop and trainer head unit, along with 6-pin data cable between head unit and brake unit…no issues

I ran FortiusAnt in manual grade mode AND power mode, varied the grade and power with no change in overall resistance on the wheel.

My Zwift PC had an old 4.2 version of FortiusAnt configured to run the python program.  I moved the USB cable from Win10 laptop to my Win10 Zwift PC to test.  To my surprise, it fired up and worked right away.  I moved the ANT+ dongle from Win10 laptop to Zwift PC (now has 2).  I ran RGT and Zwift applications and both successfully sent grade % to FortiusAnt which in turn varied the resistance.  The power reading on FortiusAnt was also in the “normal” range and was received by RGT and Zwift with no issue.

The only thing that I can come up with is that I went the lazy route and only downloaded the Windows executable on the Win10 laptop and not the full Python code.

I am up and running on a single PC with the old code, but will try to find time to download and install the full python code on the laptop and test out.

WouterJD, 2022-08-26:
Please mention: python/exe; version; works/does not work.

You could simple move the **working .exe** from **ZwiftPC** to **otherPC** and check operation.

closed 2023-03-07 by WouterJD