WouterJD/FortiusANT #394: Regression Testing with bless and RPi

MeanHat, 2022-09-01:
Hi @WouterJD, as you know I have been testing the new release of bless with Raspberry Pi (RPi3 and RPi 0 2 W).
Here is a summary so far:
1. lib_detect_testenv error resolved with `python -m pip install --upgrade lib_detect_testenv`. Did you want to add this to FortiusANT dependencies in Requirements.txt?
2. File permissions for [https://github.com/WouterJD/FortiusANT/blob/master/raspberry/3_InstallBless_2022Q2_only.sh](url) need changing to allow anyone to execute
3. When running in -s simulation mode the power jumps from 100w to c440w once Zwift is connected. I have seen this behaviour before but do not know why it happens
4. I have not needed to run [https://github.com/WouterJD/FortiusANT/blob/master/raspberry/4_InstallWxPython.sh](url) in order to use -g GUI with either RPi
5. With RPi0 2 W I have to increase swapfile size from 100 to 1024 in order to clone FortiusANT
6. Successfully run RPi3 in sim mode with different combinations of command line parameters: -g -s -d -D and -O display (with ST7789 miniTFT.
7. With RPi0 2 W (rev 1.0) there is bless error and bluetooth is not enabled. Here is the logfile with more info:
[FortiusAnt.2022-09-01 18-18-50.log](https://github.com/WouterJD/FortiusANT/files/9472109/FortiusAnt.2022-09-01.18-18-50.log)

I'll test RPi3 with Fortius trainer next but pause testing with RPi0 2 W until you are happy to proceed

MeanHat, 2022-10-11:
Hi @WouterJD - I hope this was helpful. Would you like any assistance with further testing?

WouterJD, 2022-10-11:
It sure is helpful. I'm still running/riding outside and have to undust the Tacx for further testing.
Next objective is the https://devilstrail.nl/schiermonnikoog/ in a month and the focus wil move to indoor training again.
Sorry for being slow :-)

decodeais, 2022-12-14:
I wanted to test your new versions too. I thought it would be a good occasion to test your install scripts, but I had to do a lot of changes.  The greatest problem was the new wheell for wxPython. I put it to google drive
Google Drive and changed the installation Path.
The second problem for me was the broken RollResistance calibration (-c) option on all branches.
I tried to fix it,  but then I found out the confusion starts a little deeper.
I tested to change
 `self.CalibrateRR = float(self.args.CalibrateRR.replace(',', '.'))`
 to 
`self.CalibrateRR = float(self.args.CalibrateRR.replace(",", "."))`
but then I got a Typemismatch instead. 
I would like to  send you the branch with the Installation changes. 
Shall i make a pull request ?

GewoonGijs, 2022-12-19:
Hi Meanhat, did you manage to get the bless version running on Raspberry PI? I am running in a number of problems. See the issue I opened #408. Are any of these issues known to you?

Hope to hera from you, since I don't know how to roll back to an earlier version and my setup is broken now Aaarg.

MeanHat, 2022-12-20:
Hi @GewoonGijs - I am able to run FortiusANT on an RPi3 with bless using Zwift. I needed to make a few changes as noted in my 1 Sep 22 post (https://github.com/WouterJD/FortiusANT/issues/394#issue-1359200139). I also saw the new issue you raised: https://github.com/WouterJD/FortiusANT/issues/408#issue-1503704376.

What type of Pi are you running and can you provide any more information?

mtballday, 2022-12-28:
I'm also seeing the same two problems.  With Bless I get the clsBleServer._Server(); add_gatt() exception __init__() missing 1 required positional argument error, and with the other Bluetooth option I get the module 'logfile' has no attribute 'fLogfile' error.

Both on RPI3.

WouterJD, 2022-12-28:
See also #406 #408

GewoonGijs, 2022-12-28:
> Hi @GewoonGijs - I am able to run FortiusANT on an RPi3 with bless using Zwift. I needed to make a few changes as noted in my 1 Sep 22 post ([#394 (comment)](https://github.com/WouterJD/FortiusANT/issues/394#issue-1359200139)). I also saw the new issue you raised: [#408 (comment)](https://github.com/WouterJD/FortiusANT/issues/408#issue-1503704376).
> 
> What type of Pi are you running and can you provide any more information?

Strange, I see the same error that you had n the version 2 logfile.

WouterJD, 2022-12-28:
Updates in https://github.com/WouterJD/FortiusANT/tree/ant-queue

Johanboernl, 2023-03-10:
Solution for the Bless error

The error message indicates that the dbus_next module is missing. 
The bleBless module in your Python code requires this module to be installed in order to run properly. 
You can resolve this issue by installing the dbus_next module using pip, 
which is a package installer for Python.

You can try installing the module by running the following command in your terminal:

go to this folder

/home/pi/FortiusANT/pythoncode

us underneath code

`pip install dbus-next
`

Johanboernl, 2023-03-10:
@WouterJD @MeanHat 

hi my name is johan

and still having a issue if i want to remote control the aplication with vnc

layout of program is to big

[image]

MeanHat, 2023-03-13:
Hi Johan

Possible workarounds I have used are:

   1. Use of ST7789 240x240 miniTFT display mounted on the RPi which can
   display speed, cadence, power, gears etc. as well as providing a way of
   safely shutting down the RPi;
   2. Connecting the RPi to a CTP such as Zwift whichthen displays the data;
   3. Connecting the RPi using a HDMI cable to a suitable screen (in my
   case an old TV)

Hope this helps.

WouterJD, 2023-03-13:
> 3\. When running in -s simulation mode the power jumps from 100w to c440w once Zwift is connected. I have seen this behaviour before but do not know why it happens

Please raise a separate if the problem persists, so we can handle that separately.

WouterJD, 2023-03-13:
> Hi @WouterJD - I hope this was helpful. Would you like any assistance with further testing?

It was helpful for sure.
In some ways, packages that FortiusAnt depends on do not define their dependencies properly or something else is wrong.

I have installed two rpi's tonite and had different missing modules ????
Obvious solution is to install the missing packages manually or use "3_InstallMissingPackages.sh".

----
> 5\. increase swapfile size from 100 to 1024...

The git clone FortiusANT aborts with a kill 9, but when "--depth 1" is added, it also works.
I will document both options in the manual.

----
Script `5_GetFortiusAnt_Dependencies.sh` checks for "Rpi0" using `if [ uname -m == 'armv6l' ]`
but my rpi0 2W returns armv7l.
Major reason is to know on what platform wxPython is supported.

Any suggestions?

----

So far, with some minor glitches, FortiusAnt works under Raspberry Debian Bullseye.

WouterJD, 2024-02-17:
I suggest to close this issue, since we do not further work on it.
Regression testing is always good, but I have no time to fundamentally do work here.

MeanHat, 2024-02-17:
Hi WouterJD/FortiusANT

I'm stretched for time to do this so it seems like a sensible idea.

Best regards Keith

closed 2024-02-17 by WouterJD