WouterJD/FortiusANT #395: Tacx fortius ANT dongle not found

phpmdeveloper, 2022-10-02:
Goodmorning,

I have a tacx i-genius how can i my tacx  i-genius connected on Windows 10 Computer I have your first program ( fortiusANT ) install and unfortunately the program GIVE on program the follow suggest NOT ANT-DONGLE FOUND I have two dongles install  

- ANT USB -m 
- ANT USB Stick2

Can you my helping, thanks for this. i will the tacx i-genius powerback connected on Zwift

Kind regards,

Marco

BikeBeppe64, 2022-10-02:
Ciao Marco, I-Genius non è supportato.!

[Screenshot_2022-10-02-23-38-23-04_320a9a695de7cdce83ed5281148d6f19.jpg](https://user-images.githubusercontent.com/64853498/193477526-b5ef3e30-fc0c-4c4b-b74e-4e19a85868eb.jpg)

BikeBeppe64, 2022-10-02:
Ciao Marco, come inizio dai un'occhiata al manuale che ti dà un primo aiuto, poi se incontri problematiche, posta le questioni

phpmdeveloper, 2022-10-03:
Hé wouter,

Can you this supported maked for and how expansive is it than.

Thanks kind regard marco

WouterJD, 2022-10-03:
Buon giorno Marco e @BikeBeppe64 

Ci sará un motivo che Giuseppe sappia che parli Italiano meglio del Inglese; giusto?

Si tratta di due domande:
1. "NOT ANT-DONGLE FOUND"
2. Sopporto del i-Genius

1: FortiusAnt e scritto in Python usando librerie Python e non trova la chiavetta USB senza aver installato le driver essenziale. Leggi manuale pagina 23 " install USB driver". Dopo aver installate questi driver, FortiusAnt dovrebbe trovare una chiavetta ANT. Tu puoi saltare optione 1 perche il tuo Tacx non ha un cavo USB.

2: FortiusAnt standard cerca un collegamento USB ai Tracx tradizionali. Tacx Genius e Tacx Bushido communicano tramite ANT con il loro protocollo "segreto" e per quello FortiusAnt deve essere lanciato con un parametro particolare: -t Genius; leggi manual pagina 54.

Spero che riesci combinare!

phpmdeveloper, 2022-10-04:
hello,

i have a follow problem on rouvy

the problem have you in a picture or bijlage

thanks for this one,

kind regard marco

WouterJD, 2022-10-04:
Hi Marco

You probably responded on your smartphone, in that case no picture is stored on gitbug.
Please respond on github itself.

BikeBeppe64, 2022-10-04:
> hello,
> 
> i have a follow problem on rouvy
> the problem have you in a picture or bijlage
> thanks for this one,
> 
> kind regard marco
> 
> 

Hi Marco,
Are you Italian? if so you can talk to me in Italian, posso aiutarti io..
Giuseppe

phpmdeveloper, 2022-10-05:
Hallo dir

What is clv what are the position. Doe tacx

Thanks for this,

Kind regard,

Marco

WouterJD, 2022-10-05:
Hi @phpmdeveloper Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
You could also consider to visit [the sponsor page](https://github.com/sponsors/WouterJD)
- - -

Also be more specific in formulating your question.

phpmdeveloper, 2022-10-05:
Hallo dir

I have i-genius tacx, i have a member of usage for this. My software is
zwift i can not connected on zwift and not a search of my tacx wat are is
my failed

I have de fortiusant install and working als Genius this is a good purpose
only have now zwift on my fortiusant.

Thanks for the feedback,

Kind regard,

Marco jansen

phpmdeveloper, 2022-10-06:
> Hi Marco. What is your preferred language. We can help in English, Dutch,
> German, French and Italian.
> It appears English is not your favorite.

Goededag wouter,

Ik heb een klein vraag over hoe ik fortiusant kan koppelen aan zwift

Alvast bedankt,

Met vriendelijke groet,

Marco

WouterJD, 2022-10-06:
Goedemorgen Marco

Er zijn twee vragen:
- "ANT dongle niet gevonden"
- Ondersteuning van de i-Genius

1: FortiusAnt is geschreven in Python met behulp van Python-bibliotheken en kan de USB-stick niet vinden zonder de essentiële driver te installeren. Lees manpagina 23 "USB driver installeren". Na het installeren van deze stuurprogramma's zou FortiusAnt een ANT-sleutel moeten vinden. "Option 1" kun je overslaan omdat je Tacx geen USB-kabel heeft.

2: Standaard zoekt FortiusAnt naar een USB-verbinding met traditionele Tracx. Tacx Genius en Tacx Bush communiceren via ANT met hun "geheime" protocol en daarvoor moet FortiusAnt gestart worden met een bepaalde parameter: -t Genius; lees handleiding pagina 54.

Ik hoop dat je er uit komt!

phpmdeveloper, 2022-10-06:
Beste,

De tacx werkt correct op fortiusant en heb verbinding. Alleen als ik in
zwift zoek naar krijg ik geen tacx te zien wat moet ik doen om hem in
verbinding te zetten met zwift.

Thanks,

Mvg marco

WouterJD, 2022-10-06:
Één dongle is voor FortiusAnt, de andere voor Zwift.
Misschien een idee het manual te lezen en even kijken of je dat verder helpt, vraag iemand die Engels begrijpt misschien om even te helpen.
Je bent op de goede weg en verder advies kan ik geven als je wat meer vertelt / plaatjes meestuurt; dus niet meer reageren via mail maar op github waar je dit issue ook heb aangemaakt

closed 2022-11-17 by WouterJD