WouterJD/FortiusANT #396: Trouble installing FortiusAnt on RaspberryPi

rexkraemer, 2022-10-27:
Dear FortiusAnt-Community,

my name is Felix, greetings from Hamburg, Germany.

I've been reading here for almost one year while using my Tacx Bushido in regular mode and trying to smarten it up.
Iam stoked by this project and its spirit, you make a great job, thank you.

However, i fail to install FortiusAnt according to the UserManual. Maybe you can help me out.

My Setup:
Raspberry Pi 4 B 2GB RAM
FortiusAnt Version 6.5
RaspberryPi OS

Hardware assembly and OS installation worked well.

Following the instructions in the User manual, i get several Errors and Warnings in the logs:

**4_InstallWxPython.sh**

> Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
> WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<urllib3.connection.HTTPSConnection object at 0xb533dc70>: Failed to establish a new connection: [Errno -3] Temporärer Fehler bei der Namensauflösung')': /simple/gdown/
> WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<urllib3.connection.HTTPSConnection object at 0xb533d490>: Failed to establish a new connection: [Errno -3] Temporärer Fehler bei der Namensauflösung')': /simple/gdown/
> WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<urllib3.connection.HTTPSConnection object at 0xb533d8b0>: Failed to establish a new connection: [Errno -3] Temporärer Fehler bei der Namensauflösung')': /simple/gdown/
> Collecting gdown
>   Downloading https://www.piwheels.org/simple/gdown/gdown-4.5.3-py3-none-any.whl (14 kB)
> Collecting filelock
>   Downloading https://www.piwheels.org/simple/filelock/filelock-3.8.0-py3-none-any.whl (10 kB)
> Requirement already satisfied: six in /usr/lib/python3/dist-packages (from gdown) (1.16.0)
> Requirement already satisfied: beautifulsoup4 in /usr/lib/python3/dist-packages (from gdown) (4.9.3)
> Requirement already satisfied: requests[socks] in /usr/lib/python3/dist-packages (from gdown) (2.25.1)
> Collecting tqdm
>   Downloading https://www.piwheels.org/simple/tqdm/tqdm-4.64.1-py2.py3-none-any.whl (78 kB)
>      |████████████████████████████████| 78 kB 303 kB/s
> Requirement already satisfied: soupsieve>1.2 in /usr/lib/python3/dist-packages (from beautifulsoup4->gdown) (2.2.1)
> Collecting PySocks!=1.5.7,>=1.5.6
>   Downloading https://www.piwheels.org/simple/pysocks/PySocks-1.7.1-py3-none-any.whl (16 kB)
> Installing collected packages: PySocks, tqdm, filelock, gdown
> Successfully installed PySocks-1.7.1 filelock-3.8.0 gdown-4.5.3 tqdm-4.64.1
> /usr/local/lib/python3.9/dist-packages/gdown/cli.py:121: FutureWarning: Option `--id` was deprecated in version 4.3.1 and will be removed in 5.0. You don't need to pass it anymore to use a file ID.
>   warnings.warn(
> Downloading...
> From: https://drive.google.com/uc?id=1Uk1TSc6iLArx14QH8q85c9ytrw0IKLIn
> To: /home/tacx/Downloads/wxPython-4.1.1-cp37-cp37m-linux_armv7l.whl
> 100%|█████████████████████████████████████████| 120M/120M [18:16<00:00, 110kB/s]
> Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
> ERROR: wxPython-4.1.1-cp37-cp37m-linux_armv7l.whl is not a supported wheel on this platform.
>  Pre-built Python is installed, press Enter to continue:

**5_GetFortiusAnt_Depedencies.sh**

> Paketlisten werden gelesen… Fertig
> Abhängigkeitsbaum wird aufgebaut… Fertig
> Statusinformationen werden eingelesen… Fertig
> git ist schon die neueste Version (1:2.30.2-1).
> Das folgende Paket wurde automatisch installiert und wird nicht mehr benötigt:
>   libfuse2
> Verwenden Sie »sudo apt autoremove«, um es zu entfernen.
> 0 aktualisiert, 0 neu installiert, 0 zu entfernen und 0 nicht aktualisiert.
> FortiusANT already present
> Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
> Collecting bless
>   Using cached https://www.piwheels.org/simple/bless/bless-0.2.4-py3-none-any.whl (33 kB)
> Collecting lib_programname
>   Using cached https://www.piwheels.org/simple/lib-programname/lib_programname-2.0.5-py3-none-any.whl (8.8 kB)
> Requirement already satisfied: numpy!=1.19.4 in /usr/lib/python3/dist-packages (from -r /home/tacx/FortiusANT/pythoncode/requirements.txt (line 3)) (1.19.5)
> Collecting pyinstaller
>   Using cached pyinstaller-5.6.1.tar.gz (2.4 MB)
>   Installing build dependencies ... done
>   Getting requirements to build wheel ... done
[... 180 lines cut ...]
> found 5 high severity vulnerabilities
>   run `npm audit fix` to fix them, or `npm audit` for details
> 
> The npm install command has given several warnings which can be ignored.
>  FortiusAnt Bluetooth is installed, press Enter to continue:

After Reboot, the system tries to start FortiusAnt, but it throws an error:

**FortiusAnt.sh**

> Trashcan emptied - keep 1 month
> find: ‘/home/pi/.local/share/Trash/’: Datei oder Verzeichnis nicht gefunden
> find: ‘/home/pi/.local/share/Trash/’: Datei oder Verzeichnis nicht gefunden
> find: ‘/home/pi/.local/share/Trash/’: Datei oder Verzeichnis nicht gefunden
> Log-files from previous session are deleted - keep two days
> Traceback (most recent call last):
>   File "/home/tacx/FortiusANT/raspberry/../pythoncode/FortiusAnt.py", line 44, in <module>
>     import usb.core
> ModuleNotFoundError: No module named 'usb'
> FortiusAnt finished
>   Press Enter to continue:

So, what could be the problem?

Best, Felix

WouterJD, 2022-10-30:
Hi Felix,

Welcome to the FortiusANT community

**4_InstallWxPython.sh** is always a complex module to install, let's skip that one for now.

**5_GetFortiusAnt_Depedencies.sh** is more basic and straight forward.
The essential command is `pip3 install -r ~/FortiusANT/pythoncode/requirements.txt`

There are a lot of errors there, and my first idea is: **do you have enough space on your disk?**
Check that first.

Then try to install the individual modules (using pip3 install ...) and see what happens; the modules installed are listed in **requirements,.txt**:
```
bless
lib_programname
numpy != 1.19.4
pyinstaller
pyusb
requests
service_identity
wxPython
wheel
```

rexkraemer, 2022-10-31:
Hi Wouter,

i think so (i am a complete Raspberry / Linux novice):
 
> tacx@raspberrypi:~ $ df
> Dateisystem    1K-Blöcke Benutzt Verfügbar Verw% Eingehängt auf
> /dev/root       15021656 3820396  10533912   27% /
> devtmpfs          793780       0    793780    0% /dev
> tmpfs             958644       0    958644    0% /dev/shm
> tmpfs             383460    1416    382044    1% /run
> tmpfs               5120       4      5116    1% /run/lock
> /dev/mmcblk0p1    261108   50932    210176   20% /boot
> tmpfs             191728      20    191708    1% /run/user/1000
> tacx@raspberrypi:~ $ 

I installed the modules you listed. 

However, FortiusAnt.sh still gives me Errors:

> Trashcan emptied - keep 1 month
> find: ‘/home/pi/.local/share/Trash/’: Datei oder Verzeichnis nicht gefunden
> find: ‘/home/pi/.local/share/Trash/’: Datei oder Verzeichnis nicht gefunden
> find: ‘/home/pi/.local/share/Trash/’: Datei oder Verzeichnis nicht gefunden
> Log-files from previous session are deleted - keep two days
> Traceback (most recent call last):
>   File "/home/tacx/FortiusANT/raspberry/../pythoncode/FortiusAnt.py", line 46, in <module>
>     import antCTRL
>   File "/home/tacx/FortiusANT/pythoncode/antCTRL.py", line 8, in <module>
>     import antDongle         as ant
>   File "/home/tacx/FortiusANT/pythoncode/antDongle.py", line 120, in <module>
>     import FortiusAntCommand    as cmd
>   File "/home/tacx/FortiusANT/pythoncode/FortiusAntCommand.py", line 79, in <module>
>     import settings
>   File "/home/tacx/FortiusANT/pythoncode/settings.py", line 25, in <module>
>     import lib_programname
>   File "/home/tacx/.local/lib/python3.9/site-packages/lib_programname/__init__.py", line 3, in <module>
>     from lib_detect_testenv import *
> ModuleNotFoundError: No module named 'lib_detect_testenv'
> FortiusAnt finished
>   Press Enter to continue: 

Here are the logs of the module installing process:

> tacx@raspberrypi:~ $ pip3 install bless
> Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
> Collecting bless
>   Using cached https://www.piwheels.org/simple/bless/bless-0.2.4-py3-none-any.whl (33 kB)
> Collecting bleak
>   Downloading https://www.piwheels.org/simple/bleak/bleak-0.19.1-py3-none-any.whl (131 kB)
>      |████████████████████████████████| 131 kB 848 kB/s 
> Collecting dbus-fast<2.0.0,>=1.22.0
>   Downloading https://www.piwheels.org/simple/dbus-fast/dbus_fast-1.60.0-cp39-cp39-manylinux_2_31_armv7l.whl (2.0 MB)
>      |████████████████████████████████| 2.0 MB 1.9 MB/s 
> Collecting async-timeout<5,>=3.0.0
>   Downloading https://www.piwheels.org/simple/async-timeout/async_timeout-4.0.2-py3-none-any.whl (5.8 kB)
> Installing collected packages: async-timeout, dbus-fast, bleak, bless
> Successfully installed async-timeout-4.0.2 bleak-0.19.1 bless-0.2.4 dbus-fast-1.60.0

> tacx@raspberrypi:~ $ pip3 install lib_programname
> Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
> Collecting lib_programname
>   Using cached https://www.piwheels.org/simple/lib-programname/lib_programname-2.0.5-py3-none-any.whl (8.8 kB)
> Installing collected packages: lib-programname
>   WARNING: The script lib_programname is installed in '/home/tacx/.local/bin' which is not on PATH.
>   Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
> Successfully installed lib-programname-2.0.5

> tacx@raspberrypi:~ $ pip3 install numpy!=1.19.4
> Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
> Requirement already satisfied: numpy!=1.19.4 in /usr/lib/python3/dist-packages (1.19.5)
[... 52 lines cut ...]
>      command: /usr/bin/python3 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-qd6241yt/wxpython_d3fd792311b44a7094ece41d6528dd36/setup.py'"'"'; __file__='"'"'/tmp/pip-install-qd6241yt/wxpython_d3fd792311b44a7094ece41d6528dd36/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-pip-egg-info-10e4ojtw
>          cwd: /tmp/pip-install-qd6241yt/wxpython_d3fd792311b44a7094ece41d6528dd36/
>     Complete output (7 lines):
>     Traceback (most recent call last):
>       File "<string>", line 1, in <module>
>       File "/tmp/pip-install-qd6241yt/wxpython_d3fd792311b44a7094ece41d6528dd36/setup.py", line 27, in <module>
>         from buildtools.config import Config, msg, opj, runcmd, canGetSOName, getSOName
>       File "/tmp/pip-install-qd6241yt/wxpython_d3fd792311b44a7094ece41d6528dd36/buildtools/config.py", line 30, in <module>
>         from attrdict import AttrDict
>     ModuleNotFoundError: No module named 'attrdict'
>     ----------------------------------------
> WARNING: Discarding https://files.pythonhosted.org/packages/d9/33/b616c7ed4742be6e0d111ca375b41379607dc7cc7ac7ff6aead7a5a0bf53/wxPython-4.2.0.tar.gz#sha256=663cebc4509d7e5d113518865fe274f77f95434c5d57bc386ed58d65ceed86c7 (from https://pypi.org/simple/wxpython/). Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
>   Downloading https://www.piwheels.org/simple/wxpython/wxPython-4.1.1-cp39-cp39-linux_armv7l.whl (99.2 MB)
>      |████████████████████████████████| 99.2 MB 9.2 kB/s 
> Requirement already satisfied: numpy in /usr/lib/python3/dist-packages (from wxPython) (1.19.5)
[... cut ...]
>   WARNING: The scripts helpviewer, img2png, img2py, img2xpm, pycrust, pyshell, pyslices, pyslicesshell, pywxrc, wxdemo, wxdocs and wxget are installed in '/home/tacx/.local/bin' which is not on PATH.
>   Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
> Successfully installed wxPython-4.1.1

> tacx@raspberrypi:~ $ pip3 install wheel
> Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
> Requirement already satisfied: wheel in /usr/lib/python3/dist-packages (0.34.2)

WouterJD, 2022-11-01:
As you can see in the python's traceback:
- lib_programname is used
- and then the dependency lib_detect_testenv is missing

When looking at the installation, there are WARNINGS around lib_programname (re PATH).
I do not know why this is, in case your system is not a clean install it might be due to previous installations or due to installation of other packages.

What I would try is:
- pip3 uninstall lib_programname
- pip3 install lib_programname
and then try again.

It seems you are getting closer!

rexkraemer, 2022-11-02:
I formatted the SD-Card and installed the OS withe the official installer on my Windows 10 system.

I uninstalled / installed lib_programname without any problems:

> tacx@raspberrypi:~ $ pip3 uninstall lib_programname
> Found existing installation: lib-programname 2.0.5
> Uninstalling lib-programname-2.0.5:
>   Would remove:
>     /home/tacx/.local/bin/lib_programname
>     /home/tacx/.local/lib/python3.9/site-packages/lib_programname-2.0.5.dist-info/*
>     /home/tacx/.local/lib/python3.9/site-packages/lib_programname/*
> Proceed (y/n)? y
>   Successfully uninstalled lib-programname-2.0.5
> tacx@raspberrypi:~ $ pip3 install lib_programname
> Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
> Collecting lib_programname
>   Using cached https://www.piwheels.org/simple/lib-programname/lib_programname-2.0.5-py3-none-any.whl (8.8 kB)
> Installing collected packages: lib-programname
> Successfully installed lib-programname-2.0.5
> tacx@raspberrypi:~ $ 

Still get that error at FortiusAnt.sh start.

I manually installed lib-detect-testenev.

Now this error occurs:

> Traceback (most recent call last):
>   File "/home/tacx/FortiusANT/raspberry/../pythoncode/FortiusAnt.py", line 52, in <module>
>     import bleBless
>   File "/home/tacx/FortiusANT/pythoncode/bleBless.py", line 41, in <module>
>     from bless import (
>   File "/home/tacx/.local/lib/python3.9/site-packages/bless/__init__.py", line 24, in <module>
>     from bless.backends.bluezdbus.server import (  # noqa: F401
>   File "/home/tacx/.local/lib/python3.9/site-packages/bless/backends/bluezdbus/server.py", line 9, in <module>
>     from dbus_next.aio import MessageBus, ProxyObject  # type: ignore
> ModuleNotFoundError: No module named 'dbus_next'
> FortiusAnt finished

I manually installed dbus_next.

SUCCESS! (FortiusAnt startscreen is prompted)

Thank you for leading me there. Lets see, what happens next.

Have a good week, 

Best, Felix

WouterJD, 2022-11-02:
Hi Felix

Great you have been able to move on.
I have no explanation why the dependencies for the modules were not installed correctl;y; but is resolved.

Let me hear abvout your progress.
Wouter

closed 2022-11-17 by WouterJD