WouterJD/FortiusANT #397: UserGuide unreadable

MaxiPalle, 2022-10-31:
I'd like to use a Tacx Vortex T1961 with FortiusANT as described in #261.

Before starting with the implementation, I would like to know how to proceed ;-) but cannot read the corresponding UserGuide (Unable to render code block). Am I the onl one with this problem?

WouterJD, 2022-10-31:
Hi @MaxiPalle
Please share the unreadable page, I will check

MaxiPalle, 2022-11-01:
Hi Wouter,

it is https://github.com/WouterJD/FortiusANT/blob/master/doc/FortiusANTUserManual.pdf

Thank you!

WouterJD, 2022-11-01:
I understand that, but what page is unreadable?
Screenshot...

MaxiPalle, 2022-11-01:
This is, how it looks, if you click the file. I'd expect to open and show it.

[image]

But as downloading and opening the file is possible on my local machine, there is no real problem ;-)

WouterJD, 2022-11-01:
In my browser it's as below.
It seems you have a (PDF plugin in the) browser that gives the error, which is not a FortiusAnt issue but more a browser issue; especially because a downloaded PDF can be read on your system.
Success in further investigation!

[image]

closed 2022-11-10 by MaxiPalle