WouterJD/FortiusANT #398: Broadcast HRM from ANT through bluetooth

KovMat, 2022-11-04:
Hello @WouterJD !
I love your sw. Using it from Hungary with a TACX Flow, and a rpi 3. And connect with a tablet to it for zwifting.

I would have a question, I'd like to see the hrm value (from ant+) in zwift as well. How would it be possible to broadcast that value via Bluetooth?
I'm using bless, works perfectly!
~/FortiusANT/raspberry/FortiusAntMain.sh -a -bb -g -H 0
[image]

[image]

Thanks!
Máté

WouterJD, 2022-11-06:
You approach is correct; FortiusAnt has paired with your HRM and the HR is 67. So far so good.
The Heartrate is pairable in Zwift (it is orange).

FortiusAnt **SHOULD** transmit the HR through BLE, I would have to check the code to verify (of course you may do if you like). But have no code where I am now.

Please check and if not working, let me know and I will do a test. It might take some days though.

WouterJD, 2022-11-06:
Please refer to https://github.com/WouterJD/FortiusANT/issues/389#issuecomment-1280593341 to see how to debug BLE.

MeanHat, 2022-11-11:
Hi @KovMat - I had a similar setup but sold the ANT+ dongles and bought a BLE HRM.

KovMat, 2022-11-11:
> You approach is correct; FortiusAnt has paired with your HRM and the HR is 67. So far so good. The Heartrate is pairable in Zwift (it is orange).

Yes, but sadly it does not find any hr signal.
The same setup works through ANT+ without a problem.

> FortiusAnt **SHOULD** transmit the HR through BLE, I would have to check the code to verify (of course you may do if you like). But have no code where I am now.

Yes, I've also taken a short look to the python code, but could not find the issue there.

> 
> Please check and if not working, let me know and I will do a test. It might take some days though.

I did the test. Also downloaded the LightBlue app, it shows the following for the FortiusAntTrainer, seems to me, that it somehow does not have the hrm.
[image]

KovMat, 2022-11-11:
@MeanHat sorry to hear that, my plan B is the same. But I hope there is a solution to this problem as well. 
Otherwise I would need to use my computer, not the tablet, hence it does not have an ANT+ receiver.

WouterJD, 2022-11-15:
Hi @KovMat 

I think, FortiusAnt always publishes all data (Fuitness machine, Heartrate and steering).
So I do not know why your system does not provide the data.

To do investigation, you may start with
https://github.com/WouterJD/FortiusANT/blob/master/pythoncode/bleBless.py#L272

bleBless can be run on it's own (so you do not have all FortiusAnt overhead while testing).
Please start there.
If I find some time, I can assist lateron (quite busy nowadays)

It SHOULD work

closed 2023-03-07 by WouterJD