WouterJD/FortiusANT #399: Setup BLE on Windows 10

MaxiPalle, 2022-11-05:
Hi Wouter, hi community,
just started to use FortiusANT with my old i-Vortex.
Connecting the i-Vortex (T1961) with ANT is not a big deal, connection is established and data from i-Vortex can be seen in FortiusANT. 
But as I want to use BKool or Zwift, I would like to use an Asus USB-BT500 dongle to connect with my IPhone. And here the problem starts (at least for me). As I'm new with FortiusANT, I'm a little overchallenged with troubleshooting.

This is, how FortiuANT is starting up:

`15:33:18,185: FortiusANT started
15:33:18,186: -a
15:33:18,187: -b
15:33:18,187: -bb
15:33:18,187: -B False
15:33:18,187: -c False
15:33:18,188: -C 0/0
15:33:18,188: -d 64 (64, 0b1000000)
15:33:18,188: -D None
15:33:18,189: -g
15:33:18,189: -H None
15:33:18,190: -n
15:33:18,190: -p 0.2
15:33:18,191: -O False
15:33:18,191: -R defines Runoff: maxSpeed=40 dip=2 minSpeed=1 targetTime=7 power=100
15:33:18,192: -t Vortex
15:33:18,192: -T [34, 50] x [34, 30, 27, 25, 23, 21, 19, 17, 15, 13, 11] (start=50x19)
15:33:18,192: -x
15:33:18,193: ------------------
15:33:19,379: FortiusAnt GUI started in child-process
15:33:21,852: FortiusANT broadcasts data as an ANT+ Controlled Fitness Equipent device (FE-C), id=57591
15:33:22,069: FortiusANT broadcasts data as an ANT+ Heart Rate Monitor (HRM), id=57592
15:33:22,287: FortiusANT receives data from an ANT Tacx Vortex (VTX Controller), any device
15:33:22,619: FortiusANT receives data from an ANT Tacx Vortex Headunit (VHU Controller), any device
15:33:22,886: FortiusANT broadcasts data as an ANT+ Bicycle Power Sensor (PWR), id=57596
15:33:23,105: FortiusANT broadcasts data as an ANT+ Speed and Cadence Sensor (SCS), id=57595
15:33:23,375: FortiusANT receives commands from an ANT+ Generic Remote Control
15:33:24,677: ---------- ANT- and BLE-devices are activated ----------`

BLE CTP is blinking all the time in blue, but I don't understand, whether there is a connection between FortiusANT and my IPhone (which detects the Win 10 machine via Bluetooth with the host name and not as 'FortiusANT') or not.

This is, how the log looks like:

`2022-11-05 15:33:18,186: [FortiusAnt, INFO] FortiusANT started
2022-11-05 15:33:18,186: [FortiusAnt, INFO]     Restart=False debug=64
2022-11-05 15:33:18,186: [FortiusAnt, INFO] -a
2022-11-05 15:33:18,187: [FortiusAnt, INFO] -b
2022-11-05 15:33:18,187: [FortiusAnt, INFO] -bb
2022-11-05 15:33:18,187: [FortiusAnt, INFO] -B False
2022-11-05 15:33:18,187: [FortiusAnt, INFO] -c False
2022-11-05 15:33:18,188: [FortiusAnt, INFO] -C 0/0
2022-11-05 15:33:18,188: [FortiusAnt, INFO] -d 64 (64, 0b1000000)
2022-11-05 15:33:18,189: [FortiusAnt, INFO] -D None
2022-11-05 15:33:18,189: [FortiusAnt, INFO] -g
2022-11-05 15:33:18,190: [FortiusAnt, INFO] -H None
2022-11-05 15:33:18,190: [FortiusAnt, INFO] -n
2022-11-05 15:33:18,191: [FortiusAnt, INFO] -p 0.2
2022-11-05 15:33:18,191: [FortiusAnt, INFO] -O False
2022-11-05 15:33:18,191: [FortiusAnt, INFO] -R defines Runoff: maxSpeed=40 dip=2 minSpeed=1 targetTime=7 power=100
2022-11-05 15:33:18,192: [FortiusAnt, INFO] -t Vortex
2022-11-05 15:33:18,192: [FortiusAnt, INFO] -T [34, 50] x [34, 30, 27, 25, 23, 21, 19, 17, 15, 13, 11] (start=50x19)
2022-11-05 15:33:18,192: [FortiusAnt, INFO] -x
2022-11-05 15:33:18,193: [FortiusAnt, INFO] ------------------
2022-11-05 15:33:18,193: [FortiusAnt, INFO] Windows executable started
2022-11-05 15:33:18,193: [FortiusAnt, INFO] Version info for the components
2022-11-05 15:33:18,657: [FortiusAnt, INFO] Version=Fortius Antifier v6.4, on github=Fortius Antifier v6.5.
2022-11-05 15:33:18,663: [FortiusAnt, INFO] Fortius Antifier v6.4 [version on github=Fortius Antifier v6.5]
2022-11-05 15:33:18,663: [FortiusAnt, INFO]            FortiusAnt = 2022-03-08
2022-11-05 15:33:18,663: [FortiusAnt, INFO]               antCTRL = 2020-12-27
2022-11-05 15:33:18,663: [FortiusAnt, INFO]             antDongle = 2021-12-03
2022-11-05 15:33:18,664: [FortiusAnt, INFO]                 antFE = 2020-12-28
2022-11-05 15:33:18,664: [FortiusAnt, INFO]                antHRM = 2020-12-27
2022-11-05 15:33:18,664: [FortiusAnt, INFO]                antPWR = 2020-12-28
[... 77 lines cut ...]
 'Supported Power Range'}},
 '0000180d-0000-1000-8000-00805f9b34fb': 
{'00002a37-0000-1000-8000-00805f9b34fb': 
{'Properties': <GATTCharacteristicProperties.notify: 16>,
 'Permissions': <GATTAttributePermissions.writeable|readable: 3>,
 'Value': b'\x00{',
 'value': b'\x00{',
 'Description':
 'Heart Rate Measurement'}}}
)
2022-11-05 15:33:23,674: [bless.backends.winrt.server, DEBUG] Creating a new service with uuid: 00001826-0000-1000-8000-00805f9b34fb
2022-11-05 15:33:23,675: [bless.backends.winrt.server, DEBUG] Adding service to server with uuid 00001826-0000-1000-8000-00805f9b34fb
2022-11-05 15:33:23,709: [bless.backends.winrt.server, DEBUG] Creating a new service with uuid: 0000180d-0000-1000-8000-00805f9b34fb
2022-11-05 15:33:23,709: [bless.backends.winrt.server, DEBUG] Adding service to server with uuid 0000180d-0000-1000-8000-00805f9b34fb
2022-11-05 15:33:23,716: [FortiusAnt, INFO] clsBleServer._Server(): self.BlessServer.start()
2022-11-05 15:33:23,793: [FortiusAnt, INFO] clsBleServer._Server(): 1 second loop untill Close() called
2022-11-05 15:33:24,678: [FortiusAnt, INFO] ---------- ANT- and BLE-devices are activated ----------
2022-11-05 15:33:24,680: [FortiusAnt, INFO] clsFTMS_bless.SetAthleteData(0)
2022-11-05 15:33:24,682: [FortiusAnt, INFO] clsFTMS_bless.SetTrainerData(0.0, 0, 0)
2022-11-05 15:33:24,943: [FortiusAnt, INFO] clsFTMS_bless.SetAthleteData(0)
2022-11-05 15:33:24,945: [FortiusAnt, INFO] clsFTMS_bless.SetTrainerData(0.0, 0, 0)
2022-11-05 15:33:25,193: [FortiusAnt, INFO] clsFTMS_bless.SetAthleteData(0)
...`

Any idea, what I'm missing?

WouterJD, 2022-11-06:
Actually all you copied does not show anything strange; there seems no error message.

I understand from your asking that your iPhone does not see FortiusAnt.
You could check what BLE-devices are "in the air" with the LightBlue app; please read https://github.com/WouterJD/FortiusANT/issues/389#issuecomment-1280593341

MaxiPalle, 2022-11-10:
Hi Wouter,
it was a problem with the iOS app I used. Tacx cannot connect but Rouvy did.
So FortiusANT is working as expected. Currently very laggy but I will work on it,

closed 2022-11-10 by MaxiPalle