WouterJD/FortiusANT #401: Windows 7 PRO compatibility

GerritVriesema, 2022-11-06:
Installation Fortius ANT on work-laptop (WIN 10) easily done, however on old private laptop (WIN 7 PRO) no execution after installation: a black pop-up screen is shown, whereafter this "disappears" without showing the dashboard afterwards...?!
Download of executable was correct. 
Any idea what might cause this?

WouterJD, 2022-11-06:
Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
You could also consider to visit [the sponsor page](https://github.com/sponsors/WouterJD)
- - -

First step to find out is to start the windows executable from a .BAT file, which ends with pause so you can read the messages from the console. OR start from a command-liine, so you see what is displayed.

Although windows7 is some time ago, I rememer it did work.

GerritVriesema, 2022-11-08:
Dear Wouter,

Thanks for your reply. I will try to find out hat happens exactly when the
exe-file opens!

Best,

Gerrit

closed 2022-11-17 by WouterJD