WouterJD/FortiusANT #402: Issue with installing FortiusAnt on Raspi 4

runman19, 2022-11-10:
Hi, 

I have a similar but also slightly different problem as #396 with installing FortiusAnt on Raspi4:

It also started with **4_InstallWxPython.sh**
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: gdown in /usr/local/lib/python3.9/dist-packages (4.5.3)
Requirement already satisfied: beautifulsoup4 in /usr/lib/python3/dist-packages (from gdown) (4.9.3)
Requirement already satisfied: six in /usr/lib/python3/dist-packages (from gdown) (1.16.0)
Requirement already satisfied: requests[socks] in /usr/lib/python3/dist-packages (from gdown) (2.25.1)
Requirement already satisfied: filelock in /usr/local/lib/python3.9/dist-packages (from gdown) (3.8.0)
Requirement already satisfied: tqdm in /usr/local/lib/python3.9/dist-packages (from gdown) (4.64.1)
Requirement already satisfied: soupsieve>1.2 in /usr/lib/python3/dist-packages (from beautifulsoup4->gdown) (2.2.1)
Requirement already satisfied: PySocks!=1.5.7,>=1.5.6 in /usr/local/lib/python3.9/dist-packages (from requests[socks]->gdown) (1.7.1)
/usr/local/lib/python3.9/dist-packages/gdown/cli.py:121: FutureWarning: Option `--id` was deprecated in version 4.3.1 and will be removed in 5.0. You don't need to pass it anymore to use a file ID.
  warnings.warn(
Downloading...
From: https://drive.google.com/uc?id=1Uk1TSc6iLArx14QH8q85c9ytrw0IKLIn
To: /home/pi/Downloads/wxPython-4.1.1-cp37-cp37m-linux_armv7l.whl
100%|████████████████████████████████████████| 120M/120M [01:36<00:00, 1.25MB/s]
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
ERROR: wxPython-4.1.1-cp37-cp37m-linux_armv7l.whl is not a supported wheel on this platform.
 Pre-built Python is installed, press Enter to continue: 

**5_GetFortiusAnt_Dependencies.sh** was running fine after I was using your workaround with the manual installs as in #396 

**6_SetupNodeJs_FortiusAnt.sh** was ending in the following error messages:
> usb@1.9.2 install /home/pi/FortiusANT/node/node_modules/usb
> node-gyp-build

gyp ERR! configure error 
gyp ERR! stack Error: Command failed: /usr/bin/python -c import sys; print "%s.%s.%s" % sys.version_info[:3];
gyp ERR! stack   File "<string>", line 1
gyp ERR! stack     import sys; print "%s.%s.%s" % sys.version_info[:3];
gyp ERR! stack                       ^
gyp ERR! stack SyntaxError: invalid syntax
gyp ERR! stack 
gyp ERR! stack     at ChildProcess.exithandler (child_process.js:308:12)
gyp ERR! stack     at ChildProcess.emit (events.js:315:20)
gyp ERR! stack     at maybeClose (internal/child_process.js:1048:16)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)
gyp ERR! System Linux 5.15.76-v7l+
gyp ERR! command "/usr/local/bin/node" "/home/pi/FortiusANT/node/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/pi/FortiusANT/node/node_modules/usb
gyp ERR! node -v v14.15.3
gyp ERR! node-gyp -v v3.8.0
gyp ERR! not ok 
npm WARN fortiusant@0.0.1 No repository field.
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: xpc-connect@2.0.3 (node_modules/xpc-connect):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for xpc-connect@2.0.3: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: usb@1.9.2 (node_modules/usb):
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: usb@1.9.2 install: `node-gyp-build`
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: Exit status 1

audited 207 packages in 5.702s

12 packages are looking for funding
  run `npm fund` for details

found 5 high severity vulnerabilities
  run `npm audit fix` to fix them, or `npm audit` for details

The npm install command has given several warnings which can be ignored.
 FortiusAnt Bluetooth is installed, press Enter to continue: 

With the following errors in **npm audit fix**
npm ERR! code EAUDITNOPJSON
npm ERR! audit No package.json found: Cannot audit a project without a package.json

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2022-11-10T21_04_22_760Z-debug.log

**FortiusAnt.sh** now shows the following errors:
Trashcan emptied - keep 1 month
find: ‘/home/pi/.local/share/Trash/’: No such file or directory
find: ‘/home/pi/.local/share/Trash/’: No such file or directory
find: ‘/home/pi/.local/share/Trash/’: No such file or directory
Log-files from previous session are deleted - keep two days

Traceback (most recent call last):
  File "/home/pi/FortiusANT/raspberry/../pythoncode/FortiusAnt.py", line 723, in <module>
    mainProgram()
  File "/home/pi/FortiusANT/raspberry/../pythoncode/FortiusAnt.py", line 698, in mainProgram
    parent.ListenToChild()
  File "/home/pi/FortiusANT/raspberry/../pythoncode/FortiusAnt.py", line 507, in ListenToChild
    rtn = LocateHW(self)
  File "/home/pi/FortiusANT/raspberry/../pythoncode/FortiusAnt.py", line 137, in LocateHW
    rtn = FortiusAntBody.LocateHW(self)
  File "/home/pi/FortiusANT/pythoncode/FortiusAntBody.py", line 427, in LocateHW
    TacxTrainer = usbTrainer.clsTacxTrainer.GetTrainer(clv, AntDongle)
  File "/home/pi/FortiusANT/pythoncode/usbTrainer.py", line 464, in GetTrainer
    dev.set_configuration()
  File "/home/pi/.local/lib/python3.9/site-packages/usb/core.py", line 915, in set_configuration
    self._ctx.managed_set_configuration(self, configuration)
  File "/home/pi/.local/lib/python3.9/site-packages/usb/core.py", line 113, in wrapper
    return f(self, *args, **kwargs)
  File "/home/pi/.local/lib/python3.9/site-packages/usb/core.py", line 158, in managed_set_configuration
    self.managed_open()
  File "/home/pi/.local/lib/python3.9/site-packages/usb/core.py", line 113, in wrapper
    return f(self, *args, **kwargs)
  File "/home/pi/.local/lib/python3.9/site-packages/usb/core.py", line 131, in managed_open
    self.handle = self.backend.open_device(self.dev)
  File "/home/pi/.local/lib/python3.9/site-packages/usb/backend/libusb1.py", line 804, in open_device
    return _DeviceHandle(dev)
  File "/home/pi/.local/lib/python3.9/site-packages/usb/backend/libusb1.py", line 652, in __init__
    _check(_lib.libusb_open(self.devid, byref(self.handle)))
  File "/home/pi/.local/lib/python3.9/site-packages/usb/backend/libusb1.py", line 604, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 13] Access denied (insufficient permissions)

runman19, 2022-11-13:
After formatting and following the installation procedure, I have switched to the manual installation procedure of the wxpython packages of #396 , the following error message came up: 

pip3 install wx_python
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
ERROR: Could not find a version that satisfies the requirement wx_python
ERROR: No matching distribution found for wx_python

This seems to be the core problem. Can anybody help?

runman19, 2022-11-13:
For **6_SetupNodeJs_FortiusAnt** I am getting the following issues:

npm WARN deprecated request@2.88.2: request has been deprecated, see https://github.com/request/request/issues/3142
npm WARN deprecated tar@2.2.2: This version of tar is no longer supported, and will not receive security updates. Please upgrade asap.
npm WARN deprecated har-validator@5.1.5: this library is no longer supported
npm WARN deprecated uuid@3.4.0: Please upgrade  to version 7 or higher.  Older versions may use Math.random() in certain circumstances, which is known to be problematic.  See https://v8.dev/blog/math-random for details.

> usb@1.9.2 install /home/pi/FortiusANT/node/node_modules/usb
> node-gyp-build

gyp ERR! configure error 
gyp ERR! stack Error: Command failed: /usr/bin/python -c import sys; print "%s.%s.%s" % sys.version_info[:3];
gyp ERR! stack   File "<string>", line 1
gyp ERR! stack     import sys; print "%s.%s.%s" % sys.version_info[:3];
gyp ERR! stack                       ^
gyp ERR! stack SyntaxError: invalid syntax
gyp ERR! stack 
gyp ERR! stack     at ChildProcess.exithandler (child_process.js:308:12)
gyp ERR! stack     at ChildProcess.emit (events.js:315:20)
gyp ERR! stack     at maybeClose (internal/child_process.js:1048:16)
gyp ERR! stack     at Socket.<anonymous> (internal/child_process.js:439:11)
gyp ERR! stack     at Socket.emit (events.js:315:20)
gyp ERR! stack     at Pipe.<anonymous> (net.js:673:12)
gyp ERR! System Linux 5.15.61-v7l+
gyp ERR! command "/usr/local/bin/node" "/home/pi/FortiusANT/node/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/pi/FortiusANT/node/node_modules/usb
gyp ERR! node -v v14.15.3
gyp ERR! node-gyp -v v3.8.0
gyp ERR! not ok 

> @abandonware/bluetooth-hci-socket@0.5.3-8 install /home/pi/FortiusANT/node/node_modules/@abandonware/bluetooth-hci-socket
> node-pre-gyp install --fallback-to-build

node-pre-gyp ERR! install response status 404 Not Found on https://github.com/abandonware/node-bluetooth-hci-socket/releases/download/0.5.3-8/bluetooth_hci_socket-0.5.3-8-node-v83-linux-arm.tar.gz 
node-pre-gyp WARN Pre-built binaries not installable for @abandonware/bluetooth-hci-socket@0.5.3-8 and node@14.15.3 (node-v83 ABI, glibc) (falling back to source compile with node-gyp) 
node-pre-gyp WARN Hit error response status 404 Not Found on https://github.com/abandonware/node-bluetooth-hci-socket/releases/download/0.5.3-8/bluetooth_hci_socket-0.5.3-8-node-v83-linux-arm.tar.gz 
make: Entering directory '/home/pi/FortiusANT/node/node_modules/@abandonware/bluetooth-hci-socket/build'
  CXX(target) Release/obj.target/bluetooth_hci_socket/src/BluetoothHciSocket.o
In file included from /home/pi/.cache/node-gyp/14.15.3/include/node/node_buffer.h:25,
                 from ../src/BluetoothHciSocket.cpp:7:
/home/pi/.cache/node-gyp/14.15.3/include/node/node.h:758:7: warning: cast between incompatible function types from ‘void (*)(Nan::ADDON_REGISTER_FUNCTION_ARGS_TYPE)’ {aka ‘void (*)(v8::Local<v8::Object>)’} to ‘node::addon_register_func’ {aka ‘void (*)(v8::Local<v8::Object>, v8::Local<v8::Value>, void*)’} [-Wcast-function-type]
  758 |       (node::addon_register_func) (regfunc),                          \
      |       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/pi/.cache/node-gyp/14.15.3/include/node/node.h:792:3: note: in expansion of macro ‘NODE_MODULE_X’
  792 |   NODE_MODULE_X(modname, regfunc, NULL, 0)  // NOLINT (readability/null_usage)
      |   ^~~~~~~~~~~~~
../src/BluetoothHciSocket.cpp:635:1: note: in expansion of macro ‘NODE_MODULE’
  635 | NODE_MODULE(binding, BluetoothHciSocket::Init);
      | ^~~~~~~~~~~
../src/BluetoothHciSocket.cpp: In member function ‘int BluetoothHciSocket::kernelDisconnectWorkArounds(int, char*)’:
../src/BluetoothHciSocket.cpp:401:1: warning: control reaches end of non-void function [-Wreturn-type]
  401 | }
      | ^
  SOLINK_MODULE(target) Release/obj.target/bluetooth_hci_socket.node
  COPY Release/bluetooth_hci_socket.node
  COPY /home/pi/FortiusANT/node/node_modules/@abandonware/bluetooth-hci-socket/lib/binding/bluetooth_hci_socket.node
  TOUCH Release/obj.target/action_after_build.stamp
make: Leaving directory '/home/pi/FortiusANT/node/node_modules/@abandonware/bluetooth-hci-socket/build'

> @abandonware/bleno@0.5.1-4 install /home/pi/FortiusANT/node/node_modules/@abandonware/bleno
> node-gyp rebuild

make: Entering directory '/home/pi/FortiusANT/node/node_modules/@abandonware/bleno/build'
  SOLINK_MODULE(target) Release/obj.target/bleno.node
  COPY Release/bleno.node
make: Leaving directory '/home/pi/FortiusANT/node/node_modules/@abandonware/bleno/build'

> @abandonware/bleno@0.5.1-4 install /home/pi/FortiusANT/node/node_modules/bleno
> node-gyp rebuild

make: Entering directory '/home/pi/FortiusANT/node/node_modules/bleno/build'
  SOLINK_MODULE(target) Release/obj.target/bleno.node
  COPY Release/bleno.node
make: Leaving directory '/home/pi/FortiusANT/node/node_modules/bleno/build'
npm notice created a lockfile as package-lock.json. You should commit this file.
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: xpc-connect@^2.0.0 (node_modules/@abandonware/bleno/node_modules/xpc-connect):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for xpc-connect@2.0.3: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm"})
npm WARN fortiusant@0.0.1 No repository field.
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: usb@1.9.2 (node_modules/usb):
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: usb@1.9.2 install: `node-gyp-build`
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: Exit status 1

added 204 packages from 195 contributors and audited 207 packages in 837.597s

12 packages are looking for funding
  run `npm fund` for details

found 5 high severity vulnerabilities
  run `npm audit fix` to fix them, or `npm audit` for details

The npm install command has given several warnings which can be ignored.
 FortiusAnt Bluetooth is installed, press Enter to continue:

runman19, 2022-11-14:
**Not resolved but running**

Just for everybodies information: Despite all these errors, (nearly) everything works fine. Nearly, because I am getting just 1 error displayed in the FortiusAnt-console: 
15:59:53,903: Tacx head unit returns insufficient data, len=24
15:59:53,904: To resolve, try to run without Pedal Stroke Analysis.

without any negative effect on the overall functionality.

WouterJD, 2022-11-15:
> Tacx head unit returns insufficient data, len=24

The USB interface does not have a solid protocol with error recovery; just a straight serial interface.
Sometimes data is lost and then you get the message. If occurring only once you may ignore.
As you can see in the code, the message is given after some retries.
If it keeps occurring, switch off pda.

Practice shows the message pops up at the start (warming up of USB ????? :-) )
It may also disappear after cleaning all contacts and connections.

WouterJD, 2022-11-15:
I'm not so happy with nodejs myself because it is a big overhead for the purpose.
It was the only way to get frompython to BLE at the time of writing.

That's why -bb bless has been developed; requiring a BLE5 dongle on windows however.

If your system is working now, I suggest to close the issue without further investigation.

runman19, 2022-11-15:
If it is not a further issue that WXPython does not seem to be properly installed, we can close this issue. I am still waiting for my BLE adapter for Windows but I assume that everything will be working fine.

Therefore we (I/you?) can close it then.

runman19, 2022-11-15:
Sorry, but I have to reopen the issue again:
I am getting an Bluetooth error as well for bless as for node.js

- bless: 
18:44:39,171: clsBleServer._Server(); add_gatt() exception __init__() missing 1 required positional argument: 'max_write_without_response_size'
18:44:39,172: , Bluetooth interface n/a; Check exception.
18:44:40,141: * * Bluetooth interface disabled * * 

- node.js
S**top standard Bluetooth Service
Enable Bluetooth for FortiusAnt**
---------------------------------------------------------------
FortiusANT is open source and can be used freely.

Just for the fun of knowing where you all are training,
put yourself on the FortiusANT map by making yourself known
by leaving a message with name/location/trainer on
https://github.com/WouterJD/FortiusANT/issues/14

or visit the sponsoring page https://github.com/sponsors/WouterJD
---------------------------------------------------------------
19:07:06,686: Connected to Tacx Trainer T1932
19:07:06,991: Motor Brake Unit Firmware=0x1004 Serial= 9702 year=2008 type=T1941 Version2=514 MotorBrake=True
19:07:07,091: FortiusAnt applies the MotorBrake power curve
19:07:08,453: FortiusANT broadcasts data as an ANT+ Controlled Fitness Equipent device (FE-C), id=57591
19:07:08,602: FortiusANT receives data from an ANT+ Heart Rate Monitor (HRM display), any device
19:07:08,777: FortiusANT broadcasts data as an ANT+ Bicycle Power Sensor (PWR), id=57596
19:07:08,944: FortiusANT broadcasts data as an ANT+ Speed and Cadence Sensor (SCS), id=57595
19:07:09,109: FortiusANT receives commands from an ANT+ Generic Remote Control
**19:07:09,300: * * Bluetooth interface disabled * ***

runman19, 2022-11-15:
This here is the full log -file for bless

2022-11-15 19:24:50,229: [FortiusAnt, INFO] Python version started
2022-11-15 19:24:50,229: [FortiusAnt, INFO] Version info for the components
2022-11-15 19:24:50,330: [FortiusAnt, INFO] Version=Fortius Antifier v6.5, on github=Fortius Antifier v6.5.
2022-11-15 19:24:50,331: [FortiusAnt, INFO] Fortius Antifier v6.5
2022-11-15 19:24:50,331: [FortiusAnt, INFO]            FortiusAnt = 2022-03-08
2022-11-15 19:24:50,332: [FortiusAnt, INFO]               antCTRL = 2020-12-27
2022-11-15 19:24:50,332: [FortiusAnt, INFO]             antDongle = 2021-12-03
2022-11-15 19:24:50,332: [FortiusAnt, INFO]                 antFE = 2020-12-28
2022-11-15 19:24:50,332: [FortiusAnt, INFO]                antHRM = 2020-12-27
2022-11-15 19:24:50,332: [FortiusAnt, INFO]                antPWR = 2020-12-28
2022-11-15 19:24:50,332: [FortiusAnt, INFO]                antSCS = 2020-12-27
2022-11-15 19:24:50,332: [FortiusAnt, INFO]              bleBless = 2022-04-12
2022-11-15 19:24:50,333: [FortiusAnt, INFO]         bleBlessClass = 2022-04-07
2022-11-15 19:24:50,333: [FortiusAnt, INFO]             bleDongle = 2022-03-24
2022-11-15 19:24:50,333: [FortiusAnt, INFO]             constants = 2022-03-03
2022-11-15 19:24:50,333: [FortiusAnt, INFO]                 debug = 2021-02-22
2022-11-15 19:24:50,333: [FortiusAnt, INFO]        FortiusAntBody = 2022-05-12
2022-11-15 19:24:50,333: [FortiusAnt, INFO]     FortiusAntCommand = 2022-03-03
2022-11-15 19:24:50,333: [FortiusAnt, INFO]         FortiusAntGui = 2022-01-05
2022-11-15 19:24:50,334: [FortiusAnt, INFO]               logfile = 2022-04-07
2022-11-15 19:24:50,334: [FortiusAnt, INFO]            RadarGraph = 2021-02-21
2022-11-15 19:24:50,334: [FortiusAnt, INFO]             raspberry = 2022-01-14
2022-11-15 19:24:50,334: [FortiusAnt, INFO]              settings = 2022-04-15
2022-11-15 19:24:50,334: [FortiusAnt, INFO]       structConstants = 2020-01-25
2022-11-15 19:24:50,334: [FortiusAnt, INFO]             TCXexport = 2021-04-28
2022-11-15 19:24:50,334: [FortiusAnt, INFO]            usbTrainer = 2022-03-01
2022-11-15 19:24:50,335: [FortiusAnt, INFO]              argparse = 1.1
2022-11-15 19:24:50,335: [FortiusAnt, INFO]                 numpy = 1.19.5
2022-11-15 19:24:50,335: [FortiusAnt, INFO]                    os = posix
2022-11-15 19:24:50,335: [FortiusAnt, INFO]                pickle = 4.0
2022-11-15 19:24:50,335: [FortiusAnt, INFO]              platform = 1.0.8
2022-11-15 19:24:50,335: [FortiusAnt, INFO]          sys (python) = 3.9.2 (default, Mar 12 2021, 04:06:34) 
[GCC 10.2.1 20210110]
2022-11-15 19:24:50,335: [FortiusAnt, INFO]                   usb = 1.2.1
2022-11-15 19:24:50,336: [FortiusAnt, INFO]                    wx = 4.1.1
2022-11-15 19:24:50,336: [FortiusAnt, INFO] FortiusANT code flags
2022-11-15 19:24:50,336: [FortiusAnt, INFO]    UseMultiProcessing = True
2022-11-15 19:24:50,336: [FortiusAnt, INFO]                UseGui = True
2022-11-15 19:24:50,336: [FortiusAnt, INFO]          UseBluetooth = True
2022-11-15 19:24:50,336: [FortiusAnt, INFO] ------------------
2022-11-15 19:24:50,359: [FortiusAnt, INFO] FortiusAnt GUI started in child-process
2022-11-15 19:24:50,643: [FortiusAnt, INFO] Version=Fortius Antifier v6.5, on github=Fortius Antifier v6.5.
2022-11-15 19:24:51,339: [FortiusAnt, INFO] Connected to Tacx Trainer T1932
2022-11-15 19:24:51,645: [FortiusAnt, INFO] Motor Brake Unit Firmware=0x1004 Serial= 9702 year=2008 type=T1941 Version2=514 MotorBrake=True
2022-11-15 19:24:51,746: [FortiusAnt, INFO] FortiusAnt applies the MotorBrake power curve
2022-11-15 19:24:51,848: [FortiusAnt, INFO] Retry because short buffer (len=48) or incorrect header received (expected: 0x21303 received: 0xc03)
2022-11-15 19:24:53,150: [FortiusAnt, INFO] FortiusANT broadcasts data as an ANT+ Controlled Fitness Equipent device (FE-C), id=57591
2022-11-15 19:24:53,301: [FortiusAnt, INFO] FortiusANT receives data from an ANT+ Heart Rate Monitor (HRM display), any device
2022-11-15 19:24:53,473: [FortiusAnt, INFO] FortiusANT broadcasts data as an ANT+ Bicycle Power Sensor (PWR), id=57596
2022-11-15 19:24:53,624: [FortiusAnt, INFO] FortiusANT broadcasts data as an ANT+ Speed and Cadence Sensor (SCS), id=57595
2022-11-15 19:24:53,776: [FortiusAnt, INFO] FortiusANT receives commands from an ANT+ Generic Remote Control
2022-11-15 19:24:53,975: [FortiusAnt, INFO] clsBleServer.Open()
2022-11-15 19:24:53,978: [asyncio, DEBUG] Using selector: EpollSelector
2022-11-15 19:24:53,981: [FortiusAnt, INFO] clsBleServer._Server()
2022-11-15 19:24:53,981: [FortiusAnt, INFO] clsBleServer._Server(): BlessServer(name='FortiusAntTrainer')
2022-11-15 19:24:53,983: [FortiusAnt, INFO] clsBleServer._Server(): self.BlessServer.add_gatt(
{'00001826-0000-1000-8000-00805f9b34fb': 
{'00002acc-0000-1000-8000-00805f9b34fb': 
{'Properties': <GATTCharacteristicProperties.read: 2>,
 'Permissions': <GATTAttributePermissions.writeable|readable: 3>,
 'Value': b'\x02@\x00\x00\x08 \x00\x00',
 'value': b'\x02@\x00\x00\x08 \x00\x00',
 'Description':
 'Fitness Machine Feature'},
 '00002ad2-0000-1000-8000-00805f9b34fb': 
{'Properties': <GATTCharacteristicProperties.notify: 16>,
 'Permissions': <GATTAttributePermissions.writeable|readable: 3>,
 'Value': b'@\x02{\x00\xc8\x01Y\x00',
[... 30 lines cut ...]
 'Description':
 'Heart Rate Measurement'}}}
)
2022-11-15 19:24:54,035: [FortiusAnt, INFO] Traceback (most recent call last):

[... cut ...]
2022-11-15 19:24:54,036: [FortiusAnt, INFO] clsBleServer._Server(); add_gatt() exception __init__() missing 1 required positional argument: 'max_write_without_response_size'
2022-11-15 19:24:54,037: [FortiusAnt, INFO] clsBleServer._Server(): self.BlessServer.start()
2022-11-15 19:24:54,037: [FortiusAnt, INFO] , Bluetooth interface n/a; Check exception.
2022-11-15 19:24:54,037: [FortiusAnt, INFO] clsBleServer._Server(): 1 second loop untill Close() called
2022-11-15 19:24:54,038: [FortiusAnt, INFO] clsBleServer._Server() ended
2022-11-15 19:24:54,981: [FortiusAnt, INFO] * * Bluetooth interface disabled *

WouterJD, 2022-11-17:
Bless: If the initiation fails, it may indicate that BLE dongle or driver is not BLE5 compatible.

If both bless AND nodejs give errors, there will be something wrong in the configuration (sorry for the obvious observation) but I have not seen that before nd I do not have suggestions. Perhaps somebody on the forum who has suggestions....

runman19, 2022-11-18:
I thought, that there is no additional dongle for Raspi 4 and bless on the raspi itself required?

If so, what device do you recommend?

Additionally, I could drill down the error a little bit:
a) if I start with -a but not with -n, the bluetooth server seems to work fine until the calibration has stopped. After the calibration, the BTLE server stoppes with an error message.

b) the error message refers to a "newly" implemented feature in bless from e/o August this year:
[https://github.com/hbldh/bleak/pull/962](https://github.com/hbldh/bleak/pull/962)

May be fortius Ant needs to be updated accordingly?

runman19, 2022-11-18:
Oh, actually I see that the thread above was about bleak - but I am getting the same message for bless. I haven't even installed node.js

runman19, 2022-11-18:
As a workaround, I have installed the 2022_Q2-only packages and the server seems to be working now. I am far too tired to test the connectivity to a training app - I will be doing this tomorrow and reporting immediately.

WouterJD, 2022-11-18:
> I thought, that there is no additional dongle for Raspi 4 and bless on the raspi itself required?

Of course, that's windows only

WouterJD, 2022-11-18:
See https://github.com/kevincar/bless/issues/98

WouterJD, 2022-11-19:
As explained by bless developer (see link above); version can be obtained.

It appears I use `bless = 0.2.3`.

runman19, 2022-11-19:
For the actual moment, I recommend using the bless installation mode as scripted in **3_InstallBless_2022Q2_only** . This does not refer to the 2022Q2 issue but rather installs the current development version which is not perfectly tested yes.

The error in the release branch was already resolved by kevin a couple of days ago and is already available as development version - such as installed by the script above. It works without error for me.

As soon as it is tested extensively, it will be available in the official release and this development workaround is not necessary anymore.

closed 2022-11-19 by runman19

WouterJD, 2022-11-19:
Great info, thanks