WouterJD/FortiusANT #403: How to mirror external display horizontally?

runman19, 2022-11-14:
Is there any simple way to mirror the external display horizontally?
As I have had as similar display in my "parts collection",  I am using the previous version of this display here: https://learn.adafruit.com/adafruit-pitft-28-inch-resistive-touchscreen-display-raspberry-pi (it is shown somewhere in the middle of the page). 

Everything works fine except that everything is flipped horizontally. Is there an easy way, to change it? Otherwise I need to deep dive somewhere into the drivers settings.

WouterJD, 2022-11-15:
It shopuld be easy; use the -O command line option to see whether that's what you need.
Manual 4.2 page 41

runman19, 2022-11-15:
I am actually using -O display/90 and the output itself works. But this only rotates but not flips the output. In what file/procedure of FortiusAnt is this -H option implemented?  If it is not coded yet, I can try to code the flipping and perhaps also the touchscreen functionality of this display.

runman19, 2022-11-15:
In the driver description, I can see different screen sizes and also flipping modes supported: 

[https://techatronic.com/st7789-display-pi-pico/](https://techatronic.com/st7789-display-pi-pico/)

These modes just don't seem to be implemented in FortiusAnt

runman19, 2022-11-15:
I have done some code analysis and meanwhile I seem to understand a little bit more: it seems to me that we are currently using the adafruit driver for the ST7789 which looks a bit more basic than the techatronic driver with some more or different capabilites. During the next weekend I'll try my luck to replace the driver and implement some more options like rotating, flipping and resizing.

WouterJD, 2022-11-15:
You could also implement a new class to support the techatronic driver. If you install the adafruit, you'll probably not have those options.

closed 2023-12-15 by WouterJD