WouterJD/FortiusANT #404: Tacx Cosmos/Fortius compatibility

TamasAdam, 2022-11-17:
Hi, I,m Tamas from Hungary. Ibm planning to buy a Tacx Cosmos/Fortius trainer. Is it compatible with FortiusAnt?. I,m not famiar with thos trainer. I attache a picture from the advertisment.
[image]
Thanks in advance.

WouterJD, 2022-11-17:
Hi @TamasAdam, welcome

I think it is. It looks like my Tacx Fortius with th T1932 headunit, connected to computer with USB cable.
Good luck

TamasAdam, 2022-11-17:
Thanks for the prompt reply.

TamasAdam, 2022-11-18:
Hi @WouterJD 
One more question. Whar are the system requirements of FortiusAnt?
I have a laptop win 10, pentium t4300 dual core 2.1 Ghz, 4 Gb RAM, NVIDIAGeForce. 512 Mb. 
For CTP I would use my android mobile. Is it sufficient?

WouterJD, 2022-11-18:
I have no explicit system requirements, probably because we never ran into restrictions.
If python3 is running, you're usually fine.
FortiusAnt has been used on Windows7 upwards.

So I do not expect issues.

TamasAdam, 2022-11-18:
Many thanks.

WouterJD, 2022-11-22:
Yes any magnet will do
Try the magnet, of course size may influence correct functioning

If you see cadence in FortiusAnt it works

WouterJD, 2022-11-22:
Succes

TamasAdam, 2022-11-22:
Thanks. I will buy a Fortius trainer on thursday. But the owner still doesn't find the magnet. I have a few magnets for old bike computers. I will try them.

WouterJD, 2022-11-22:
👍

TamasAdam, 2022-11-26:
Hi Wouter,
I have the Fortius trainer. I'm trying to install Fortiusant and neccessary tools. I choose the executable version. It's running without any problem. But then i tried to install BLE dongle. Installation of node.js went without problem (i think). Then I replaced the driver of the dongle. No problem. Then i tried to install node dependencies as discribed in manual 3.8.3. First problem that there is not any node folder in my Fortiusant folder. I have found node.js folder in Program Files. In command prompt I tried to run "npm install" but error occured. I cannot understand clearly the priblemb but i gev premission for node in firewall but did not helped.             
Log file attached, in case I succeeded.
Can you halp me? I'm not an IT expert at all .
[2022-11-26T08_10_24_188Z-debug-0.log](https://github.com/WouterJD/FortiusANT/files/10095922/2022-11-26T08_10_24_188Z-debug-0.log)

TamasAdam, 2022-11-26:
I went further. I ran command promt as administrator, and in node.js folder typed npm install. The came up less error lines in log file. see  
[2022-11-26T09_08_07_510Z-debug-0.log](https://github.com/WouterJD/FortiusANT/files/10096061/2022-11-26T09_08_07_510Z-debug-0.log)
attached.
When i start Fortiusant, it says bluetooth doerface avaliable (node).
Trainer still hasn't been connected.

TamasAdam, 2022-11-26:
I tried other way. According to manual 3.4 Download Fortiusant from Github. And so on...
At the same step "npm install" goes more further, but at the and some error messages came. See logfile 
[2022-11-26T12_06_33_178Z-debug-0.log](https://github.com/WouterJD/FortiusANT/files/10096224/2022-11-26T12_06_33_178Z-debug-0.log)
Starting Fortiusant Bluetooth interface available. But starting with Simulate trainer option, Bluetooth interface cannat activated, and disabled.
Trainer still hasn't been connected, but i will try it soon.

WouterJD, 2022-11-26:
Why not use bless with -bb?

TamasAdam, 2022-11-26:
Manual says bless needs ble 5 songle. I have asus bt400. Is it compatible?

WouterJD, 2022-11-26:
I don't know, please check at Asus

TamasAdam, 2022-11-26:
Thanks. I gave it a try. In simulating hom trainer mode it says BLE5 required. Specification of bt400 says bluetooth 4.0.
But trainer is now connected and can be seen by FortiusAnt.

TamasAdam, 2022-11-26:
Connected the trainer, but Fortiusant cannot be seen by CTP.

TamasAdam, 2022-11-26:
I managed to complete node.js installation, I had to install Visual Studio.
Then the BLE aktivation problen still hasn"t been sold. And then I ticked "Bluetooth" in in the log file section of setup.
With this parameter in simulation mode CTP can see Fortius trainer. :)
I cannot try it with the trainer itself because it"s too late now.. But tomorrow i will try it keeping my fingers crossed.

WouterJD, 2022-11-27:
Good luck; stepping in the right direction

TamasAdam, 2022-11-28:
Hi Wauter,
It seems finally succeded to make Fortiusant and my trainer working. :)
At the first attempt I had extremly high power numbers. Then I left all the equipment in the garage and searched for similar issues had been solved. I haven't find the exact solution but noticed there are some parameters to set the power ratio. I checked the setup after starting the executable file and found that the "Adjust target power..." parameter was 20(%). Had been changed for 100(%) everything went properly...  almost, because zwift couldn't find the Trainer. Soon I found that I had selected more log file parameters in the setup. After leaving only "Bluetooth" ticked, BLE signal appeard in Zwift.
As I have mentioned previously that in my setup BLE signal appears only in case "Bluetooth" is ticked at log file parameters. For me it hasn't any relevance and I can live with that but maybe it is useful information for you.
Long story short, thank you for this congenious solution that makes VR cycling possible without expensive equipments.  I will put myself onto the FortiusAnt map. :)

WouterJD, 2022-11-28:
> As I have mentioned previously that in my setup BLE signal appears only in case "Bluetooth" is ticked at log file parameters.

I will check

WouterJD, 2022-11-28:
@TamasAdam could you please share the data how FortiusAnt is started (command-line and/or jsonfile) together with the logfile for the situation that Bluetooth is selected but does NOT work.

Thanks

TamasAdam, 2022-11-29:
@WouterJD 
What parameters should I set for logfile?

TamasAdam, 2022-11-30:
@WouterJD 
see attached the json files but file extension must have been changed because json files can't be attached here. The only thing I changed is at the bottom right of the setup window the bluetooth option ticked or not ticked.
[1json_FortiusAntSettings_working.txt](https://github.com/WouterJD/FortiusANT/files/10126485/1json_FortiusAntSettings_working.txt)
[2json_FortiusAntSettings no BLE signal.txt](https://github.com/WouterJD/FortiusANT/files/10126486/2json_FortiusAntSettings.no.BLE.signal.txt)

and one more problem accures, every time I start the exe file, the Target power multiplying factor (-p) is set to 20% despite I change it to 100% and save.

WouterJD, 2022-11-30:
I will investigate. Perhaps it will be next week, since quite busy

TamasAdam, 2022-11-30:
Thanks. It's not an emergency. Finally I can use Fortiusant. :) And its fantastic.

WouterJD, 2022-11-30:
Thanks, but this is an interesting case

WouterJD, 2022-12-28:
> and one more problem accures, every time I start the exe file, the Target power multiplying factor (-p) is set to 20% despite I change it to 100% and save.

Should be solved in versions as of now.
Unfortunately I did not stick to command-line only; now there is json and a user-interface, which complicates things.
But when implemented it should work.

WouterJD, 2022-12-28:
Incorrect usage of () caused the issue; Solved.

WouterJD, 2022-12-28:
Updates in https://github.com/WouterJD/FortiusANT/tree/ant-queue

TamasAdam, 2023-01-06:
> Should be solved in versions as of now. Unfortunately I did not stick to command-line only; now there is json and a user-interface, which complicates things. But when implemented it should work.

Thanks Wouter.:)

closed 2023-03-07 by WouterJD