WouterJD/FortiusANT #406: Ble(ss) install on Macbook question

gitremko, 2022-12-12:
Hi Wouter,  
This is Remko from the behindcorner (achterhoek).  I have been re-enjoying my tacx fortius since 2 years with your great program, thanks a lot! 

I have been working on a bootcamp-windows on my macbook pro with two ant-dongles and Tacx Desktop and sofar with hardly any issue.
This weekend I saw the BLE option with bless directly on the macbook and I tried to get that working but I ran into the issue with 1 missing argument beneath: 

18:26:57,982: Dongle - No (free) ANT-dongle found, Bluetooth interface available (bless)
18:26:57,984: Connected to Tacx Trainer T1932
18:26:58,513: Tacx head unit returns incorrect header 0x21303 (expected: 0xc03)
18:26:58,727: Motor Brake Unit Firmware=0x1007 Serial= 6839 year=2010 type=T1941 Version2=3089 MotorBrake=True
18:26:58,833: FortiusAnt applies the MotorBrake power curve
18:26:59,048: Tacx   - Connected to Tacx Trainer T1932
18:26:59,049: AntHRM - No heartrate monitor connected
18:27:00,064: Ctrl-C to exit
18:27:00,065: Tacx   - Connected to Tacx Trainer T1932
18:27:00,132: clsBleServer._Server(); add_gatt() exception __init__() missing 1 required positional argument: 'max_write_without_response_size'
18:27:00,132: , Bluetooth interface n/a; Check exception.
18:27:01,067: Dongle - No (free) ANT-dongle found, Bluetooth interface n/a; Check exception.
18:27:01,067: * * Bluetooth interface disabled * * 
18:27:01,067: ---------- devices are activated ----------
18:27:01,068: Tacx   - Connected to Tacx Trainer T1932
18:27:01,068: Target=100W  0.0km/h Current=  0W Cad=  0 r=   0 100% t-- 

Do you have any idea what I could do about this, for further information it is running on a macbook pro 2017 with BLE (BT4.2),  python 3.9.7 and all packages from the requirements.txt installed including bless 0.2.4. The only thing I can think of sofar would be that I would need BLE 5 like you prescribe for the the windows-dongles. 

Thank you again for the great work,

Kind Regards
Remko

GewoonGijs, 2022-12-19:
Hi, I am having the exact same error when trying to install the new version with Bless on raspberry pi.

WouterJD, 2022-12-28:
> 18:27:00,132: clsBleServer._Server(); add_gatt() exception **init**() missing 1 required positional argument: 'max_write_without_response_size'

See https://github.com/WouterJD/FortiusANT/issues/408#issuecomment-1360040656
It seems an error in bleak

GewoonGijs, 2022-12-28:
@WouterJD what I found out is that bleak and bless changed their interface and are expecting a different number of arguments since this summer.

WouterJD, 2022-12-28:
According https://github.com/kevincar/bless/blob/master/examples/gattserver.py#L66 the calling sequence is correct

WouterJD, 2022-12-28:
Please refer to: https://github.com/kevincar/bless/issues/92

WouterJD, 2022-12-28:
Updates in https://github.com/WouterJD/FortiusANT/tree/ant-queue

closed 2023-03-07 by WouterJD