WouterJD/FortiusANT #408: Installation broken on Raspberry pi

GewoonGijs, 2022-12-19:
Today I tried to install the newest version of FortiusAnt and now my installation is broken.

- the most recent version (as of after 0.18.1) doesn't use __version__ anymore.
- the most recent version of bless throws an error (see #406)

When shifting back to node.js I got an error on logfile not being initialized.

gitremko, 2022-12-20:
@GewoonGijs  I reverted the bleak package to version 0.15.1  " pip install bleak==0.15.1"  to get rid of the max_write_without_response_size error. 

I did run into a next errror which I cannot recall now, but maybe only going back to the older bleak version might help for you

WouterJD, 2022-12-28:
> The most recent version (as of after 0.18.1) doesn't use **version** anymore.
What code-line causes the break?

GewoonGijs, 2022-12-28:
I guess it is in bleBleak.py line number 75

GewoonGijs, 2022-12-28:
The exception though is thrown because of a missing argument in a call to bless/bleak. I believe the interface has changed last summer. Put I could not find out yet from where this arguments are coming.

WouterJD, 2022-12-28:
> I guess it is in bleBleak.py line number 87

Code removed; coding error to be investigated.

WouterJD, 2022-12-28:
> I believe the interface has changed last summer.

If you find out what is changed, please let me know.

WouterJD, 2022-12-28:
See also #404 and https://github.com/kevincar/bless/issues/92

WouterJD, 2022-12-28:
Updates in https://github.com/WouterJD/FortiusANT/tree/ant-queue

GewoonGijs, 2022-12-29:
I will try later this week. Thanx

GewoonGijs, 2023-01-04:
@WouterJD sorry to say, but I cannot get it to work with Bless. I tried to make a corrected version of Bless that adds the 'max_write_without_response_time' to the Bluezdbus backend.  With this edited version of Bless, I can start FortiusAnt and I see a FortiusAntTrainer being advertised at my iPhone, but I cannot 'koppel' connect to it. The Node.js bluetooth is working fine again. 

I noticed however that when I try to start with -D-1 because I don't use Ant it forces a crash.

MeanHat, 2023-01-08:
Hi @WouterJD and @GewoonGijs - I did 2 sets of further testing:
_**Test 1**_
- FortiusANT **Master**
- Bless
- RPi3 running os11 (Bullseye) - fresh install
- display (ST7789)
- simulation mode
- Zwift

I had to make the following changes to get it to work successfully:
- https://github.com/WouterJD/FortiusANT/blob/master/raspberry/3_InstallBless_2022Q2_only.sh needs to have permissions changed to allow it to be executed in terminal
- https://github.com/WouterJD/FortiusANT/blob/master/raspberry/4_InstallWxPython.sh failed with warning that cp37 is not supported on this platform (armv7l)
- when running FOrtiusAnt.sh with -a -bb -s -D-1 -O display there was (a) no module named lib_detect_testenv (fixed with `pip3 install lib_detect_testenv` ) and (b) no module named dbus_next (fixed with `pip3 install dbus_next`)

_**Test 2**_
- FortiusANT **ant-queue** branch
- Bless
- RPi3 running os11 (Bullseye) - fresh install
- simulation mode
- Zwift

I also had to make the following changes:
- https://github.com/WouterJD/FortiusANT/blob/master/raspberry/3_InstallBless_2022Q2_only.sh needs to have permissions changed to allow it to be executed in terminal
- https://github.com/WouterJD/FortiusANT/blob/master/raspberry/4_InstallWxPython.sh failed with warning that cp37 is not supported on this platform (armv7l)
- when running FOrtiusAnt.sh with -a -bb -s -D-1 -O display there was (a) no module named lib_detect_testenv (fixed with `pip3 install lib_detect_testenv` ) and (b) no module named dbus_next (fixed with `pip3 install dbus_next`)

However, it then failed with the following error:
14:19:02,767: clsBleServer._Server(); add_gatt() exception Rejected send message, 1 matched rules; type="method_call", sender=":1.50" (uid=1000 pid=1595 comm="python3 ../pythoncode/FortiusAnt.py -a -bb -g -s ") interface="org.freedesktop.DBus.Introspectable" member="Introspect" error name="(unset)" requested_reply="0" destination="org.bluez" (bus)

14:19:02,767: , Bluetooth interface n/a; Check exception.

14:19:03,739: * * Bluetooth interface disabled * * 

_In summary, I was able to get FortiusANT to run successfully on os11 RPi3B using current master and Bless; I was not able to get it to run with the ant-queue branch. Hope this helps - I suggest I do not do further testing with the RPi0 until the RPi3B is fully functional_

[See also #394 #404 ]

MeanHat, 2023-01-19:
Hi @WouterJD & @GewoonGijs 

I've done some further testing with the RPi Zero 2W. I have not been successful in getting it to run Bless. I was able to get it to run BLE with node.js on os 10 (Buster) but it needed a number of fixes:

- EAUDITNOPJSON error with npm fixed with `npm init`  to create create a package.json file
- EAUDITNOLOCK error with npm fixed with `npm install` & `npm audit fix`
- "usb.core" module missing error fixed with `sudo apt-get install libusb-1.0-0-dev` and `pip3 install pyusb`
- "No module named lib_programname" error fixed with `pip install lib_programname`
- commented out lines 52 & 53 in FortiusAnt.py `import bleBless' & 'import bleBlessClass` and lines 613 & 614 `logfile.Write(s % ('bleBless', bleBless.__version__ ))' & 'logfile.Write(s % ('bleBlessClass',   bleBlessClass.__version__ ))`

Would it be helpful to allow users the choice of using BLE with node.js or Bless with RPi Zero 2 until we are able to fix Bless?

WouterJD, 2023-03-07:
Hi @MeanHat 

With -b, nodeJs/bleno is used.
With -bb, bless is used.

NodeJS is not ideal, since it's quite big for the purpose.
Bless is not ideal looking at the number of issues caused.

...

MeanHat, 2023-03-09:
Hi @Johanboernl

This is a known issue as reported in https://github.com/WouterJD/FortiusANT/issues/394. This also includes a fix as well as information on other fixes for issues you may encounter.

WouterJD, 2023-03-09:
Hi @johanboernl Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
You could also consider to visit [the sponsor page](https://github.com/sponsors/WouterJD)
- - -

Johanboernl, 2023-03-10:
solved bles issue

The error message indicates that the dbus_next module is missing. 
The bleBless module in your Python code requires this module to be installed in order to run properly. 
You can resolve this issue by installing the dbus_next module using pip, which is a package installer for Python.

You can try installing the module by running the following command in your terminal:

pip install dbus-next

WouterJD, 2023-03-13:
I have installed some rpi's tonight and -with some help on missing modules- it all seems to work well.
I close the issue, if problems persist please reopen and let me know.

closed 2023-03-13 by WouterJD

MeanHat, 2023-03-14:
Great work @WouterJD !