WouterJD/FortiusANT #410: Does FortiusANt connect to FulGaz

charlesyjs, 2023-01-09:
Hi, I use FortiusANT to connect to Zwift successfully but I have an offer for a FulGaz subscription. I believe FulGaz requires the bluetooth and FE-C style connection that FortiuANT provides to Zwift but wondered if anybody has actually used FortiusANT on our older Tacx Flow USB style trainers to connect and use FulGaz.

Thanks
Charlesy

szczurem, 2023-01-20:
> There is one known exception; Tacx Trainer Software 
Hah! I always thought there is something wrong with my installation. I was using Tacx for a while in 2022 and then it stopped working, killing FortiusANT instances all of a sudden...

jujroy, 2023-01-20:
Good afternoon Wouter and users around the World.

Please find some Issues encountered with FulGaz.

I am using 2 Windows 10 laptops and 2 Ant Sticks
T1932
Motor Brake
Using Fortiusant Version 6.4
I start the console using the Windows Executable file.
I use ANT+ - not Bluetooth.

In Jan-Feb 2022, I used Fulgaz and I did not encounter any problems.
I loved the game especially for the quality of their videos across the world.  I am not a big fan of Avatars and Virtual World.

Picture 1 shows the Device Connection page that was taken circa Feb 2022.  Simple – select all 3 options for Trainer, Resistance and Cadence.  FulGaz behaved properly

--------------------------------

Early Jan 2023, I subscribed to FulGaz again.    

See Picture 2 (Jan 2023) for the Device Connection page.  One can now see what I call a ghost  Tacx Neo 2T with ID 0 ??? One can also see a Garmin (2161) ID 57596 for Cadence and RPM connection, furthermore 57595 is also available for Cadence.

I normally select Tacx Neo 2T (57591) for all 3 connections.  If I start pedalling immediately following my ride selection, the system works properly for approx.. 10-15 minutes.  Weird problems start such as locked Grade Slope on the FortiusAnt GUI.  GUI may be locked at 2% while FulGaz indicated -7%.  Wattage and RPM on Fulgaz may track or stop completely.  The system becomes unusable!      

I also tried various combinations of connections for RPM and Cadence i.e. Garmin (57596), Tacx or 57595 without success.
  
I also tested over the last 3 years, Rouvy, Wahoo/RGT, Swift and they all worked as expected. 

I suspect that FulGaz has changed their software over the last 10 months or so ??? which makes it unusable with FortiusANT.  (Ant+ version).  Fulgaz also recommends using Bluetooth for trainer connection which I do not have.

I am switching to Rouvy next month therefore I am not expecting anybody to spend much time looking at this issue but I am curious to know if other users are experiencing similar issues or have a functional system.

I also attached the json file renamed as a JPEG

Thank you so much.  Cheers

[image]
[image]
[image]

WouterJD, 2023-03-07:
I see @bikebeppe64 has completed a FulGaz today https://www.strava.com/activities/8674537646.
So it does work somehow.

closed 2023-03-07 by WouterJD

jujroy, 2023-03-07:
Thank you for this.  Now riding on Rouvy.  I may retry FG next Fall.  Cheers

WouterJD, 2023-03-07:
:-)