WouterJD/FortiusANT #412: Bluetooth interface error (bleno)

barsk, 2023-01-15:
Hi!
I'm from Sweden. Been using an old T1901/T1932 combo for years with that troublesome TTS4 and wanted to move to newer software. And so stumbled on to FortiusANT. Great piece of software!

I have everything up and running on a **Win10** computer. Trainer is connected and running. However my Asus BT400 with bleno-library seems to not be working. I have used Zadig to change drivers to WinUSB, and I have installed node.js and run npm install in the node directory. I get this error:
Bluetooth interface cannot be opened; module 'logfile' has no attribute 'fLogfile'

What can be wrong?

WouterJD, 2023-01-16:
Hi @barsk Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
You could also consider to visit [the sponsor page](https://github.com/sponsors/WouterJD)
- - -

To start with; the logfile error is solved in next version and available (python only) in the current branch. Sorry for that.
The -b bleno implementation is quite complex due to thge number of components required.
You could consider the -bb bless implementation, which is easier but requires a BLE/5 dongle.

I'm pretty busy these days but plan to publish current development asap.
Please let me know...

barsk, 2023-01-16:
Trying the direct Python route then. I seem to have gotten a windows version of Python311 installed in my C:\ directory by something I did yesterday related to FortiousANT. Not sure what it was (and a bit concerning). But never mind that, I used that installation and ran pip install -r pythoncode/requirements.txt.
It stumbled on attrdict that I needed to install also. But then I got:

`Requirement already satisfied: service_identity in c:\users\krist\appdata\roamin
g\python\python311\site-packages (from -r requirements.txt (line 7)) (21.1.0)
Collecting wxPython
  Using cached wxPython-4.2.0.tar.gz (71.0 MB)
  Preparing metadata (setup.py) ... error
  error: subprocess-exited-with-error

  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [12 lines of output]
      Traceback (most recent call last):
        File "<string>", line 2, in <module>
        File "<pip-setuptools-caller>", line 34, in <module>
        File "C:\cygwin64\tmp\pip-install-wb0lwthw\wxpython_6d8f420b1e024826a106
183adcdf34ac\setup.py", line 27, in <module>
          from buildtools.config import Config, msg, opj, runcmd, canGetSOName,
getSOName
        File "C:\cygwin64\tmp\pip-install-wb0lwthw\wxpython_6d8f420b1e024826a106
183adcdf34ac\buildtools\config.py", line 30, in <module>
          from attrdict import AttrDict
        File "C:\Python311\Lib\site-packages\attrdict\__init__.py", line 5, in <
module>
          from attrdict.mapping import AttrMap
        File "C:\Python311\Lib\site-packages\attrdict\mapping.py", line 4, in <m
odule>
          from collections import Mapping
      ImportError: cannot import name 'Mapping' from 'collections' (C:\Python311
\Lib\collections\__init__.py)
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem wit
h pip.
error: metadata-generation-failed
`
I also tried requirementsNOGUI.txt and go a few other requirements covered. Decided to give it a try:

`$ python.exe FortiusAnt.py -a -g -H0 -A -l -b
Traceback (most recent call last):
  File "C:\FortiusANT\pythoncode\FortiusAnt.py", line 52, in <module>
    import bleBless
  File "C:\FortiusANT\pythoncode\bleBless.py", line 41, in <module>
    from bless import (
  File "C:\Users\krist\AppData\Roaming\Python\Python311\site-packages\bless\__in
it__.py", line 41, in <module>
    from bless.backends.winrt.server import (  # noqa: F401
  File "C:\Users\krist\AppData\Roaming\Python\Python311\site-packages\bless\back
ends\winrt\server.py", line 20, in <module>
    from bless.backends.winrt.ble import BLEAdapter
  File "C:\Users\krist\AppData\Roaming\Python\Python311\site-packages\bless\back
ends\winrt\ble\__init__.py", line 1, in <module>
    from .adapter import BLEAdapter
  File "C:\Users\krist\AppData\Roaming\Python\Python311\site-packages\bless\back
ends\winrt\ble\adapter.py", line 5, in <module>
    from pysetupdi import devices  # type: ignore
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ModuleNotFoundError: No module named 'pysetupdi'
`
 I'm a Java guy so I'm stuck here. Ideas?

szczurem, 2023-01-20:
Hello to the entire FortiusANT community! I have exact same problem with starting python version of fortiusant (adapter.py, actually). Tried Python 3.11, 3.7.9, same problem with 

      from pysetupdi import devices # type: ignore
      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
      ModuleNotFoundError: No module named 'pysetupdi'

pytsetupdi is an odd module, almost absent from the internet. Not even sure what module it is part of.

---------
Since this is my first post: I use (exe version, two ANT+ dongles) of FortiusANT with Tacx Fortius and Rouvy since 2021. Decided to try the Python route, attempting to add functionality of Nanoleaf lights reacting to either power generated or virtual gear selected. I think I have the code ready, but cannot test the integrated version ...

szczurem, 2023-01-20:
Found the problem. requirements.txt are missing this line:
`git+https://github.com/gwangyi/pysetupdi; platform_system == "Windows"`

Or you need run the following:
`python -m pip install git+https://github.com/gwangyi/pysetupdi; platform_system == "Windows"`

closed 2023-03-07 by WouterJD