WouterJD/FortiusANT #415: Raspberry bullseye BLE issues

kjoglum, 2023-01-21:
Have been running FortiusANT (BLE/Nodejs mode) with success over a couple of years on Raspberry Pi 3b+ (buster), running Zwift on an iMac (Zwift Companion app on iPhone historically needed as bridge in my case), and love your work. However, as part of a system crash, I needed to start fresh.

Having read through different issues, my case seems to be a combination.

I have installed bullseye, and cloned latest FortiusANT repo. As reported by @MeanHat, I also needed to run `pip install lib_detect_testenv` and `pip install dbus_next` before getting FortiusANT trainer up and running. Following the manual, I initially went down the bless route (-bb).  When running `TestbleBless.sh`, even with the trainer at rest, values are reported at incremental increase/decrease (which might be the intention, but I choose to mention it anyways). Having the FortiusANT trainer up and running, without log errors, I attempted starting ZC app on iPhone / Zwift on iMac, although ended up with intermittent connectivity, and no response when starting a ride (ZC app running on iPhone 11/iOS 16). 

Decided to switch to the BLE/Nodejs approach (-b). Initially, I got the `Bluetooth interface cannot be opened; module 'logfile' has no attribute 'fLogfile'` as reported by @barsk, but after adding -d b (bluetooth debug) the error was gone, and bluetooth interface opened. Tried the same approach running ZC app / Zwift on iMac, but same intermittent connectivity as with bless (no error reported in debug logs). Switched bluetooth setting from ZC app to built-in bluetooth within Zwift on the iMac (BLE 4.0), and suddenly I managed a successful connection between FortiusANT and Zwift, with expected behaviour during a ride.

A couple of sidenotes to my issues:
- When trying to connect directly to Zwift app on iPhone, the issue persists, not able to get connectivity. Which might indicate some iPhone connectivity issue: Neither Zwift directly nor Zwift Companion app along with Zwift on MacOS will work.
- Tried LightBlue debugging from iPhone, but when trying to connect to the FortiusANTTrainer, the connection attempt stops with message: "The peripheral disconnected while being interrogated". So, there seems to be an issue connecting to iPhone in general.
- Historically, when running FortiusANT, my Garmin FR 745 has also automatically detected the trainer components when published, but does not do so anymore (and is not able to search/find the components).

kjoglum, 2023-01-23:
Continuing troubleshooting, and have switched to running the Zwift app on Apple TV HD, 4th gen (BT 4.0). Also switched (git checkout) from using the steering or master branch to ant-queue branch on my RPi. 

When attempting -bb (bless), I still have intermittent connectivity (NO SIGNAL appearing at regular intervals in Zwift), and the steering feature does not seem to connect at all. When running -b from ant-queue branch, I get expected/desired connectivity, including steering feature, and I manage to carry out a ride.

So, I guess direct connectivity to iMac/ATV when running -bb is unlikely due to both running BT 4.0..? Still, I do not understand why neither iPhone nor my Garmin FR 745 are able to connect, neither for -bb nor -b.

kjoglum, 2023-01-24:
I guess this issue can be closed, as I have identified a satisfactory solution for my use case: Use BLE/Node js approach, and connect directly to Zwift on iMac/ATV or similar (BLE 4.0). Added a ANT dongle to the RPi, and then also achieved parallel data transfer to my Garmin FR 745.

erwin4u-alt, 2023-01-28:
Hello, thank you guys! Had the same issues (missing packages not working BLE) but now it works also on my side with my configuration: Tacx T1932, RPI 3+, ATV

WouterJD, 2023-03-07:
I'm sorry to close this one, but I do not have the devices to test the issue.
Also sorry for the missing dependencies, which are most likely caused by dependencies in other packages (that do not define their dependencies well).
And since you got your systems running, I hope you are ok.

If not, feel free to reopen.

closed 2023-03-07 by WouterJD