WouterJD/FortiusANT #416: Problem with bluetooth connection

peterlp75, 2023-01-22:
Hi, does someone have this issue with bluetooth connection, When i start FortiusAnt program says bluetooth disabled, I don't know how to enable it.

[image]

MeanHat, 2023-01-23:
Hi @peterlp75 - I have had this error and used the logfile to diagnose it. Have you tried that?

peterlp75, 2023-01-24:
Hi, thanks for answer
I'm not that familiar with FortiusAnt, can you explain it to me with more detail?

MeanHat, 2023-01-24:
Hi @peterlp75, one suggestion is to look at the manual [https://github.com/WouterJD/FortiusANT/wiki](url). On page 42 it describes how to turn on the log file and select what you want logged. You could use  `-d log` in the start line (see page 27, section 3.9). This should show you what prompts the "Bluetooth interface disabled" warning. 
You can also post the log file to the community here so that others can help you if you are stuck.