WouterJD/FortiusANT #421: Read from USB trainer error

tkathulevej, 2023-02-25:
Hi,

I'm I the only one getting this error now and then:

[image]

This happens not always. - Maybe 1 out of 10 training sessions.. I seems also to never happen at low intensities (only after longer intervals at +200W I would say).

In the beginning I thought this had something to do with maybe a faulty Fortius motor unit overheating, but now I have had the opportunity to use FortusANT with 2-3 different Fortius trainers (T1941/1932) and even different kombinations of motor and head unit. - And I still see the error now and then :-/    ....Only way to recover from is to fully shut the FortiusANT program and restart (...meaning your training session in BigRing VR which I use is spoiled.

I've not tried to run Fortius with logging. Maybe I should?

BR /tka

Thomas Kjaer-Andersen, Denmark

WouterJD, 2023-02-25:
The real issue is that there is no error recovery on the serial onterfa è to the headunit. If insufficient data us received, FortiusAnt tries to recover, but... Not always successful.

Om my trainer it also happens sometimes, properly reconnecting helps. The error also happens when the mains cable is not properly connected; perhaps it causes interference or so.

WouterJD, 2023-02-25:
Any suggestions welcome, no solution found until now

WouterJD, 2023-02-25:
Logging will help you to understand the problem but will not help me to investigate. It will confirm that records are received incomplete

WouterJD, 2023-02-25:
I assume you disabled pedal stroke analysis, causing to tune down the sampling frequency. Not all headunit manage to respond fast enough

tkathulevej, 2023-02-25:
...Okay, thx for the information. - And yes, stroke analysis disabled.... /tka

WouterJD, 2023-03-07:
Hi how are things?

closed 2023-03-13 by WouterJD

WouterJD, 2023-03-16:
See also #425

WouterJD, 2023-03-17:
See https://github.com/WouterJD/FortiusANT/issues/425#issuecomment-1473961356

tkathulevej, 2025-01-06:
FYI:

...Just adding an additional comment to this old issue:

This problem (for me) slowly got worse and worse. After a while I found that pulling the USB plug from Fortius head unit AND killing FortiusANT session followed by starting a new one (without calibration), would most often make me able to continue stucked training sessions instead of losing them entirely.

However things continued getting worse, and one day I desperately searched my drawers for something that could make some miracle. And I found this:

[image]

....I.e. an USB cable at least APPEARING more expensive that the formerly used. And not any longer than absolutely necessary either. 
  
An guess what. IT DID THE TRICK 😁😁   ...From then on, I've run numerous training sessions and I've not experienced one single failure.   ....So this might be a tip for others annoyed by same type of problems.

BR /tka

Denmark

WouterJD, 2025-01-06:
Hey @tkathulevej that is a good solution!
Connection problems can surely improve using better cabling/connectors; not the least because of a better fit and/or corroded connectors.

Great that you are training again!

WouterJD, 2025-01-06:
I've added this to manual section 5.11