WouterJD/FortiusANT #425: Tacx head unit returns insufficient data (raspberry slow)

WouterJD, 2023-03-15:
FortiusAnt on raspberry is working fine
But when in another window, the Enter key is pressed...
and when the Enter is is released, we turn back to normal operation

The message "Tacx head unit returns insufficient data" is therefore caused by the Raspberry where FortiusAnt is running.
[Check video on Youtube](https://youtube.com/shorts/3v5SHleuc78)

Who has experience with this issue
How could this situation be detected
Or even better: avoided?

[Python code](https://github.com/WouterJD/FortiusANT/blob/master/pythoncode/usbTrainer.py#L2480)

WouterJD, 2023-03-15:
Interesting; this does not happen in the calibration loop

WouterJD, 2023-03-17:
This problem occurs when -b (BLE/Nodejs) or -bb (BLE/bless) is activated.

It seems that the raspberry is not fast enough to do all that work.
Why it affects the USB-interface to the Tacx-trainer is not clear to me.

WouterJD, 2023-10-09:
Anybody still experiencing this problem?
If not, I will close the issue.

MeanHat, 2023-10-09:
Happy to close Wouter

closed 2023-12-15 by WouterJD