WouterJD/FortiusANT #426: Does FortiusANT work on Windows7?

CarstenRLV, 2023-03-19:
Hi,
I'm starting FortiusAnt with FortiusAnt.exe -a -g -bb -A -H0
but getting the error message:
bleak.exc.BleakError: Only Windows 10 is supported. Detected was ('7', '6.1.7601', 'SP1', '')
I have not read anywhere that there is a restriction to W10, is there anything I can do?
Regards
Carsten

WouterJD, 2023-03-19:
FortiusAnt sure works on windows 7.
Evidently, bleak dies not. Bleak is a subcomponent of bless.

I understand you want to use BLE (why?) , so use Node.js which does work on windows 7.

Or upgrade to windows 10
Or use ANT

WouterJD, 2023-03-19:
@CarstenRLV Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
You could also consider to visit [the sponsor page](https://github.com/sponsors/WouterJD)
- - -

WouterJD, 2023-04-06:
Hi @CarstenRLV 
Since there is no communication here, I assume can be closed.
If not happy to reopen

closed 2023-04-06 by WouterJD