WouterJD/FortiusANT #427: Probleme de vitesse avec kinomap

regis65, 2023-03-19:
Bonjour,j ai mis la nouvelle version la 6.7 et j ai toujours le même problème la vitesse n est pas la même que sur kinomap,
Voir photo:
[image]
[image]
Sportivement

WouterJD, 2023-03-19:
@regis65 bonjour. Tu utilise toujours le Fortius avec T1932 headunit? Quel moteur/frein (T1942??)

WouterJD, 2023-03-19:
+ Tu me peu envoyer une imprime de l'écran noir (console); est-ce-quíl-y-a des messages?

WouterJD, 2023-03-20:
_Looking at your screen; Kinomap asks you to cycle -30%. It's quite steep!_
_You're going 91 rpm and that produces 52 watts, bringing the speed to 112 km/h. It's not surprising._
_But -30% is not realistic._
_Rouvy has the same problem and that's why the -G parameter was introduced._

_Manual page 42:_

---
Regardant ton écran; Kinomap te demande de cycler -30%. C'est assez raide !
Vous roulez à 91 tr/min et cela produit 52 watts, ce qui porte la vitesse à 112 km/h. Ce n'est pas surprenant.
Mais -30% n'est pas réaliste.
Rouvy a la mêmem problème et c'est pourquoi le paramètre -G a été introduit.

Manuel page 42:

-G valeur Modifier la note demandée avec un facteur/factorDownhill ; voir section 6.4 "PowerCurve pour i-Flow (T1901-T1932)". Notez que GradeAdjust n'est pas actif en mode manuel.
FortiusAnt -a -g --> la valeur par défaut : targetGrade = 100 % de la note demandée
FortiusAnt -a -g -G50 --> recommandé lors de l'utilisation de Rouvy : targetGrade = 50 %
FortiusAnt -a -g -G50/50 --> targetGrade = 50% (en montée) et 25% en descente (50% * 50%)
FortiusAnt -a -g -G100/0 --> targetGrade = montée uniquement (remplace l'ancien drapeau -u)
FortiusAnt -a -g -G0/0 --> targetGrade = toujours plat

regis65, 2023-03-22:
Salut je t envoie l ecran noir
[image]

regis65, 2023-03-22:
Salut,je t envoie les photos de l ecran noir
[image]

[image]

regis65, 2023-03-25:
Salut,wouter je t es envoyé les photos de l ecran noir sur Github.
Sportivement
Regis

WouterJD, 2023-03-25:
Merci Regis.
j'ai vue les photos; pas des messages inquiétant.

Alors je reste avec la raccomndation de essayer avec -G

regis65, 2023-03-26:
Comment fais t on pour le changer

WouterJD, 2023-03-26:
Aussi dans le GUI tu peut changer le parameter - G
Regarder le manuel
Peut etre cest meux de demander un ami pour aider et Faire les expwrimwnts

regis65, 2023-03-26:
Salut, tu pourrais m'envoyer le manuel svp.
Merci
Regis

regis65, 2023-03-26:
Je connais personne qui pourrait me le modifier.

WouterJD, 2023-03-26:
Parce que c'est toi!

[image]

Cliquer sur 
[image]

[image]

"Modifier la note demandée avec un facteur/facteur en descente (-G)"

Normalement 100% / 100%; dans ton cas tu pourrait specifier 100/10, et ça signifie que -quand kinomap demande une descente de -30%, FortiusAnt línterprete come -3%.

PS. Le manuel en englais se trouve a: https://github.com/WouterJD/FortiusANT/blob/master/doc/FortiusANTUserManual.pdf

Succes!

regis65, 2023-04-02:
Salut,je l es réglé c est toujours pareil au niveau des kilomètres heures je suis a 50 sur fortius et sur kinomap je suis 12 kilomètres heures .

WouterJD, 2023-04-02:
FortiusAnt montare la vitesse de la rouw.
Kinomap calcule la vitesse, basé sur power e grade.

C'è Sint deux ciffres completament different

WouterJD, 2023-04-06:
@regis65 laisse-moi savoir ci s'est claire maintenant

WouterJD, 2023-04-11:
Since there is no communication here, I assume can be closed.
If not happy to reopen

closed 2023-04-11 by WouterJD

regis65, 2023-05-14:
Salut,
Sa fonctionne toujours mal avec kinomap,j ai plus de dénivelé  les
kilometre son pas les mêmes sur kinomap et ton application je ne sais pas
quoi faire,sa m embête d en acheter  un nouveau .
Sportivement
Regis

regis65, 2023-05-14:
Salut,
Sa fonctionne toujours mal avec kinomap,j ai plus de dénivelé  les kilometre son pas les mêmes sur kinomap et ton application je ne sais pas quoi faire,sa m embête d en acheter  un nouveau.
Sportivement 
Régis

WouterJD, 2023-05-14:
FortiusAnt fournit puissance et cadence à kinomap.  Kinomap traduit cela en vitesse.  Cette vitesse n'a aucun rapport avec la vitesse de la roue (la vitesse affichée dans FortiusAnt).  Alors ne t'en fais pas

WouterJD, 2023-05-14:
Kinomap fournit à FortiusAnt un pourcentage de pente sur la base duquel la résistance est calculée.  Vous pouvez influencer cela avec les paramètres (-G).

WouterJD, 2023-05-14:
Je lai ecrit april2, puis no response. Maintenant meme question. Questeque tu attend de ma part

regis65, 2023-05-14:
Moi je pensais que l home trainer et l aplication (kinomap)aurait la même
vitesse,et quand la vidéo montre u dénivelé l aplication (fortiusant)montre
le même dénivelé

WouterJD, 2023-05-14:
FortiusAnt: vitesse = vitesse de la roue
Kinomap, Zwift, rouvy...: vitesse = f (power, degree, weight)

regis65, 2023-05-14:
Désolé, je n es pas compris

WouterJD, 2023-05-14:
Ignorez la vitesse affichée par FortiusAnt et commencez à vous entraîner !