WouterJD/FortiusANT #428: Fortiusant setup seems to not calibrate correctly.

WouterJD, 2023-03-20:
Hi all, and thank you Wouter for this very good project.

I am in Stavanger, Norway. 

Setup is this:
- Fortius Multiplayer USB
- win10 laptop running Fortiusant 
- ant+ dongle (Garmin)
- win10 laptop running Zwift
- ant+ dongle (Tacx antenna thing)

 I get the message I should have been a pro cyclist or my equipment is not calibrated correctly. 

Zwift shows what I believe is at least 5x actual power.
The same power is shown in Fortiusant.
Calibration in Fortiusant uses the full countdown time, but does not give any error.

Any help with this issue would be greatly appreciated.

_Originally posted by @Bjohje79 in https://github.com/WouterJD/FortiusANT/issues/14#issuecomment-1455080810_

WouterJD, 2023-03-20:
Hi @Bjohje79 

If the calibration on your setup takes the full time, your tyre may be too loose.
Try some adjustment there.

You can also do a rundown, so you have a better feeling on what happens (instead of the automatic calibration).

And you can do a test-run, using the manual mode (-m) and select the resistance using the headunit.

The Fortius Multiplayer is a great machine and usually can be setup to get working correctly.

szczurem, 2023-04-01:
> Zwift shows what I believe is at least 5x actual power

I believe the problem might be an odd power multiplier put in the default settings. I had a similar problem - open settings and check if you do not have 0.2 (or 5, I don't remember) put in a "power", "factor" or "calibrate rolling resistance" section of the configuration file. On of the production versions was published with this as a default.

WouterJD, 2023-04-02:
The two factors you mention are part of the Power curve adjustment and I think it works as described:
-c value	Calibrate the rolling resistance for magnetic brake. See section 6.4 “PowerCurve for i-Flow (T1901-T1932)”. Default is 15. If the power, displayed by FortiusAnt is higher than your power-meter, specify -c 14 to reduce the RollingResistance by 1 Newton.
-p value	Adjust target Power by multiplying by this factor for static calibration.
If the power, displayed by FortiusAnt is always 10% higher than your power-meter, specify -p90 to reduce resistance by 10%.

As @szczurem describes, -p 20 would cause the problem; either specified on the command line or the json file.

szczurem, 2023-04-03:
> As @szczurem describes, -p 20 would cause the problem; either specified on the command line or the json file.

That was my issue. After an upgrade some time in 2022 the power numbers got crazy. It looks like you released the package with .json in the windows executable folder with that adjustment?

WouterJD, 2023-04-06:
@Bjohje79 
Since there is no communication here, I assume can be closed.
If not happy to reopen

Bjohje79, 2023-04-12:
Hi, thanks for all the feedback. As I am new to all this I had several issues. First issue of not calibrating within countdown was solved by getting a trued wheel (!) Did not realize the one I had was so bad... 
Still when using my setup in Bkool STP today, it seems I get to high power output. Will investigate more the next days...

WouterJD, 2023-04-12:
Hi @Bjohje79 An important factor to realize is, that the old-skool trainers were never developed with the modern setup in mind; "real live video" was new in the world and the fun was to be training and seeing what happens.
Absolute power-measurement was not in the game by then.

When using trainer road, one of the first tests is an FTP-test. Let's say, the result is 456 Watts.
All training is then relative to that 456Watts; 50% - 90% - 110%.

The absolute figure is not relevant in that case.
Today, the outcome of the test is compare to calibrated power-meters.

So, if your power outcome is 456Watts and you estimate it should be 228 - use -c50 and the readings will match.

This all is not lineair, but hey - we're using hardware technology from +15 years ago!

Bjohje79, 2023-04-12:
Hi again, I realize it is an old setup, I believe that is part of the fun 😉 Getting good use out of old school quality products "abandoned" by the producers and mainstream makes a lot of sense, at least in a resource conserving perspective.
Big thumbs up to you for the great effort put in, an even more so for supporting your development in this excellent way!

Will check with different -c and -p settings soon.

Sendt fra Outlook for Android<https://aka.ms/AAb9ysg>

WouterJD, 2023-04-12:
👍

Bjohje79, 2023-05-01:
Hi again. Have not used my setup much lately, but after getting a "true" wheel the calibration works fine. Most importantly, thanks to tips from @szczurem I changed the multiplier factor in the json file, and voila. No more "Superman" like powers output to ctp. Much more enjoyable with realistic inputs, now I can put in that extra effort keeping up with a group at similar power or see someone more fit pace away from me 🤪
Thanks again for keeping this hardware and forum alive!

WouterJD, 2023-05-01:
Hi, please explain what you changed in the json-file; perhaps there is an error there ...

szczurem, 2023-05-08:
Apologies if this is off topic, but I started to play with calibration fine-tuning and realise there is perhaps a missing functionality there.
My scenario is this. 
1. I do a long workout, getting the tire up to temperature. 
2. Auto-calibrate Tacx Fortius (again). 
3. Do another workout, this time with a parallel measurement of power sensing pedals.
4. Compare the average power, getting a benchmark difference between heated up, stable and calibrated Fortius and Garmin pedals.

Now what I am missing is the possibility to **statically enter an absolute measure of "calibration correction" in such state - without doing the initial calibration upon starting FortiusANT.** This not only saves time, but also resolves the problem of insufficiently stable reading and a subsequent drift as the tire heats up.

I know I might be chasing too much precision, but hey - it is always good to tweak out whatever we can from the equipment we have.

WouterJD, 2023-05-08:
Hi @szczurem  your message triggers a thought I had before.

First some backgroup on calibration; a special command to the Fortius brake is used to let it run at a predefined speed (20 km/hr) untill a stable resistance is found. That resistance is then further used as Calibrate value.

The thought was to store that value and use it as a preset value to skip the calibration, assuming that the calibration value would be the same everytime you use the system. Of course, initially it would be incorrect but after 5 minutes (or so) the calibration value would be ok. I did not make it.

You CAN skip calibration with the **-n flag** (no calibration) and work with -p to adjust power (with a percentage, no lineair offset).

If you have a clear idea, I'm happy to assist to add it (why not).

Perhaps, first experiment with -n and -p.
You can also look in the code; https://github.com/WouterJD/FortiusANT/blob/master/pythoncode/FortiusAntBody.py#L961

Always happy to hear from you!

szczurem, 2023-05-08:
The easiest "mixed" approach I can think of is reporting the "Calibrate value" in the console (or GUI even) after a normal calibration. Then you can introduce a numerical value to the -n flag for static, but "calibrated" calibration. I suspect that -n assumes some default - we just need a way to influence that default.

rgerber001, 2023-09-27:
Hello Mr. Wouter,
I am appreciative and impressed with FotiusAnt. I have made a contribution in hopes I can get it working on my setup!
I started in 2005 with an iMagic. I loved it! I burned out the green magnetic brake, around 2006, so when Tacx offered a Fortius upgrade, I bought it. Now I have a Fortius Multiplayer on the original (green-accented) iMagic frame with the steering attachment.

With **FortiusAnt**, here is what I see:
1. When I connect it all, load the drivers and start FortiusAnt.exe (using 2 BT 4.0 devices), it starts fine, and it suggests kicking the pedal to start calibration.
2. I rotate the crank, and the wheel/motor begins spinning, settling at 20kph for over 30 sec.
3. The speed then drops to 7.5kph, and the middle "dial" no longer shows that it is calibrating.
4. It continues at 7.5kph indefinitely, but (never stops) unless I use the rear bike brake or something to stop it.
5. Before I start it or while the calibration is running, I can steer full left or right, and the console relates that it resolved the steering limit(s).
6. I suspected that maybe the 6 conductor cable might not be connecting well between the power/control brick and the Fortius head unit (this stuff is 15 years old or more...), so I examined it and cleaned all contacts. I did this for the other cables and sockets as well. That has not resolved the issue. 
7. I setup Zwift, but I did not see any FortiusAnt-related choices in the hardware setup dialogs, so now I am just stalled.

If you have seen this pattern before, I hope you have a couple of simple options to try. I would appreciate any feedback that may occur to you as to what might be going wrong.
I'll attach my .JSON (renamed as .txt) in case it is helpful...
I did not send a log file, but will if you think it would be helpful.

I appreciate in advance any suggestions or troubleshooting tips you can share.

Regards,

Roy Gerber
Southwest Florida, USA

[FortiusAntSettings.json.txt](https://github.com/WouterJD/FortiusANT/files/12732129/FortiusAntSettings.json.txt)

WouterJD, 2023-09-27:
Hi @rgerber001 
Welcome to the FortiusANT community.
Thanks for the contribution, which is very much apreciated.
- - -

Based upon your description, the FortiusAnt / USB side works well.
I did not yet get a problem description like yours, starting to calibrate but then <as described>.
My first reaction was to check all cabling and you already did.

What you might try is to skip the calibration (-n flag) to see whether there is something odd there.

Unfortunately, Zwift will see FortiusAnt only after calibration when the interface is really active. (I'm not so happy about that and have it on my whish list to improve)

A screen print at the moment that "nothing happens" anymore is usefull; please include the console AND the interface.

closed 2023-10-12 by WouterJD