WouterJD/FortiusANT #429: Tacx Training Desktop App (Win) crashes FortiusANT

szczurem, 2023-04-05:
I occasionally use FortiusANT with TTDA on windows (current version, Ant+) and now 80% of the time it crashes Fortius with the following. The TTDA then "detects" Tacx Neo2T, but shows no wattage. Same config no problems with Rouvy. 

`Exception in thread Thread-1:
Traceback (most recent call last):
  File "threading.py", line 932, in _bootstrap_inner
  File "threading.py", line 870, in run
  File "FortiusAnt.py", line 486, in Tacx2DongleThread
  File "FortiusAnt.py", line 167, in Tacx2Dongle
  File "FortiusAntBody.py", line 676, in Tacx2Dongle
  File "FortiusAntBody.py", line 1098, in Tacx2DongleSub
  File "usbTrainer.py", line 689, in Refresh
  File "usbTrainer.py", line 2553, in SendToTrainer
AssertionError`

The weird thing is that things work great for a spell of time (I was able to start it day after day last month), and then stops. The last time I got the error with ver 6.7, I managed to start FortiusANT 6.4 without problems. Today both fail to work (the lines reported are different, but the error is the same). The current version of Tacx software is 4.41.0.0, it says it was upgraded on 29 March (but I was using it on the 1 April with FortiusANT 6.4).

WouterJD, 2023-04-05:
I'll have a look tonight

WouterJD, 2023-04-06:
Hi
Line 2553 = "assert(self.UserAndBikeWeight >= 0x0a)              # Avoid surprises"

Could it be that you have defined a UserAndBikeWeith of less than 10 kilo's?

I have updated usbTrainer.py; so the python version should recover from a weard setting. Or do you need windows.exe?

WouterJD, 2023-04-06:
Please refer to https://github.com/WouterJD/FortiusANT/blob/master/pythoncode/usbTrainer.py#L559

And let me know whether the message "UserAndBikeWeight is set to %d, which is below the expected minimum of 70kg; corrected to 85kg." appears and what value is reported.

szczurem, 2023-04-06:
Success! Indeed, I do now get:
"UserAndBikeWeight is set to 0, which is below the expected minimum of 70kg; corrected to 85kg"
The message was repeated three times within 0.5 second and then it stopped.

I take it user and bike weight are taken from the trainer software? That in turn is taken dynamically from Garmin Connect in Tacx case. So perhaps there is a short period when Tacx downloads the weight parameters and was reporting zero and that was killing USBtrainer.py? I tried to start FortiusANT after Tacx and that did not help though.

szczurem, 2023-04-14:
I run the code and it works - the error popped out numerous times for about 30secs, then stopped, allowing trainer detection by Tacx software. From then on it seems to work well.

I am afraid I am not experienced enough in debugging python code to trace what value of the userweight is eventually set.

WouterJD, 2023-04-14:
I can add that if you like

szczurem, 2023-04-14:
Many thanks, but I guess need to learn to do such stuff on my own. So far my coding learning curve is very erratic- I did a fork of your code to play with live led wall panels changing colours with power generated and it kind of works, but I still have not learned how to make sure I have the latest versions from the repository.

closed 2023-04-14 by WouterJD