WouterJD/FortiusANT #431: TACX Bushido 1980

gibabuscape, 2023-05-03:
I have a Bushido T1980 and I would like to know if there is anyone who has already installed FortiusAnt with Bkool? Are there any specific instructions for the T1980? @WouterJD

WouterJD, 2023-05-03:
Hi @gibabuscape 

I assume you have read the manual and managed to get FortiusAnt linked to your bushido; **please confirm that this is correct.**
From that moment onwards, your bushido is not communicating with Bkool, but FortiusAnt does.

Please provide some more details
- what set-up do you use
- what happens after starting FortiusAnt - how far do you get
- does any other CTP work in your environment?

gibabuscape, 2023-05-03:
> Please, see below the erro when  started FortiusAnt

[image]

gibabuscape, 2023-05-03:
> > Please, see below the erro when  started FortiusAnt
> 
> [image]

[FortiusAntSettingjson.txt](https://github.com/WouterJD/FortiusANT/files/11389794/FortiusAntSettingjson.txt)

WouterJD, 2023-05-05:
Hi @gibabuscape the problem is caused by an imort from bless and is resolved in the current version; please update.
See https://github.com/WouterJD/FortiusANT/blob/master/pythoncode/FortiusAnt.py#L644

gibabuscape, 2023-05-05:
WouterJD, tks a lot for you support.

I don't know Python, is it possible to have the executable?

Enviada em: sexta-feira, 5 de maio de 2023 10:08
Para: WouterJD/FortiusANT ***@***.***>
Cc: Gilberto Faria ***@***.***>; Mention ***@***.***>

WouterJD, 2023-05-05:
> is it possible to have the executable?

Yes, I just uploaded it

gibabuscape, 2023-05-05:
Tks man!!!

Enviada em: sexta-feira, 5 de maio de 2023 11:06

WouterJD, 2023-05-07:
Good luck!

gibabuscape, 2023-05-09:
Hi Wouter. Not yet because my computer is in maintenace. As soon as possible I Will update you.  Tks Verry much

Obter o Outlook para iOS<https://aka.ms/o0ukef>

Hi @gibabuscape<https://github.com/gibabuscape> how are things proceeding?

bredy73, 2023-05-11:
> I have a Bushido T1980 and I would like to know if there is anyone who has already installed FortiusAnt with Bkool? Are there any specific instructions for the T1980? @WouterJD

I think is not compatible, same as kinomap. At the moment it works with Rouvy and Zwift.

WouterJD, 2023-05-11:
Of course, I do not doubt that there are issues.
I know that many of the known CTP’s like Zwift, Trainer Road, Rouvy, Sufferfest, Golden Cheetah, Road Grand Tour, BigRingVR, TTS work with FortiusAnt. FortiusAnt uses a standard ANT-protocol, not tailored to any of thesae CTP's.

It might be an idea to use bluetooth and see what happens.

Could you please provide some more information on the pairing process and where things go wrong?

WouterJD, 2023-05-11:
With FortiusAnt and Bkool on the same Windows 10 PC, after activating FortiusAnt, Bkool finds the FortiusAnt ANT-devices right away. Note you need two ANT-dongles; for FortiusAnt and for Bkool.

@bredy73 and @gibabuscape I hope it helps, please let me know.

[image]

[image]

WouterJD, 2023-10-09:
Since there is no communication here, I assume can be closed.
If not happy to reopen

gibabuscape, 2023-10-09:
> Enviada em: segunda-feira, 9 de outubro de 2023 07:46

Good morning

I apologize deeply for not having dedicated this topic, but I promise that as soon as I can test it again and if necessary, I would like to thank you for leaving the doors open to talk to you.

Best Regards
Gilberto Faria

WouterJD, 2023-10-09:
Hi @gibabuscape 

No problem at all; I'm waiting for your inputs and then we'll see whether I can help.

bredy73, 2023-11-08:
> With FortiusAnt and Bkool on the same Windows 10 PC, after activating FortiusAnt, Bkool finds the FortiusAnt ANT-devices right away. Note you need two ANT-dongles; for FortiusAnt and for Bkool.
> 
> @bredy73 and @gibabuscape I hope it helps, please let me know.
> 
> [image]
> 
> [image]

I do not why, I use two dongle ant+, but my Bkool not recognized the bushido.

WouterJD, 2023-11-08:
Hi @bredy73 thanks for the update, this is as designed and looks good.

---
What do you mean with last phrase;
- did it not work in the past and does it work now 
- does it not work now?

bredy73, 2023-11-09:
For me it doesn't work. I don't know why.

WouterJD, 2023-11-09:
The screen looks good, note that FortiusAnt at the ANT+ side is "Tacx Neo" for the dimple reason that the non-Ant tacxes are not defined on the interface. So use tacx neo

bredy73, 2023-11-11:
I think the problem is the Ant+ because it doesn,t recognize even the cadence, speed and heart belt sensor.

bredy73, 2023-11-12:
My solution:
Connect only one Ant+ dongle, start Fortiusant.
Start Bkool and now connect the second Ant+ dongle. Now Bkool pairing Bushido and other sensor.

closed 2023-11-28 by WouterJD