WouterJD/FortiusANT #432: No Tacx trainer found

ProYoran, 2023-05-06:
Hallo, ik ben Yoran Krol en kom uit Noord-Brabant. Ik heb een Tacx Genius gekocht en 2 ANT+ dongles. Ik wil de software met 1 pc gebruiken.

Ik heb de software gedownload en de Switchabl driver geïnstalleerd. Als ik FortiusANT open en op "Locate HW" klik, staat er "No Tacx trainer found".

Wat is het probleem?

WouterJD, 2023-05-07:
Om de tacx Genius te laten werken moet je FortiusAnt vertellen dat het zo'n machine is; dat wordt niet automatisch gedetecteerd.

Zie manual https://github.com/WouterJD/FortiusANT/blob/master/doc/FortiusANTUserManual.pdf, pagina 55.

Start FortiusAnt met **-t Genius** of selecteer dat trainer type in de instellingen, zie hoofdstuk 4.3 op pagina 45.

Succes en anders hoor ik van je.

WouterJD, 2023-05-09:
> Dat had ik al gedaan.

Da's mooi.
Ik heb toch de indruk dat FortiusAnt op zoek gaat naar een Tacx trainer die via een USB-kabel is aangesloten en dat is toch bij jouw trainer NIET het geval?

"No Tacx trainer found" is een bericht dat hoort bij de USB-aangesloten trainer.
Bij de Genius verwacht ik berichten als "Pair with Tacx Genius", "Tacx Genius paired: %d", ...

Door FortiusAnt te starten met `-d f` of zelfs `-d all` krijg je meer berichten op het console en in de logfile; succes - zou moeten lukken!

ProYoran, 2023-05-09:
Na herstart is het gelukt, ik weet alleen niet hoe ik resistance krijg. Die voel ik niet.

WouterJD, 2023-05-09:
Vertel wat meer ;
Welke CTP gebruik je
Wat zie je op FortiusAnt
Wat zie je op je CTP

closed 2023-10-09 by WouterJD