WouterJD/FortiusANT #435: Stuttering Fortius on longer high power stints

szczurem, 2023-07-20:
I am not sure this is really a FortiusAnt issue, or a hardware one, but the major remaining trouble I have with my Fortius setup is the stuttering at high wattage. For ~240W+ continuous output, the brake tends to momentarily literally brake. Power reported drops to zero than things return to normal, but pedaling becomes a nightmare, as you have to accelerate on a high gear back to the previous cadence.
The stuttering tends to get more frequent over time - sprint finishes are definitely out of question.

I recall some users talking about firmware upgrade resolving the issue. Any hints if this is indeed the problem, where to find the firmware and how to do it?

WouterJD, 2023-07-20:
Hmmmm. The first thing I'd think of is the cadence sensor. No cadence, no power

szczurem, 2023-07-20:
I wish it was that simple! It does happen sometimes, but my problem is different. For sensor issues cadence falls to zero first then, after about 5s,  power is cut completely until the sensor is fixed. In this case, the cut off is instant and very short lived.

szczurem, 2023-09-24:
> Sprint at the end of a ride and very likely it will lock up repeatedly. 
Thanks - you are describing exactly my problems. More frequent lockups towards the end of the ride could indicate heat or movement (as the rider gets more tired?) being a problem. I can't remember well, but it may indeed be that the issue was already there before FortiusANT.
Weather is wonderful still, so it has been a while since I have been riding indoors (and I managed to borrow a more modern direct-drive kit for a while).

WouterJD, 2023-09-27:
Hey, interesting discussion.
Please let me know how I can help.

You could always make a trace of the USB-interface.
Please describe in detail what happens.

What is a "lock-up"? Does the resistance of the brake suddenly goes up? What does the interface say (resistance is displayed)?

oflahero, 2023-09-27:
Hi Wouter. Yes, the "lock-up" happens when the resistance immediately goes to maximum, and your legs are left stuck, mid-stroke, and then it immediately eases, and you have to wind up the power again from zero (and you are usually in a high gear at this point ).

I suspect it's related to Fortius' notorious low-cadence issue. At the end of a ride you're more tired than you think and  are trying to sprint in a high gear at a relatively low cadence with tired legs.

I will try running FortiusAnt with '-d log' next time anyway and will post it here when the issue recurs.

szczurem, 2023-09-27:
> I suspect it's related to Fortius' notorious low-cadence issue. At the end of a ride you're more tired than you think and are trying to sprint in a high gear at a relatively low cadence with tired legs.

I do not think it is cadence-related in my case. The scenario described by @oflahero also occurs in high power - high cadence scenarios (although indeed more frequently over time). With higher power output (~400W) such stuttering is certain to happen within seconds even at the beginning of the ride.

WouterJD, 2023-09-27:
It is important to keep the Fortius at high wheel speed, low wheel speeds are not handled well.

I have done a lot of experimenting here and therefore implemented the "virtual gears".
Note that the physical gears and wheel speed are COMPLETELY irrelevant (even though that sounds very strange); all that matters is cadence and power. You can confirm this, having experience with the direct drive: there is no wheel, hence no wheelspeed and no confusion.

So in case of a steep slope, where high power and low cadence is the natural behaviour, the following may happen. I have put two approaches next to each other:

| Virtual gear solution | Physical gear stuttering | 
| ---- | ----- | 
| Zwift tells FortiusAnt the slope you are riding (see note below) | Same |
|  **Keep your bike at the highest gear** (==> high wheel speed at low cadence) | Ride your bike at lower gear, corresponding to slope |
| Cycle at low cadence | Same |
| FortiusAnt calculates required power based upon wheelspeed and slope% | Same |
| Because of the high wheel speed / combined with the slope (e.g. +10%), the required power would be excessive |Although the required power is realistic, Fortius brake does not work well with low wheel speed |
| **Now use the virtual gearbox**, "shifting" to a lower gear. FortiusAnt will adjust the calculation and require a lower power, correspoding to your abilities | |
| The power you produce is communicated to Zwift and **Zwift calculates the speed** (using slope% and power) | Same |
| The displayed wheel speed on the FortiusAnt console (not relevant for your excersize) will be higher than Zwift speed | Wheel speed may correspond to Zwift speed |

Although at first sight you might think there is some cheeting here, this is not the case:
- The brake is set at a certain resistance
- You ride your bike at a certain cadence (which results in wheel speed, which results in power)
- The power you produce is communicated to Zwift without conversion
- Zwift calculates the displayed speed

Note:
Where Zwift displays a slope%, Zwift already communicates a lower number to FortiusAnt to reduce the required power (knowing that usually bikes are not fit for the slopes used). This is different for Rouvy where the real slopes are communicated and the -g parameter can be used to reduce it.

szczurem, 2023-09-27:
This is a very eloquent explanation of the virtues of the virtual gears, which I love and almost always use (and abuse - a clash with reality of actual gears available at 20% slope is painful). However, this problem occurs even with high (80-90rpm) cadence/high wheel speed in my case. What is more, it is actually more annoying, as you have to struggle with a sudden stop and subsequent spinning back to the high cadence.

WouterJD, 2023-09-27:
Hi @szczurem thanks for reply.

The question to answer is: who causes the sudden behavioural change, FortiusAnt-software or the Tacx-Fortius-brake?

Of course the expected question: can you reproduce this?
And if you can:
- please provide a log (-d log) and I will have a look.
- together with a description "the lock-up occurred at, when [I was cycling as before | suddenly increassed cadence to overcome a hill | whatever other important change you noticed]"

oflahero, 2023-09-27:
I would very much suspect the Fortius brake here, as I think I recall similar happening after long rides just using the Tacx software, long ago.

It can be improved (although I've no hard data here) by ensuring your wheel pressure is high (120-130psi) before applying the brake and calibrating. But it won't eliminate the issue.

I went for a VO2 intervals Zwift lunch ride today, and could feel the brake beginning to lock up *very slightly* during the latter efforts. But at the end I just sprinted very hard and experienced a brake lock. Here's a link to the '-d all' log. It should be during the big effort near the very end: 
https://www.dropbox.com/scl/fi/gjb9nmd58utveu9fzi9o3/FA_log.zip?rlkey=qv9iodtac41rs9va3lzt1x6s9&dl=0

Again though I would be very surprised if it were FortiusAnt's issue.

WouterJD, 2023-09-28:
@oflahero this is usefull information.
I'm looking at the provided JSON file at line 11783, see attached excel sheet and image (below).

TargetMode=2  Grade = 1 resulting in a required power of 347 Watt and the target sent to the brake of 4342.
The brake reports back that the current resistance is 3450 and the current power is calculated as 275 Watt.

Without that a significant different required resistance is sent to the brake, The current resistance moves up to 5216, 4903, 5508, 6015, 5647, 6317, 6449 which means a current power of 400...537 watts.

Cadence drops, but the reported wheelspeed is increasing (which is strange if you do not change gears).

After 20 measurements (5 seconds) the current resistance drops again to normal values.

Something similar happens at line 13102.

[image]
[FortiusANT JSON Analysis.xlsx](https://github.com/WouterJD/FortiusANT/files/12753931/FortiusANT.JSON.Analysis.xlsx)

WouterJD, 2023-09-28:
Since the Brakes resistance jumps up without FortiusAnt requiring to do so, I would like to confirm that it's not a software error.

> I recall similar happening after long rides just using the Tacx software

In that case, fiddling with FortiusAnt-parameters would not solve the issue either.

If it would be overheating I do not see why the issue would be solved within 5 seconds without actually stopping and cooling down.
It could be a hardware issue (headunit or USB interface) which resolves itself; but the data does not seem completely corrupted

oflahero, 2023-09-30:
Great analysis, thank you. It's a relatively minor problem considering the benefit I've had from being able to extend the life of my ancient Fortius, so much appreciated.

WouterJD, 2023-09-30:
@oflahero thanks for reply.
If the problem were caused by overheating, could it help to cool the brake?
If the problem would exist in the headunit, it might be worth to try to install another headunit.

PS. Did you ever introduce yourself? It seems all I know is that you have a Fortius :-)
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
You could also consider to visit [the sponsor page](https://github.com/sponsors/WouterJD)
- - -

oflahero, 2023-10-03:
I did! Back in 2020 I think. I'm on the 'Put yourself...' page. I've a Fortius VR trainer with video reality/VR software, standard blue and white head unit. I bought it second-hand here in Ireland in 2016. All USB IDs for the dongles I bought are probably in that log file!

I have had three good years out of what really should have been a dead Fortius, thanks to FortiusAnt, but will probably be treating myself to a direct drive trainer very soon as the Fortius is really starting to get rusty, and a pain to maintain.

WouterJD, 2023-10-09:
Hi @oflahero you sure made a [remark](https://github.com/WouterJD/FortiusANT/issues/14#issuecomment-605529276)! I now know you're from Ireland!

> Fortius is really starting to get rusty

I think that is the real reason to buy a new trainer and I'm happy we could move that date for at least 3 years!
After all the effort in creation FortiusAnt I would certainly be very sad to scrap the trainer and since I have two that dat willl be in future.

I sure believe a direct drive trainer is better that the Fortius, the hardware sure made evolution!

WouterJD, 2023-10-09:
What shall we do with this issue? I suggest to close, since it seems hardware which I cannot resolve.

closed 2023-10-12 by WouterJD