WouterJD/FortiusANT #437: Tacx training app can't find my genius multilayer trainer

ArcaBeina, 2023-08-24:
Hi!
My name is Artis, im from Latvia and i don't understand codes and that stuff.  I recently bought a used tacx genius multiplayer and i would like to connect it to tacx app but it can't find my trainer. I have a windows 10 laptop, T2021 motorbrake, Black track T2420 and a T028 ant+ antenna.

WouterJD, 2023-08-24:
Welcome @ArcaBeina 

First step is to read the manual, try the correct script with the correct parameters (-t tacxtype). I will assist when back from holiday.

WouterJD, 2023-08-28:
Hey @ArcaBeina If you react, I can help.
https://github.com/WouterJD/FortiusANT/blob/master/doc/FortiusANTUserManual.pdf is the manual, if you start there it should be doable.

If not, just drop a message,

WouterJD, 2023-10-09:
Hi @ArcaBeina I pick up this issue after some holidays, sorry for delay...
Do you still need help?

I would have to dig into the Genius again, since I do not have that hardware.
Please confirm first you're still busy with the issue.

closed 2023-10-12 by WouterJD