WouterJD/FortiusANT #439: Fortius

PUSKAS6, 2023-10-10:
Hi
I have a Tacx Fortius, and I would greatly appreciate it if you could give me instructions to how I can get to use it again.
I will sponsor you monthly if you can help.
Cheers Jeff

WouterJD, 2023-10-11:
Hi Jeff, welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
- - -

The manual with installation- and usage instructions can be found [here](https://github.com/WouterJD/FortiusANT/blob/master/doc/FortiusANTUserManual.pdf)

PUSKAS6, 2023-10-12:
Hi
Thanks for your email.
My brake  = T1941-50.
Head = T1932.
Frame = T193005.
Do you think that there is a chance that it could actually work again?
Cheers Jeff.

WouterJD, 2023-10-12:
Hi Jeff, if your Fortius is working well, there is no reason at all why it would not work. Asking for a chance? I would say 100%.

It seems you look for a quick start, check the manual section 3.3 - assuming you use windows.

By the way, where are you located? country/city?

PUSKAS6, 2023-10-12:
Hi (I'm not sure how I address you),
I am in New Zealand, The Bay of Islands, Russell.
I really appreciate your help.
Cheers Jeff.

WouterJD, 2023-10-13:
Hi Jeff, my name is Wouter and I'm located in the Netherlands.

PUSKAS6, 2023-10-13:
Cheers Wouter, nice to meet you.

PUSKAS6, 2023-11-01:
Hi Wouter,
Can you please help me?
I have installed The Manual,USB drivers and Executable, I do not understand
what I have to do next.
Cheers Jeff.

WouterJD, 2023-11-01:
Hi Jeff

If I understand well, you're a Windows user and considering your questions section 3.3 is suited for you.
You say you have "installed" the software; I assume you have downloaded to it your computer.
So you are at bullet 2 (in section 3.3).

I would suggest you now execute bullets 3 (installation of the USB drivers) and then 6 (Start FortiusAnt) and tell me what happens; please include a screen-print of the results.

Succes!
Wouter

PUSKAS6, 2023-11-02:
[image: image.png]

PUSKAS6, 2023-11-02:
[image: image.png]
This is what I got when I unclicked the Pedal Analysis.
Cheers Jeff.

>
>
>
>
> Message ID: ***@***.***>
>>
>

PUSKAS6, 2023-11-02:
[image: image.png]

On Thu, Nov 2, 2023 at 1:42 PM Jeff Tucker ***@***.***> wrote:

> [image: image.png]
> This is what I got when I unclicked the Pedal Analysis.
> Cheers Jeff.
>
>>
>>
>>
>>
>> Message ID: ***@***.***>
>>>
>>

PUSKAS6, 2023-11-02:
Green light on head is flashing.

On Thu, Nov 2, 2023 at 1:53 PM Jeff Tucker ***@***.***> wrote:

> [image: image.png]
>
> On Thu, Nov 2, 2023 at 1:42 PM Jeff Tucker ***@***.***>
> wrote:
>
>> [image: image.png]
>> This is what I got when I unclicked the Pedal Analysis.
>> Cheers Jeff.
>>
>>>
>>>
>>>
>>>
>>> Message ID: ***@***.***>
>>>>
>>>

PUSKAS6, 2023-11-02:
Hi Wouter
I've got it to work,is it possible to use the Real Life Videos with this
setup?
You are very helpful and I greatly appreciate it.
Cheers Jeff.

On Thu, Nov 2, 2023 at 1:55 PM Jeff Tucker ***@***.***> wrote:

> Green light on head is flashing.
>
> On Thu, Nov 2, 2023 at 1:53 PM Jeff Tucker ***@***.***>
> wrote:

WouterJD, 2023-11-02:
Hi Jeff.

It seems you post your comments, using email replying on posts.
The texts of your replies appear in github, not the images.

Please use github in a browser: https://github.com/WouterJD/FortiusANT/issues/439

---

When asking for Real Live Videos', it seems you refer to Tacx' old software, which is discontinued by tacx; I understand it does not work anymore.

The idea is:
- You have your Tacx Fortius
- You connect it to FortiusAnt using USB
- FortiusAnt broadcasts an ANT-signal (using an ANT-dongle)
And thus has converted you Tacx Fortius into a "modern" trainer, using an open protocol.

Then you install a "standard" (non proprietary like Tacx Training Software) packages, such as Zwift or Rouvy which also use an ANT-dongle and communicate with your Tacx.

Zwift is a virtual environment, Rouvy has real-live video's.

WouterJD, 2023-11-03:
Hey Jeff how are things going?

PUSKAS6, 2023-11-03:
It works , thanks so much.Cheers Jeff.Sent from my iPhone

WouterJD, 2023-11-03:
Hey Jeff, that's good news.
Tell me more, probably some photo's of your setup. What CTP do you use?

PUSKAS6, 2023-11-07:
Hi Wouter
I'm not sure what CTP is,can you explain it to me?
Cheers Jeff.

WouterJD, 2023-11-07:
CTP = cycling training program

TSS was the first, tacx' proprietary software.
Modern CTPs are Trainer Road, Zwift, Rouvy,... and many more. These programs communicate to a trainer through a standard protocol (e.g. ANT)

PUSKAS6, 2023-11-08:
Hi Wouter
I am just using my Fortius as a stand alone trainer ,ie no CTP.
Cheers Jeff.

WouterJD, 2023-11-08:
Hi Jeff, does this mean you use it in manual mode only?

Regarding your question regarding "Real Life Videos" I assumed you'd be interested in Rouvy or so

PUSKAS6, 2023-11-08:
Hi Wouter
Yes ,manual only.
If I wanted to use Rouvy ,what would I have to do and would I be able to
use my old RLV's?
Cheers Jeff.

WouterJD, 2023-11-08:
Rouvy asany other CTP use two dongles.

Old RLV? I think not, the videos are comprised in the Rouvy license. Start with test license for free and check what it does

PUSKAS6, 2023-11-09:
Hi Wouter
When I start ,I have to recalibrate each time , is this what should happen?
Cheers Jeff.

WouterJD, 2023-11-09:
Yes it's normal (as designed).
I have never tested whether the calibration always ends in the same calibration values.

The reason to calibrate is to get a baseline which is dependant on tire pressure, temperature and roller-tension.

If you don't like it, you can disable using the parameterization

PUSKAS6, 2023-11-13:
Hi Wouter,
I'm sorry to bother you again,but I've used it as a trainer twice and each
time after about 5 minutes the screen goes blank and the PC turns off.
Have you any ideas ?
If I'm too much trouble, please let me know.
Cheers Jeff.

WouterJD, 2023-11-13:
Hi Jeff, no worries - keep asking, I hope I can help.

First thing I would do is check the Windows power settings. Check what has been set for 5 minutes non-activity. Shutting down is not a usual action but I think it could be.

Note that there is no mouse/keyboard activity, so the PC should remain on at least as long as you want to train.

PUSKAS6, 2023-11-13:
Cheers Wouter, I'll try that.

PUSKAS6, 2023-11-14:
Hi Wouter
It worked ,thanks heaps.
Cheers Jeff.

WouterJD, 2023-11-14:
@PUSKAS6 Jeff; Great that things work. If there is anything, just let me know.
Thanks for sponsorship, which motivates to keep going.

I assume this issue can now be closed, we can always reopen if needed.

closed 2023-11-14 by WouterJD

PUSKAS6, 2023-11-14:
Hi Wouter
Thanks for all your help.
Cheers Jeff.