WouterJD/FortiusANT #440: WxPhyton installation failure on Raspberry 3B+:  wxPython-4.1.1-cp37-cp37m-linux_armv7l.whl is not a supported wheel on this platform.

orrmany, 2023-11-10:
WxPhyton installation fails on Raspberry 3B+: with error message:
 wxPython-4.1.1-cp37-cp37m-linux_armv7l.whl is not a supported wheel on this platform.

output of 'uname -a:' Linux raspberrypi3b 6.1.21-v7+ #1642 SMP Mon Apr  3 17:20:52 BST 2023 armv7l GNU/Linux

Printout of the install script: 
pi@raspberrypi3b:~/FortiusANT/raspberry $ ./4_InstallWxPython.sh
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: gdown in /usr/local/lib/python3.9/dist-packages (4.7.1)
Requirement already satisfied: six in /usr/lib/python3/dist-packages (from gdown) (1.16.0)
Requirement already satisfied: tqdm in /usr/local/lib/python3.9/dist-packages (from gdown) (4.66.1)
Requirement already satisfied: filelock in /usr/local/lib/python3.9/dist-packages (from gdown) (3.13.1)
Requirement already satisfied: requests[socks] in /usr/lib/python3/dist-packages (from gdown) (2.25.1)
Requirement already satisfied: beautifulsoup4 in /usr/lib/python3/dist-packages (from gdown) (4.9.3)
Requirement already satisfied: soupsieve>1.2 in /usr/lib/python3/dist-packages (from beautifulsoup4->gdown) (2.2.1)
Requirement already satisfied: PySocks!=1.5.7,>=1.5.6 in /usr/local/lib/python3.9/dist-packages (from requests[socks]->gdown) (1.7.1)
/usr/local/lib/python3.9/dist-packages/gdown/cli.py:126: FutureWarning: Option `--id` was deprecated in version 4.3.1 and will be removed in 5.0. You don't need to pass it anymore to use a file ID.
  warnings.warn(
Downloading...
From (uriginal): https://drive.google.com/uc?id=1Uk1TSc6iLArx14QH8q85c9ytrw0IKLIn
From (redirected): https://drive.google.com/uc?id=1Uk1TSc6iLArx14QH8q85c9ytrw0IKLIn&confirm=t&uuid=f7fa5c09-b40e-49c7-b0e8-6ead504b4a13
To: /home/pi/Downloads/wxPython-4.1.1-cp37-cp37m-linux_armv7l.whl
100%|█████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 120M/120M [00:16<00:00, 7.35MB/s]
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
ERROR: wxPython-4.1.1-cp37-cp37m-linux_armv7l.whl is not a supported wheel on this platform.
 Pre-built Python is installed, press Enter to continue:

WouterJD, 2023-11-17:
Hi @orrmany 
Unfortunately, I'm not a raspberry specialist and do not have an answer for you.
I have had FortiusAnt running on a Rspberry 3; but that's some time/years ago.

Perhaps somebody else reading this issue has a suggestion. @meanhat ?

MeanHat, 2023-11-18:
Hi @orrmany

Please excuse me if you already know this but if not hopefully it might help. The PiWheel version for wxPython that is used in the installation guide is 4.1.1. This is only compatible with Debian 11 (Bullseye) - see [https://www.piwheels.org/project/wxpython/](url) for details.

You can check which version you are operating using [cat /etc/os-release](url).

In order to proceed you could either install Bullseye (if you not already running it) or try using a wxPython that is compatible with your version of Debian.

Please let @WouterJD and me know how you get on.

WouterJD, 2023-11-18:
Hey @MeanHat thanks for joining in

orrmany, 2023-11-20:
> Hi @orrmany
>
> Please excuse me if you already know this but if not hopefully it might help. The PiWheel version for wxPython that is used in the installation guide is 4.1.1. This is only compatible [...]
Hi @MeanHat, here it is: 

pi@raspberrypi3b:~ $ cat /etc/os-release 
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

MeanHat, 2023-11-26:
HI @orrmany 

Sorry for the delay. If you don't need the FortiusANT GUI then you can skip installing WxPython in step `4_InstallWxPython.sh` which would then avoid the problem altogether. 

If you really need / want the GUI then please let me know and I can spend some more time recreating the problem and looking for a solution.

MeanHat, 2023-11-26:
> HI @orrmany
>
> Sorry for the delay. If you don't need the FortiusANT GUI then you can skip installing WxPython in step `4_InstallWxPython.sh` which would then avoid the problem altogether.
>
> If you [...]

I should have added that with a CTP such as Zwift you can manage very easily without the GUI. I run RPi3 with Zwift on Tacx Fortius.

orrmany, 2023-11-28:
Hi @MeanHat , sorry for the slow response: it would be nice to have the GUI, indeed. My guess: I have a freshly installed (i.e., latest version of everything) Raspberry. I guess this  "freshness" might break the installation process, i.e., a new wheel(?) might need to be built.... 
I am happy to produce debug logs, as needed, just instruct me what to do.

decodeais, 2023-12-15:
I always starting with this old compressed [image](https://github.com/WouterJD/FortiusANT/issues/407#issuecomment-1358388594). It is working out of the box and you have only to start via  start menue. If you want to install the latest version it is a way to make the the Raspi "unfresh". If you have another Ant-Stick you might have to reinstall the stick.

MeanHat, 2023-12-16:
That's a great suggestion @decodeais, Thank you for sharing.

@WouterJD - would it be useful to add a link to the image in the User Manual in the Raspberry Pi chapter?

WouterJD, 2023-12-16:
Yep.

I never studied "artefacts"; the .pdf and .exe should de stored different from how it's done now as well. Something to do in January I think

decodeais, 2023-12-16:
> That's a great suggestion @decodeais, Thank you for sharing.
> 
> @WouterJD - would it be useful to add a link to the image in the User Manual in the Raspberry Pi chapter?

It is an old version without steering. If you have updated it, somebody should try to compress it too. The result is a little bigger then the pure OS-image  but it is so much easier to use. I think it helps you, because it give everybody  the standardized  environment. Most questions concerning installation will be solved.

orrmany, 2023-12-16:
Well, it seems I somehow managed to install wxPython. I tried many things, but while trying to follow https://wxpython.org/blog/2017-08-17-builds-for-linux-with-pip/index.html I noticed that 
`pip download wxPython`
resulted in downloading a wxPython wheel (which is different what the script tries to install):
`wxPython-4.2.1-cp39-cp39-linux_armv7l.whl`
so I issued a 
`pip install wxPython-4.2.1-cp39-cp39-linux_armv7l.whl`
and "it just worked (TM)". 
:) 

Now I check my W0, too

orrmany, 2023-12-16:
Now on my Raspberry W0 I tested:
`pip install -U pip
pip install -U six wheel setuptools
pip download wxPython`
has cached 
`wxPython-4.2.1-cp39-cp39-linux_armv6l.whl`
then 
`pip install wxPython`
succeded, but for the succesful run of the test as per the install tutorial:
`python -c "import wx; a=wx.App(); wx.Frame(None,title='hello world').Show(); a.MainLoop();"`
I needed to install one more additional library (which was also needed for the 3B+, to be honest):
`sudo apt install libpcre2-32-0`

sudo apt install libpcre2-32-0