WouterJD/FortiusANT #441: Doesn't detect TacX Flow Multiplayer T2250

ktippey-hzdr, 2023-11-11:
Just purchased an older T2250 planning to run it with Zwift, but running into all kinds of issues. Running on Windows 10 with an ANT+ stick. Tried your software (Windows executable) but Locate HW just leads to 'No Tacx trainer found'. USB is plugged in. Not sure what I'm doing wrong or if there's a bug?

WouterJD, 2023-11-17:
Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, wqhere are you from?
You could also consider to visit [the sponsor page](https://github.com/sponsors/WouterJD)
- - -

Some applications connect as many ANT dongles as they can find, some applications start in the background and use a dongle even when not active (e.g. Garmin Express). These applications also do not worry if there is no dongle, the keep trying.

So check whether some other software is running and using an ANT-dongle, I bet you willl find one.
If not, let me know and we will further investigate.

ktippey-hzdr, 2023-11-18:
Hi, and thanks for the welcome! From the US but living in Germany.  I did some investigating (device manager) and realized that the device driver for my second ANT+ stick was the issue. Your Fortius software saw the stick and was trying to use it but I believe Zwift had a problem with it, thus the (!). **After installing a new driver, it works!**  Pretty amazing what you've done here

I did have an intermittent-type failure when running Zwift where the software starting scrolling a lot of errors and Zwift froze up with the last received data just staying constant.  I restarted the software and it again worked, it just messed up my last interval of the day.  I'm not sure if that's something other people have noticed or how to address it?  

Also, I have some experience with python and app programming and I'd be interested in helping with the programming side if/when that might be needed.  I was pretty psyched to see this app existed after realizing the used trainer I had just bought wasn't supported by Zwift.  

I saw that you addressed steering in a recent update, do you have recommendations for where to try that out in Zwift?

WouterJD, 2023-11-18:
Hey Kristy thanks for your update. Great to see that you'd like to support on the python side.
Most things are quite stable these days; obviously no new trainers are added and in the end FortiusAnt is just the woman-in-the-middle enabling the Zwift-trainer communication.

The steering works only through BLE using a private Zwift protocol which has been opened by one of the contributors.
I do not use Zwift myself (for test purpose only). I understand and recall it's for (some) zwift rides that you can actually steer and position yourself on the road/track; I do not know whether this is always active.

ktippey-hzdr, 2023-11-19:
Ah yes I now see a comment about ANT+ not supporting the steering .... I'll have to grab a bluetooth dongle and give it a try. It looks like Repack Rush is the new Zwift steering route, but certain controllers (not sure if ours will count) allow you to steer around the regular roads. Thanks!

ktippey-hzdr, 2023-11-19:
I guess I have one question... is it possible to make an option that triggers the app to auto-restart when the usb connection fails?  At the moment it just spews error messages and continues sending the last known data to Zwift until I restart it then it works again

WouterJD, 2023-11-19:
Hi @ktippey-hzdr

It would be possible, the question was not asked before though.

Please check the USB cable to be properly connected. Especially All cabling, including the brake's mains cable. The headunit picks up electrical noise easily and then misfunctions.

If problem persists please provide info, such as screen shot.

ktippey-hzdr, 2023-11-19:
The connections are tight, those are the first things I checked. I'm concerned the issues might be due to connecting through a usb hub, but I only have a few usb connections available on my laptop, so I can't really connect another way. Things break worse when I try to connect straight into the computer with the usb and through the hub with an ant+ stick.  I will provide screenshots next time it happens, so far it averages 2-3x times per hour

WouterJD, 2023-11-19:
I have not experienced problems with ANT+ in USB hub. I have trainer on laptop.
Try to connect directly to computer and move other devices to hub.

WouterJD, 2023-11-19:
Regarding error recovery on the UsbDevice combined with your proposal to asist on the python side...

https://github.com/WouterJD/FortiusANT/blob/master/pythoncode/usbTrainer.py#L404
Is where the UsbDevice is connected

https://github.com/WouterJD/FortiusANT/blob/master/pythoncode/usbTrainer.py#L2427
This is the only place where the UsbDevice is read.

You might try to add some code in  USB_Read() 
- to detect the error
- to do whatever is neede to recover from the error

It's not possible (easy) to leave the USB-reading loop, go back to the point where the USB-device is connected and then restart due to the setup of the software.

If you find out what is needed.
As soon as you know, it could mean that that the GetTrainer() function must be split into two (GetTrainer and GetTrainerSub) where GetTrainerSub could be called from UISB_read(). Of course the connection and initialization code should not be copy-pasted.

Happy to hear from you!

ktippey-hzdr, 2023-11-21:
I found a way to plug my trainer and the ant+ sticks straight into the laptop and no further issues

WouterJD, 2023-11-21:
Great. Best solution