WouterJD/FortiusANT #442: indieVelo & FortiusANT/i-Vortex: target power ineffective

orrmany, 2023-11-15:
Hi, I try to use a T1961/T1962 brake/headunit combo ("i-Vortex") with FortiusANT. My ultimate goal is to control the trainer according to power-based structured workouts from intervals.icu. When I change target power (resistance) via the Tacx headunit, then the trainer braking effect changes noticeably. 

However, when indieVelo connects to FortiusANT via ANT FE-C, then the brake's resistance seems to be fixed whatever resistance have been set prior to the connection, regardless of the target sent by CTP (indieVelo). The target seemingly changes (see attached screenshot) 
[image], but the brake's effective resistance does not change, i.e., it remains constant. If the target power in the workout changes then I need to raise the physical speed of the wheel. (Last time I launched indieVelo, the brake's resistance was seemingly stuck around 100W@22..4km/h. The workout asked for 140W as start, then I already had to raise wheel revolution speed, as opposed to my expectations that I can keep the pedalling speed and the the brake would increase braking. Then the CTP sent a target for 220W, then I really had to physically shift gears and increase the wheel's revolution to 45...48km/h to reach 220W, which is close to the limit of the brake and made the training session extremely noisy. T1961's rated wattage capacity shall be around 950W at top revolution speeds, so 220W should have been possible at very moderate wheel-speed, say well below 25km/h spinning speed according to the Vortex device-profile, see [image]

Command line has been (Raspberry Pi 3B, latest OS, latest FortiusANT):
 ~/FortiusANT/raspberry/FortiusAntMain.sh -a -bb -t Vortex -x -H 0 

Is this behavior normal? 

- If yes, then how should I use the tool if properly for the described use case (i.e., I want constant spinning speed with braking effect varying according to the the power target from CTP)? 
- If abnormal, then pls, consider this as bug-report and advise how to debug it.

orrmany, 2023-11-17:
Now upgraded firmwares as per #219. 

- T1961 brake is now v.0.2.7
- T1962 HU is now 0.2.5
Target power is still ineffective, I will attach logs

WouterJD, 2023-11-17:
> When I change target power (resistance) via the Tacx headunit, then the trainer braking effect changes noticeably.

How did you do that?

> Then the CTP sent a target for 220W

How do you know this?
- Do you see the target in CTP being 220W
- Or do you see something on the FortiusAnt screen?

> ~/FortiusANT/raspberry/FortiusAntMain.sh -a -bb -t Vortex -x -H 0

This is OK.

-----

Is your CTP indieVelo paired with FortiusAnt?
It just *seems* that the target is not received by FortiusAnt for some reason.

To investigate, you could use the `-d f` flags, which produces a logfile with functional lines.
When the CTP sends the target, the logfile contains `"SetPower(Power)"`

--- So far assuming that the target  power is NOT correct

-----

But in the image you send (see above)
https://user-images.githubusercontent.com/16136500/283086082-a345f975-5324-490e-9eef-721821b75eb9.png

I see a black square
[image]
saying Target 140W

So indieVelo wants you to cycle with a resistance of 140W

> The workout asked for 140W as start, then I already had to raise wheel revolution speed, as opposed to my expectations that I can keep the pedalling speed and the the brake would increase braking.

This expectation is correct.

-----
As you see, the TargetPower is 140 and for Tacx-trainers, the Target-Resistance is calculated (r= 140).
The vortex Tacx Trainer is smarter than the older trainers; the required power is directly sent to the trainer.

-----

> Then the CTP sent a target for 220W, then I really had to physically shift gears and increase the wheel's revolution to 45...48km/h to reach 220W, which made the training session extremely noisy.

What puzzles me is that
- when you set the resistance manually (again: how do you do that); your trainer works well
- when you use a CTP it does not work

I can help, if you describe in more detail the two procedures, including a screen print.

Sorry for the long text, it shows the troughts I had while trhinking of your issue.

orrmany, 2023-11-17:
I made a short workout in intervals.icu:
- 1min @155W target
- 1 min @190W 
- 1min @155W
I started FortiusANT on a Raspberry3B+, with the command line
'~/FortiusANT/raspberry/FortiusAntMain.sh -a -bb -t Vortex -x -H 0 -d127 2>&1 | tee FortiusANT-lastexecution.txt  '

When the trainer used in standalone mode, then I was / am able to adjust target power from the HU and the brake reacts accodingly. I was /am able to adjust braking anywhere between 100-200W easily whe spinning the wheel  25km/h, as the power-curve suggests, too. When I change wheel speed then brake adjust braking to keep the target power, like it should.

When I start FortiusANT and the CTP connects the resistance "feels" as ca. 100W. I start the workout (indieVelo pulls the workout from intervals.icu and I start it), while spinning the wheel at moderate speed, producing ca. 60..80W. 

When the workout starts, then the printout  (the lines looking as:
"16:31:44,496: Target=100W  0.0km/h hr=  0 Current=  0W Cad=  0 r= 100 100% --b-") seems to react to the changing target power (_both_ the target watt and the target "r=xxx" changes to the target watt numbers, i.e., at 155W target it prints like:
".... Target=155W [....] r= 155 ...." 
but the brake keeps at constant braking, i.e., as hard (or soft) as was before starting the CTP. If I change the spinning speed, then the observed wattage changes accordingly and reported to the CTP.
See logs:

- [FortiusANT-lastexecution.txt](https://github.com/WouterJD/FortiusANT/files/13396593/FortiusANT-lastexecution.txt)
- [FortiusAnt.2023-11-17 16-17-35.log](https://github.com/WouterJD/FortiusANT/files/13396602/FortiusAnt.2023-11-17.16-17-35.log)
-  [FortiusAnt.2023-11-17 16-17-35.json](https://github.com/WouterJD/FortiusANT/files/13396604/FortiusAnt.2023-11-17.16-17-35.json)

orrmany, 2023-11-17:
Here is the screenshot explained 
[image]

orrmany, 2023-11-17:
> > When I change target power (resistance) via the Tacx headunit, then the trainer braking effect changes noticeably.
> 
> How did you do that?
With buttons on the Head Unit (HU, an (i-)Vortex T1962 graphical head unit). The HU can accept either HR, or slope-%, or Wattage as target. I specify the target type via presing the [OK] button then cycling-through the modes with the arrows, then choose mode with [OK] again. If in target-Wattage mode I can adjust the target wattage with the up-down buttons. 
Note, that fortiusANT must NOT operate during this time, since if it does, then the HU displays only "--PC--" and shows nothing else.
> 
> > Then the CTP sent a target for 220W
> 
> How do you know this?
The CTP (indieVelo) displays the target. As soon the CTP changes the target, say for "155W", then in the Fortius ANT console printout _both_ the value of the "...Target =...." field and the "...r = ..." field adapt. That is, these fields eventually and gradually (during 2...3 printout time) became like 
"...Target = 155W  [.....] r = 155 ..."

> 
>     * Do you see the target in CTP being 220W
Yes 

> 
>     * Or do you see something on the FortiusAnt screen?
Both.

> Is your CTP indieVelo paired with FortiusAnt? 
Yes. 

> It just _seems_ that the target is not received by FortiusAnt for some reason.
According to its console printouts it receives and adapts _both_ the "Target = " and the "r =  " fields to the same numeral value. I.e., when the TCP seems to sends XXX watt as target, then FortiusANT prints both "Target = XXX W" and " r = XXX " 

> 
> To investigate, you could use the `-d f` flags, which produces a logfile with functional lines. 
I did, with "-d 127" 

> When the CTP sends the target, the logfile contains `"SetPower(Power)"`
I have attached the logs above.

 
> --- So far assuming that the target power is NOT correct
> 
> But in the image you send (see above) https://user-images.githubusercontent.com/16136500/283086082-a345f975-5324-490e-9eef-721821b75eb9.png
> 
> I see a black square [image] saying Target 140W
I have uploaded a new version of the screenshot augmented with labels

> 
> So indieVelo wants you to cycle with a resistance of 140W
Yes

>     * when you set the resistance manually (again: how do you do that); your trainer works well
Yes. I explained the how above, 

> 
>     * when you use a CTP it does not work
Yes, the resistance seems to remain unaffected, 

> Sorry for the long text, it shows the troughts I had while trhinking of your issue.
I really appreciate your effort and the great software you have created. :) Thank you!

WouterJD, 2023-11-17:
Correct
As you see, the CTP transmits it's required power (130, 190, ...)
This is also visible in the logfile (not that `-d f` only shows a subset of what can be traced;` -d all` traces everything and - I agree - the logfile is not a beauty)
```
2023-11-17 16:18:33,245: [FortiusAnt, INFO] SetPower(155.0)
2023-11-17 16:18:34,219: [FortiusAnt, INFO] Target=155W 18.3km/h hr=  0 Current= 35W Cad= 46 r= 155 100% t-ba
```

**Conclusion 1: the communication between FortiusAnt and the CTP works.**

-----

Thanks for describing the headunit again: when modifying the power manually, FortiusAnt is not used at all.

When FortiusAnt is activem, the displpay shows --PC-- 
**Conclusion 2: there is communication between FortiusAnt and the Vortex.**

-----

**Question now is: is the target power sent to the Vortex?**

I would suggest to redo the test, with more logging; just log all then I willl find out what I need.

After the message
`2023-11-17 16:18:33,245: [FortiusAnt, INFO] SetPower(155.0)`
I would expect ANT-messages being sent to the Vortex (now not traced)
`... Dongle send ... (ch=5 p=16 ...`
where ch=5 is the Vortex channel and page 16=msgPage16_TacxVortexSetPower

WouterJD, 2023-11-17:
By the way Gábor, I read ". I recently started to use FortiusANT with a T1961 break with a T1962 head unit, which I have bought second hand. " but I have the impression you use FortiusAnt already since Nov 2020. Is that correct?

orrmany, 2023-11-17:
I will test with "-d all" tomorrow, now it is too late to fire up the trainer for pedallinh: I live in a multi-apartment house.

Yes, I briefly tested Fortius ANT a few years ago, but I never wanted to pay for Zwift and found a gym nearby with half-smart stationary bikes: they report ANT+power (and have manual, analogue braking adjustment). I used those for a while for winter  time and I went outdoor during summers. As such, I never really started to use the Tacx trainer until recently. 

Now I got potential company for indieVelo/Zwift, so I got interested.

orrmany, 2023-11-17:
We follow each other @Strava, too.  :) 
If it suits you, then I would be happy to have a voice chat via a suitable client (Messenger, Skype, Teams, whatever) to discuss the problem, sometimes an oral discussion is quicker than writing comments, but of course I would like to honor your privacy, so notes only here are totally fine.

WouterJD, 2023-11-17:
That's ok; good fun.
Let's see next week when you're able to do some testing.

It just seems as if the power does not get to the trainer.

orrmany, 2023-11-17:
FYI: while scanning the log myself, I saw some strange lines: like trying to set power to 700+ Watts (temporarily), receiving from "unknown ANT+ channel", etc.  It would be worth to record the ANT communication between the HU and the brake during the succesfull target-wattage setting communication in standalone mode. Maybe ExploreANT can do such "ANT+ eavesdropping"?

orrmany, 2023-11-17:
[image]

orrmany, 2023-11-20:
@WouterJD , I am sorry, I am/will not able to redo the test at least for 5..6 days: the bike has been re-fitted with outdoor tyres for the weekend and I will need to travel abroad this week: simply the timing prevents me to rerun the test before next week.

WouterJD, 2023-11-20:
Hi Gabor, thanks for update.
No worries I'll see you back as soon as you can.

orrmany, 2023-11-28:
Now I came back from my trip, changed back tyre to training tyre, changed debug setting to "-d all" , rebooted the the laptop running the indieVelo CTP --- and suddenly target power setting started to work normally.....
Anyway, I am uploading the logs here, then I will close the ticket. If the problem reappers, then I will reopen.
[FortiusANT_logs_2023-11-28-13.03.30.zip](https://github.com/WouterJD/FortiusANT/files/13492051/FortiusANT_logs_2023-11-28-13.03.30.zip)

orrmany, 2023-11-28:
Cannot replicate the problem, so closing

closed 2023-11-28 by orrmany

WouterJD, 2023-11-28:
Almost human. Working better after a good rest