WouterJD/FortiusANT #443: Windows 11

bredy73, 2023-11-19:
I'll try FortiusAnt executable in windows 11, but it don't work. Why?

WouterJD, 2023-11-19:
The answer as short as the question would be "because there's something wrong".

Please provide some more details.
- did you have it working on Windows 10?
- same hardware?
- on windows 11 please show error...

bredy73, 2023-11-19:
I use a new hardware, with win 11 pro. In another PC i use fortiuant with win 10 and works fine. In the newit Say me that this software Is not compatibile with win 11version. I use the executable versione.

WouterJD, 2023-11-19:
So windows 11 does not want to start it.
Try to startup in compatibility mode

There will be something in the properties of the executable

bredy73, 2023-11-20:
I just try in comatible mode, same result. None use win11 with fortiusant?

WouterJD, 2023-11-20:
Most likely a windows.exe must be created on a windows 11 system, which I do not have.

Or run the python version

bredy73, 2023-11-20:
Yes I try the executable fortiusant.exe. Then I install phyton and run FortiusAnt (use json file).bat and it works perfect.

WouterJD, 2023-11-21:
Hi @bredy73, this sounds good.

You might try to create a Windows11 executable on your system
Running `Development\FortiusANT CreateWindowsExecutable.bat` would do the job, resulting in a new `WindowsExecutable\FortiusAnt.exe`

You can then test the FortiusAnt.exe; if succesful create a PR in github to upload it, I will then promote it to the base.
If you need further guidance, just le me know.

WouterJD, 2023-11-21:
You're welcome to the FortiusANT community
Now we are communication some more; I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
You could also consider to visit [the sponsor page](https://github.com/sponsors/WouterJD)
- - -

WouterJD, 2023-11-25:
PS. I tried to reproduce on Windows 11 but no succes; it started up without issues.

Editie	Windows 11 Home
Versie	22H2
Installatiedatum	‎8-‎10-‎2022
Build van besturingssysteem	22621.2428
Ervaring	Windows Feature Experience Pack 1000.22674.1000.0

WouterJD, 2023-11-25:
Let's closed for now. 
Always happy to reopen if needed/wanted

closed 2023-11-25 by WouterJD