WouterJD/FortiusANT #445: Support I-Vortex T2160

Rien1980, 2023-11-27:
Hoi Wouter,

Om te kunnen trainen voor de Alpe d'HuZes heb ik een I-Vortex T2160 in gebruik. Ik krijg deze echter niet aan de praat met Fortius ANT. Als ik de ANT dongle aansluit wordt deze herkend. De trainer zelf wordt echter niet gevonden. 

Groet,

Rien

WouterJD, 2023-11-28:
Welkom in de FortiusANT community
- - -
Ik ben altijd nieuwsgierig met wie ik communiceer, waar FortiusANT wordt gebruikt en in welke configuratie.
Welke bundle heb je gekocht, en welke rem en head-unit gebruik je?
Vertel meer over jezelf, eventueel onder issue #14.
- - -

Het eerste advies is om FortiusAnt op te started met de `-t vortex `parameter, zodat bekend is welk type trainer gezocht moet worden.

WouterJD, 2023-11-29:
Hi Rien, welcome again. 

> My ANT-dongle is recognized as you can see below.

Did you attach something, I don't see it😬

Mind that you have to tell FortiusAnt to look for the vortex with the `-t vortex ` parameter

WouterJD, 2023-11-29:
`-t vortex ` specified?

GewoonGijs, 2023-11-30:
Hoi Rien, als je het niet werkend krijgt, mag je mijn werkende setup overnemen. Die bestaat uit een Fortius met een raspberry pi. Verbinding met je PC/laptop over Bluetooth. 

Voor €75,- mag je het geheel overnemen. Ik ben overgestapt op een direct drive trainer.

Rien1980, 2023-12-05:
> Referenc

> `-t vortex ` specified?

Yes I did. Doesn't recognize the trainer. I will try with my MacBook also.

WouterJD, 2023-12-05:
@Rien1980 wanneer je op [Locate HW] drukt, wat gebeurt er dan? Met nadruk op het veld, waar nu "No tacx trainer found" staat.

WouterJD, 2023-12-05:
Ik verwacht in ieder geval dat bericht "Tacx Vortex paired:...  (pairing can take a minute)"
langs komt.

WouterJD, 2023-12-06:
[image]

Zie voorgaande scherm
- zie je in het console de opstartparameters? 
- zie je het bericht "Pair with Tacx Vortex ... (pairing can take a minute)"

Heb je geexperimenteerd met het opslaan van de settings in een JSON file?
De settings in de JSON file overrulen de command-l;iine

Rien1980, 2023-12-07:
Hoi Wouter,

Mijn Tacx gaat aan de stroom, de stuurunit zet ik ook aan. Ik begin dan alvast te trappen. Via USB heb ik de ANT dongle aangesloten aan mijn laptop. Alles staat redelijk dicht bij de unit. Dan start ik FortiusANT vanuit de WindowsExecutable map. In de instellingen zet ik dan Vortex aan. Ook als ik op locate hw klik vindt hij de Vortex niet. 

Groet,

Rien

WouterJD, 2023-12-07:
Deel even een screen print nadat hij de trainer niet heeft gevonden, waar het FortiusAnt EN het zwarte scherm op te zien zijn aub

Rien1980, 2023-12-07:
Hoi Wouter,

Bij deze.

[image]

Groet,

Rien

WouterJD, 2023-12-07:
Als je de exe opstart en verder niets hebt gedaan, zoekt FortiusAnt naar een USB trainer.

Er wordt nu NIET naar een vortex trainer gezocht

Rien1980, 2023-12-07:
Ik heb in de settings aangegeven dat ik een Vortex gebruik.

[image]

WouterJD, 2023-12-07:
Ik ga vanavond (hoop ik) even kijken

WouterJD, 2023-12-08:
Rien

Het ziet er naar uit dat je alles goed hebt gedaan, maar dat er met de -t wijziging via het instellingen scherm iets niet werkt.

Start windows verkenner
In de folder **FortiusANT\StartUp** vindt je het opstart script **FortiusAnt (exe).bat**
Klik met de rechtemuisknop op dat bestand en klik op bewerken.
Maar van de eerste regel `..\WindowsExecutable\FortiusAnt.exe -g -t Vortex`
Klik op Bestand, Afsluiten en bewaar het bestand.

Start FortiusAnt door hierop te dubbelklikken

Je ziet dan een scherm als onderstaand:
[image]

Klik op [Locate HW] en dan zou je onderstaande moeten zien:
[image]

Klik dan op [Start], FortiusAnt moet dan naar je trainer gaan zoeken
[image]

Ik hoor graag van je en ik ga kijken wat er met de instellingen gebeurt...

Rien1980, 2023-12-08:
Beste Wouter,

Je bent een held. Hij doet het. Nu nog even klooien met een fietsapp en gaan met de banaan. Ik ga van het weekend helemaal los de Alpe 'd Huez op.

Super dat jij hiermee een bijdrage hebt geleverd aan mijn training. 

https://www.opgevenisgeenoptie.nl/fundraisers/RienVerbeek/alpe-dhuzes

Groet,

Rien

WouterJD, 2023-12-08:
Top!

Zit je ook op strava? https://www.strava.com/athletes/2885978 Zie je daar!

En als je blij bent met deze Tacx/fietsapp (Zwift?) bridge, kijk dan nog even op de [sponsorpagina](https://github.com/sponsors/WouterJD/).

Succes!

Rien1980, 2023-12-08:
Hoi Wouter,

Ik heb me op Strava net geconnect met de FortiusAnt groep en heb net in
Paypal een sponsoring gedaan. Ben je zeer dankbaar.

Groet,

Rien

WouterJD, 2023-12-08:
Ik hoop dat je d'r plezier mee hebt!
Ik ga binnekort eens kijken waarom de vortex instelling niet werkt. Maar dat voor later.
En al vast bedankt voor de sponsoring.

WouterJD, 2023-12-13:
When -t is specified interactively, the setting was not effectuated.
Improved in next version.

WouterJD, 2023-12-13:
> Paypal een sponsoring gedaan. Ben je zeer dankbaar.

Hoi Rien, geen big deal. Maar ik zie dat nog niet in Paypal.

Belangrijker is dat je trainer het doet!

WouterJD, 2023-12-15:
The user interface is improved and published including EXE in 6.8****

closed 2023-12-15 by WouterJD