WouterJD/FortiusANT #446: Continuously message "insufficient data"

WouterJD, 2023-11-30:
Hi @WouterJD, love this app, 99% of the time ... I was doing a Zwift race this morning when the app froze up (continuously read "insufficient data"), and reported 234W indefinitely. I quickly checked all the cables and restarted the app, which got me dropped, and it continued just outputting 234W. I just let it run out the race while I tried to pedal, hoping it'd reconnect and tapping stop/start and find hardware ....  I ended up rechecking all the cables several times and having to restart the app several times and also disconnect/reconnect from the Zwift connection panel.  I honestly don't know what finally got it back working.  It was just extra frustrating a few km from the finish of a race. Everything was plugged directly into my laptop usb ports.  I don't know what other tricks I can try to get a more consistent connection??

Setup: Windows 10, Fortius T1932 head unit, 2x ANT+, FortiusAnt.exe

_Originally posted by @ktippey-hzdr in https://github.com/WouterJD/FortiusANT/issues/14#issuecomment-1833334415_

ktippey-hzdr, 2023-11-30:
Hmm ok, so aside from double checking all the connections there may not be much else I can do (if it's a hardware thing). Yes, I suppose I could try adjusting error recovery but I don't know how much I could improve that as it already looks quite reasonable.

The trainer is located in my office in my apartment. It can get a little chilly when I'm not riding but not particularly cold. It had run just fine for ~ a workout a day prior to this, which is why I was surprised when it flaked this morning.  I don't know if it could related to the intensity of the workout; this race was going faster/longer than I had tried before

ktippey-hzdr, 2023-12-01:
Wow, thanks!!  I will give this a pull and try it a try this morning :)

ktippey-hzdr, 2023-12-06:
I haven't had any new instances where the head unit froze up since this implementation (I did get a complete Zwift 30s freeze-up but that was likely due to my old computer or the internet connection). As far as I can tell, I think it's a good fix

WouterJD, 2023-12-06:
Good to hear and I saw your freeze on Strava.

Did you experience a re over-action?
You should have seen a bunch of messages in the black console

WouterJD, 2023-12-07:
Great to hear. I had a successful training yesterday and had no issues, your training was saved by the recovery (next time please share the copy of the console) so the update seems OK.

I have create a pull request
I have invited you to collaborate; please accept
I can then ask you to validate the PR

This because of your interest to join in development

ktippey-hzdr, 2023-12-07:
Done! Thanks for the invite. Will screenshot or save debugging log next time

ktippey-hzdr, 2023-12-09:
Sorry to bug you so much.  I have a new problem though. My HRM is no longer recognized in Strava. FortiusAnt reads it but it doesn't broadcast right.  Every once in a while Zwift has a signal for a second then it drops again. I tried cleaning my HRM, connecting with and without -H0 option, unplugging/repluggin ANT+ sticks ... I don't know what else to try.

I don't know if it matters but this is a new miniPC with Windows 11, otherwise same ANT+ sticks, 1932 head unit

[image]

WouterJD, 2023-12-10:
Hi, I see HRM arrived in Strava today so problem seems solved.

Note that Zwift connects to the trainer, which  is FortiusAnt. There are three devuces: trainer, power meter, speed/cadence.

Zwift also connects to HRM. Directly, FortiusAnt is not involved there.

Do you use an ANT or BLE HRM?
I assume ANT.
Note that BLE usually connects to one device only. So if your watch is linked, Zwift will not see it (BLE!)

WouterJD, 2023-12-10:
I read your edit now.
👍

ktippey-hzdr, 2023-12-13:
Real issue this time: Trainer froze up at 267W. Gave it a minute to try to figure it out itself. Restarted app twice to no avail (it continued to output last known reading).  Unplugged and replugged in USB and restarted app and it reset correctly.  App initialized with `python FortiusAnt.py -g -a -dau -H11896`.  Oddly, despite USB read errors, it never logged 'Try to reconnect to Tacx head unit'.  I don't know what to make of things?

Here it was still working:
07:41:12,984: ANT Dongle:Unknown channel: synch=164, len= 9, id=0x4f, check=210, channel=0, page=51(0x33) info="00 33 ff ff ff ff b3 4f ff"
07:41:12,985: devAntDongle.__ReadAndRetry() returns ["a4", "03", "40", "03", "01", "03", "e6"] 
07:41:12,985: Dongle    receive: synch=0xa4, len= 3, id=0x40 Channel Response     , check=0xe6, info="03 01 03" (ch=3, msg=0x1)
07:41:13,059: devAntDongle.__ReadAndRetry() returns ["a4", "03", "40", "00", "01", "03", "e5"] 
07:41:13,059: Dongle    receive: synch=0xa4, len= 3, id=0x40 Channel Response     , check=0xe5, info="00 01 03" (ch=0, msg=0x1)
07:41:13,070: devAntDongle.__ReadAndRetry() returns ["a4", "03", "40", "06", "01", "03", "e3"] 
07:41:13,070: Dongle    receive: synch=0xa4, len= 3, id=0x40 Channel Response     , check=0xe3, info="06 01 03" (ch=6, msg=0x1)
07:41:13,212: devAntDongle.__ReadAndRetry() returns ["a4", "03", "40", "02", "01", "03", "e7"] 
07:41:13,212: Dongle    receive: synch=0xa4, len= 3, id=0x40 Channel Response     , check=0xe7, info="02 01 03" (ch=2, msg=0x1)
07:41:13,222: devAntDongle.__ReadAndRetry() returns ["a4", "09", "4e", "01", "04", "00", "ad", "09", "25", "0b", "a7", "a2", "69"] 
07:41:13,222: Dongle    receive: synch=0xa4, len= 9, id=0x4e Broadcast Data       , check=0x69, info="01 04 00 ad 09 25 0b a7 a2" (ch=1 p=4(HRM Previous Heart beat))
07:41:13,232: devAntDongle.__ReadAndRetry() returns ["a4", "03", "40", "03", "01", "03", "e6"] 
07:41:13,232: Dongle    receive: synch=0xa4, len= 3, id=0x40 Channel Response     , check=0xe6, info="03 01 03" (ch=3, msg=0x1)
07:41:13,234: Sleep(0.25) to fill 0.25 seconds done.

Here it started outputting constant watt-stream:
**07:41:13,277: Trainer recv hdr=-0x1 data="array('B')" (len=0)**
**07:41:13,277: Retry because short buffer (len=0) or incorrect header received (expected: 0x21303 received: -0x1)**
07:41:13,308: devAntDongle.__ReadAndRetry() returns ["a4", "03", "40", "00", "01", "03", "e5"] 
07:41:13,309: Dongle    receive: synch=0xa4, len= 3, id=0x40 Channel Response     , check=0xe5, info="00 01 03" (ch=0, msg=0x1)
07:41:13,321: devAntDongle.__ReadAndRetry() returns ["a4", "03", "40", "06", "01", "03", "e3"] 
07:41:13,321: Dongle    receive: synch=0xa4, len= 3, id=0x40 Channel Response     , check=0xe3, info="06 01 03" (ch=6, msg=0x1)
07:41:13,418: Trainer recv hdr=-0x1 data="array('B')" (len=0)
**07:41:13,418: Retry because short buffer (len=0) or incorrect header received (expected: 0x21303 received: -0x1)**
07:41:13,461: devAntDongle.__ReadAndRetry() returns ["a4", "03", "40", "02", "01", "03", "e7"] 
07:41:13,461: Dongle    receive: synch=0xa4, len= 3, id=0x40 Channel Response     , check=0xe7, info="02 01 03" (ch=2, msg=0x1)
07:41:13,467: devAntDongle.__ReadAndRetry() returns ["a4", "03", "40", "01", "01", "02", "e5"] 
07:41:13,467: Dongle    receive: synch=0xa4, len= 3, id=0x40 Channel Response     , check=0xe5, info="01 01 02" (ch=1, msg=0x1)
07:41:13,478: devAntDongle.__ReadAndRetry() returns ["a4", "03", "40", "03", "01", "03", "e6"] 
07:41:13,478: Dongle    receive: synch=0xa4, len= 3, id=0x40 Channel Response     , check=0xe6, info="03 01 03" (ch=3, msg=0x1)
07:41:13,559: devAntDongle.__ReadAndRetry() returns ["a4", "03", "40", "00", "01", "03", "e5"] 
07:41:13,559: Dongle    receive: synch=0xa4, len= 3, id=0x40 Channel Response     , check=0xe5, info="00 01 03" (ch=0, msg=0x1)
07:41:13,561: Trainer recv hdr=-0x1 data="array('B')" (len=0)
07:41:13,561: Retry because short buffer (len=0) or incorrect header received (expected: 0x21303 received: -0x1)
07:41:13,570: devAntDongle.__ReadAndRetry() returns ["a4", "03", "40", "06", "01", "03", "e3"] 
07:41:13,570: Dongle    receive: synch=0xa4, len= 3, id=0x40 Channel Response     , check=0xe3, info="06 01 03" (ch=6, msg=0x1)
07:41:13,702: Trainer recv hdr=-0x1 data="array('B')" (len=0)
07:41:13,702: Retry because short buffer (len=0) or incorrect header received (expected: 0x21303 received: -0x1)
07:41:13,711: devAntDongle.__ReadAndRetry() returns ["a4", "03", "40", "02", "01", "03", "e7"] 
07:41:13,711: Dongle    receive: synch=0xa4, len= 3, id=0x40 Channel Response     , check=0xe7, info="02 01 03" (ch=2, msg=0x1)
07:41:13,711: devAntDongle.__ReadAndRetry() returns ["a4", "03", "40", "01", "01", "09", "ee"] 
07:41:13,711: Dongle    receive: synch=0xa4, len= 3, id=0x40 Channel Response     , check=0xee, info="01 01 09" (ch=1, msg=0x1)
07:41:13,725: devAntDongle.__ReadAndRetry() returns ["a4", "03", "40", "03", "01", "03", "e6"] 
07:41:13,726: Dongle    receive: synch=0xa4, len= 3, id=0x40 Channel Response     , check=0xe6, info="03 01 03" (ch=3, msg=0x1)
07:41:13,808: devAntDongle.__ReadAndRetry() returns ["a4", "03", "40", "00", "01", "03", "e5"] 
07:41:13,808: Dongle    receive: synch=0xa4, len= 3, id=0x40 Channel Response     , check=0xe5, info="00 01 03" (ch=0, msg=0x1)
07:41:13,814: devAntDongle.__ReadAndRetry() returns ["a4", "09", "4f", "00", "33", "ff", "ff", "ff", "ff", "90", "4f", "ff", "f1"] 
[... cut ...] , info="00 33 ff ff ff ff 90 4f ff" (ch=0 p=51(Track Resistance))
07:41:13,820: devAntDongle.__ReadAndRetry() returns ["a4", "03", "40", "06", "01", "03", "e3"] 
07:41:13,821: Dongle    receive: synch=0xa4, len= 3, id=0x40 Channel Response     , check=0xe3, info="06 01 03" (ch=6, msg=0x1)
**07:41:13,841: Trainer recv hdr=-0x1 data="array('B')" (len=0)
07:41:13,841: Tacx head unit returns insufficient data, len=0
07:41:13,841: To resolve, check all (signal AND power) cabling for loose contacts.**

WouterJD, 2023-12-15:
> I was trying to build the EXE

Building EXE on Windows 10 still works, perhaps some checks required for Windows 11.

WouterJD, 2023-12-15:
The USB error recovery is implemented and published including EXE in 6.8

closed 2023-12-15 by WouterJD