WouterJD/FortiusANT #448: bleBless error: control requested by client, control already granted

rcambrj, 2023-12-01:
Hello and thanks for this! :)

I keep getting a problem whereby after some time FortiusANT stops working. The amount of time seems to vary, but the error is always the same:

<details>
<summary>show log</summary>
```
$ journalctl -f
[...]
Dec 01 16:19:19 tacx python3[394]: 16:19:19,888: Target=0.9%(108W) 20.5km/h Current=109W Cad= 39 r=2338 100% t-c-
Dec 01 16:19:20 tacx python3[394]: 16:19:20,893: Target=0.9%(108W) 20.5km/h Current= 92W Cad= 39 r=2346 100% ---b
Dec 01 16:19:21 tacx python3[394]: 16:19:21,898: Target=0.9%(106W) 20.3km/h Current= 85W Cad= 39 r=2323 100% t---
Dec 01 16:19:22 tacx python3[394]: 16:19:22,901: Target=0.9%(105W) 20.2km/h Current=102W Cad= 39 r=2307 100% --cb
Dec 01 16:19:23 tacx python3[394]: 16:19:23,906: Target=0.9%(104W) 20.1km/h Current=100W Cad= 38 r=2303 100% t---
Dec 01 16:19:24 tacx python3[394]: 16:19:24,913: Target=0.9%(103W) 20.0km/h Current= 86W Cad= 38 r=2285 100% --cb
Dec 01 16:19:25 tacx python3[394]: 16:19:25,917: Target=0.9%(104W) 20.1km/h Current=103W Cad= 38 r=2303 100% t---
Dec 01 16:19:26 tacx python3[394]: 16:19:26,922: Target=0.9%(102W) 19.9km/h Current=102W Cad= 38 r=2279 100% ---b
Dec 01 16:19:27 tacx python3[394]: 16:19:27,927: Target=0.9%(102W) 19.9km/h Current= 88W Cad= 38 r=2278 100% t-c-
Dec 01 16:19:28 tacx python3[394]: 16:19:28,931: Target=0.9%(101W) 19.7km/h Current= 86W Cad= 38 r=2274 100% ---b
Dec 01 16:19:29 tacx python3[394]: 16:19:29,936: Target=0.9%(100W) 19.6km/h Current= 90W Cad= 38 r=2264 100% t---
Dec 01 16:19:30 tacx python3[394]: 16:19:30,942: Target=0.9%(98W) 19.4km/h Current= 84W Cad= 37 r=2241 100% --cb
Dec 01 16:19:31 tacx python3[394]: 16:19:31,945: Target=0.9%(98W) 19.4km/h Current= 93W Cad= 37 r=2242 100% t---
Dec 01 16:19:32 tacx python3[394]: 16:19:32,951: Target=0.9%(98W) 19.4km/h Current= 86W Cad= 37 r=2241 100% ---b
Dec 01 16:19:33 tacx python3[394]: 16:19:33,956: Target=0.9%(101W) 19.7km/h Current=126W Cad= 37 r=2276 100% t---
Dec 01 16:19:34 tacx python3[394]: 16:19:34,477: bleBless error: control requested by client, control already granted
Dec 01 16:19:34 tacx python3[394]: 16:19:34,961: Target=0.9%(101W) 19.7km/h Current=105W Cad= 37 r=2281 100% ---b
Dec 01 16:19:35 tacx python3[394]: 16:19:35,965: Target=0.9%(101W) 19.7km/h Current= 94W Cad= 37 r=2278 100% t-c-
Dec 01 16:19:36 tacx python3[394]: 16:19:36,971: Target=0.9%(99W) 19.5km/h Current= 83W Cad= 37 r=2256 100% ---b
Dec 01 16:19:37 tacx python3[394]: 16:19:37,975: Target=0.9%(98W) 19.4km/h Current= 81W Cad= 37 r=2249 100% t---
Dec 01 16:19:38 tacx python3[394]: 16:19:38,980: Target=0.9%(95W) 19.1km/h Current= 83W Cad= 37 r=2212 100% --cb
Dec 01 16:19:39 tacx python3[394]: 16:19:39,502: bleBless error: control requested by client, control already granted
Dec 01 16:19:39 tacx python3[394]: 16:19:39,986: Target=0.9%(94W) 19.0km/h Current= 75W Cad= 36 r=2204 100% t---
Dec 01 16:19:40 tacx python3[394]: 16:19:40,990: Target=0.9%(93W) 18.9km/h Current= 79W Cad= 36 r=2186 100% ---b
Dec 01 16:19:41 tacx python3[394]: 16:19:41,994: Target=0.9%(92W) 18.7km/h Current= 83W Cad= 36 r=2184 100% t---
Dec 01 16:19:43 tacx python3[394]: 16:19:43,000: Target=0.9%(91W) 18.6km/h Current= 69W Cad= 36 r=2177 100% ---b
Dec 01 16:19:44 tacx python3[394]: 16:19:44,004: Target=0.9%(87W) 18.2km/h Current= 67W Cad= 35 r=2126 100% t-c-
Dec 01 16:19:44 tacx python3[394]: 16:19:44,527: bleBless error: control requested by client, control already granted
Dec 01 16:19:45 tacx python3[394]: 16:19:45,009: Target=0.9%(86W) 18.0km/h Current= 63W Cad= 34 r=2125 100% ---b
Dec 01 16:19:46 tacx python3[394]: 16:19:46,014: Target=0.9%(83W) 17.7km/h Current= 57W Cad= 34 r=2084 100% t-c-
Dec 01 16:19:47 tacx python3[394]: 16:19:47,018: Target=0.9%(83W) 17.7km/h Current= 75W Cad= 34 r=2080 100% ---b
Dec 01 16:19:48 tacx python3[394]: 16:19:48,024: Target=0.9%(84W) 17.8km/h Current= 82W Cad= 34 r=2101 100% t---
```
</details>

From the moment where the log says `bleBless error: control requested by client, control already granted`, Zwift begins to report a bluetooth connection failure. FortiusANT seems to continue to work fine reporting the current power in the log, but in order to resume a connection with Zwift I have to reboot the machine.

I'm starting FortiusANT commit 8ee2a5a8a8020b6445d4c0c6e264bc0de00a99f7 with:

```
/usr/bin/python3 /home/pi/FortiusANT/pythoncode/FortiusAnt.py -a -bb -D-1 -H-1
```

on a Raspberry Pi 3B running Raspberry Pi OS 32 bit (minimal, no graphical environment).

I ran with `-d65535` and the debug output around the first error is:

<details>
<summary>show log</summary>
```
2023-12-01 16:19:33,696: [FortiusAnt, INFO] Sleep(0.23) to fill 0.25 seconds done.
2023-12-01 16:19:33,697: [FortiusAnt, INFO] clsTacxTrainer.Refresh(True, 2)
2023-12-01 16:19:33,698: [FortiusAnt, INFO] clsTacxNewUsbTrainer._ReceiveFromTrainer()
2023-12-01 16:19:33,699: [FortiusAnt, INFO] Trainer recv hdr=0x21303 data="array('B', [18, 204, 18, 0, 0, 2, 0, 0, 9, 0, 0, 0, 37, 0, 0, 0, 0, 0, 13, 10, 0, 0, 0, 0, 3, 19, 2, 0, 5, 220, 1, 0, 81, 22, 72, 97, 48, 8, 194, 9, 200, 8, 1, 0, 37, 8, 2, 221, 163, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0])" (len=64)
2023-12-01 16:19:33,700: [FortiusAnt, INFO] ReceiveFromTrainer() = hr=37 Buttons=0 Cadence=37 Speed=19.7 TargetRes=2248 CurrentRes=2498 CurrentPower=110, pe=1 hdr=0x21303 Connected to Tacx Trainer T1932
2023-12-01 16:19:33,701: [FortiusAnt, INFO] Grade2Power (TargetGrade= 0.9%, Speed=19.7, Weight= 85, rR=0.004, wR=0.51, wS=0, d=1.0) = TargetPower=101W
2023-12-01 16:19:33,702: [FortiusAnt, INFO] clsTacxNewUsbTrainer.TargetPower2Resistance(101, 5713) = 2278
2023-12-01 16:19:33,702: [FortiusAnt, INFO] clsTacxUsbTrainer.SendToTrainer(T=2, M=2, P=101, G=0.9, R=2278, W=85, PE=1, S=19.7, C=37)
2023-12-01 16:19:33,703: [FortiusAnt, INFO] Trainer send data="01 08 01 00 e6 08 01 00 02 55 10 04" (len=12)
2023-12-01 16:19:33,703: [FortiusAnt, INFO]                   tacx mode=2 target=2278 pe=1 weight=85 cal=1040
2023-12-01 16:19:33,706: [FortiusAnt, INFO] clsFTMS_bless.SetAthleteData(0)
[... 77 lines cut ...]
2023-12-01 16:19:35,204: [FortiusAnt, INFO] Sleep(0.23) to fill 0.25 seconds done.
2023-12-01 16:19:35,205: [FortiusAnt, INFO] clsTacxTrainer.Refresh(True, 2)
2023-12-01 16:19:35,205: [FortiusAnt, INFO] clsTacxNewUsbTrainer._ReceiveFromTrainer()
2023-12-01 16:19:35,207: [FortiusAnt, INFO] Trainer recv hdr=0x21303 data="array('B', [18, 204, 18, 0, 0, 2, 0, 0, 9, 0, 0, 0, 37, 0, 0, 0, 0, 0, 13, 10, 0, 0, 0, 0, 3, 19, 2, 0, 137, 220, 1, 0, 88, 22, 80, 94, 104, 8, 58, 8, 229, 8, 1, 0, 37, 6, 2, 132, 176, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0])" (len=64)
2023-12-01 16:19:35,208: [FortiusAnt, INFO] ReceiveFromTrainer() = hr=37 Buttons=0 Cadence=37 Speed=19.7 TargetRes=2277 CurrentRes=2106 CurrentPower=93, pe=1 hdr=0x21303 Connected to Tacx Trainer T1932
2023-12-01 16:19:35,208: [FortiusAnt, INFO] Grade2Power (TargetGrade= 0.9%, Speed=19.7, Weight= 85, rR=0.004, wR=0.51, wS=0, d=1.0) = TargetPower=101W
2023-12-01 16:19:35,209: [FortiusAnt, INFO] clsTacxNewUsbTrainer.TargetPower2Resistance(101, 5720) = 2275
2023-12-01 16:19:35,209: [FortiusAnt, INFO] clsTacxUsbTrainer.SendToTrainer(T=2, M=2, P=101, G=0.9, R=2275, W=85, PE=1, S=19.7, C=37)
2023-12-01 16:19:35,210: [FortiusAnt, INFO] Trainer send data="01 08 01 00 e3 08 01 00 02 55 10 04" (len=12)
2023-12-01 16:19:35,211: [FortiusAnt, INFO]                   tacx mode=2 target=2275 pe=1 weight=85 cal=1040
2023-12-01 16:19:35,213: [FortiusAnt, INFO] clsFTMS_bless.SetAthleteData(0)
2023-12-01 16:19:35,217: [FortiusAnt, INFO] clsFTMS_bless.SetTrainerData(19.7, 37, 93)
2023-12-01 16:19:35,222: [FortiusAnt, INFO] SetGrade(0.9)
2023-12-01 16:19:35,222: [FortiusAnt, INFO] SetRollingResistance(0.004[def=0.004])
2023-12-01 16:19:35,455: [FortiusAnt, INFO] Sleep(0.23) to fill 0.25 seconds done.
2023-12-01 16:19:35,456: [FortiusAnt, INFO] clsTacxTrainer.Refresh(True, 2)
2023-12-01 16:19:35,456: [FortiusAnt, INFO] clsTacxNewUsbTrainer._ReceiveFromTrainer()
2023-12-01 16:19:35,458: [FortiusAnt, INFO] Trainer recv hdr=0x21303 data="array('B', [18, 204, 18, 0, 0, 2, 0, 0, 9, 0, 0, 0, 37, 0, 0, 0, 0, 0, 13, 10, 0, 0, 0, 0, 3, 19, 2, 0, 159, 220, 1, 0, 81, 22, 128, 109, 104, 8, 64, 9, 233, 8, 0, 0, 37, 6, 2, 152, 253, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0])" (len=64)
2023-12-01 16:19:35,458: [FortiusAnt, INFO] ReceiveFromTrainer() = hr=37 Buttons=0 Cadence=37 Speed=19.7 TargetRes=2281 CurrentRes=2368 CurrentPower=104, pe=0 hdr=0x21303 Connected to Tacx Trainer T1932
```
</details>

Googling the error doesn't seem to reveal much 🤔 any idea where I should be looking?

My guess is that this is likely a problem with [kevincar/bless](https://github.com/kevincar/bless). FWIW I'm running 0.2.5 (currently latest).

rcambrj, 2023-12-01:
I fixed the problem by switching to the nodejs flavoured bluetooth stack. Seems to be stable now. If you can think of things which I should test, I'm happynto switch back and try them.

WouterJD, 2023-12-02:
Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
You could also consider to visit [the sponsor page](https://github.com/sponsors/WouterJD)
- - -

I will look into this issue shortly, you though that it's a kevincar/bless issue may be correct.
Good that you managed to getr nodejs to work,.

WouterJD, 2023-12-15:
Since there is no communication here, I assume can be closed.
If not happy to reopen

closed 2023-12-15 by WouterJD

rcambrj, 2023-12-18:
Were you waiting on me for further information?

WouterJD, 2023-12-29:
Hi @rcambrj 

i was curious about your set-up, what tacx you use etc.
Regarding the issue; since you siolved it I did not take action.

As documented, bleBless is the nicer architecture, but now and then lacks stability...