WouterJD/FortiusANT #450: Adding Cadence Sensor

decodeais, 2023-12-16:
I would like to add the Cadence-Sensor into the Fortius-Ant Software.  I think it would be nice without the annoying magnet sensor mounting and adjusting. 
I would like  to add an additional input like the Input for the heartrate-Sensor. With this information I can calculate a pulse frequency to  send it via optocoupler to the Fortius candence sensor plug.
Do you see special problems?
I n which program section should I place the pulse Timer?
What  do you think about it?

WouterJD, 2023-12-16:
Unfortunately that is not possible ; the headunit of the Fortius (and likes) use the cadense sensor ITSELF for proper functioning. FortiusAnt cannot change that.

There is no way to send cadense sensor info TO the headunit through USB

WouterJD, 2023-12-16:
Creating an external signal to input to the cadense sensor plug is far more complex than the current simple cadense sensor.

I also guess you might be the only one using the implementation

WouterJD, 2023-12-16:
Open for comments

decodeais, 2023-12-16:
if I read the Cadence from ta BLE-Sensor, make a correct pulse frequency an a Raspi there would be no difference for the Fortius trainer.  The needed hardware would be one resistor and one optocoupler ( 20Cent). The only problem would be the Round display for the Pedal torque. I would get a random phase shift in the simulated sensor signals.

WouterJD, 2023-12-16:
I understand getting BLE stable is more a challenge than getting the cadense sensor work.
It's a very complex way to replace the magnet driven switch, I think.

WouterJD, 2023-12-16:
"it would be nice without the annoying magnet sensor mounting and adjusting"

Is that the problem we're trying to solve?

decodeais, 2023-12-22:
May be this feature is too special. I implemented  this part of software in a  "Nano BLE33" . I will test it this week. I am curious  how stable it works. So you can close this request if you like.

WouterJD, 2023-12-22:
Agree, but always curious to follow results.
Please post here or at the pictures page

decodeais, 2023-12-24:
IT works!
https://www.strava.com/activities/10428981150
The Low Cadence ist quite normal for me. Thats reason i bought this Part to have a Look in it.

WouterJD, 2023-12-26:
Smart solution, independant of FortiusAnt and doing exactly what's needed.

Regarding the code, I prefer using the struct functions to fill buffers. But if it works, great job👍

decodeais, 2023-12-28:
The device makes only sense if you use your normal road bike on the trainer. This solution has the advantatage that it also works together with Fortius and running on a PC . 
At the end  the very  simple schematic  :
[image]

closed 2024-01-19 by WouterJD