WouterJD/FortiusANT #451: Running bluetooth on the same device

firux88, 2024-01-08:
Hi, I'm trying to run my device (T1932) with FortiusAnt on the same computer where I run Zwift.

I'm using NodeJs (-b) and bluetooth stream correctly on my phone showing FortiusAnt Trainer.
I'm using a TPLINK U500 Bluetooth 5 with WinUSB (6.1.7600.16385) and a U400 Bluetooth 4.
I'm unable to connect my U400 to the U500 FortiusAnt from the U400.

Any idea how to make it work ? Windows 10, WinUSB 

Thank you

WouterJD, 2024-01-08:
Hi Jimmy, Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
Thanks for sponsoring!
- - -

Now back to your questions

You use a T1932 headunit, connect with USB to your Windows PC.
Using NodeJs, the data comes on your windows phone; I would be interested to know what software you use there (Question1).
So far so good.

Then I do not understand the text; I understand you have two bluetooth-dongles in the windows PC.
I guess you want a CTP (dongle 1) to communicate to FortiusAnt (dongle 2).
_Note that, two ANT-dongles in one PC work well for the same purpose._

I do not know whether this has been done before.
So, some trial and error may be applicable.

NOTE: A BLE-device can connect to ONE other device only; for example if you have a HRM (Heart rate monitor) you can read the heartrate with one watch only!

NOTE2: There are HRM's that support two monitors, but they have implemented two BLE-devices to achieve that purpose.

I guess that, as long as your phone is connected to FortiusAnt, no other device will see FortiusAnt BLE devices anymore.
So when doing the trials, please remove the phone connection.

I hope this helps you further, I will read the results.

firux88, 2024-01-08:
Thank you Wrouter for the quick answer.
I've successfully make it work by removing the bluetooth 5 dongle and I'm using U400 tplink for FortiusAnt and the inboard for the CTP.
The issue is the U400 was used by bleno as the bt5 is probably not supported by bleno AND it was used by windows. (How maybe I mixed some driver ID) but it's working now.

I've added my informations in #14 
Cheers !

WouterJD, 2024-01-09:
Great problem solved.

closed 2024-01-09 by WouterJD