WouterJD/FortiusANT #452: Tacx (Garmin) Training App on IOS does not see the Trainer

TomX6601, 2024-01-14:
Hi Wouter,
first of all, excellent app and thank you for it! I have installation on Rpi 3B+ with my old i-Magic. Through BLE my Zwift app on IOS see it well and it works. Unfortunatelly Tacx(Garmin) Training App on the same IPhone does not see it. What could be wrong?

WouterJD, 2024-01-14:
Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used;  configuration used is known - Magic. 
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
You could also consider to visit [the sponsor page](https://github.com/sponsors/WouterJD)
- - -

Whats wrong with the tacx app. All polite remarks apart: tacx does not want to support the so-called legacy trainers and has builtin proprietary measures to block. I may be misinformed, but it's the only app not working.

TomX6601, 2024-01-15:
Thanx

TomX6601, 2024-01-15:
Wouter, I am little bit confused with command line parameters. I have T1902 i-Magic connected to Rpi. On PC I run Zwift, where the FortiusAnt Trainer is connected through BLE.I have separate BLE HRM.
1. Should I connect HRM to FortiusANT through BLE, or directly to Zwift? What parameters?
2. What are the best commandline parameters for Zwift usage? I use workout plan sent from Trainigpeaks to Zwift
3. Rpi is asking before every usage to pair device through BLE and I have to confirm it pushing the button on a keyboard, but I do not have keyboard connected every time to Rpi.How to avoid it?

WouterJD, 2024-01-16:
1. Connect HRM to Zwift (FortiusAnt does not use an HRM, old tacx trainers connected to HRM but that's not applicable for you)
2. No specific command line parameters are required, so focus on getting the setup working first
3. Please provide more info (screen shot, foto) since I do not recognize this question

TomX6601, 2024-01-19:
Works already well with Zwift, because I am using the pairing through Iphone. I have the problem with Rouvy, because FortiusAnt is paired somehow strange to my W11 notebook. When I start Rouvy, it sees Fortius, but do not transfer Power. Should I use also the external BLE dongle for pairing Rpi to my notebook?

closed 2024-02-10 by TomX6601