WouterJD/FortiusANT #453: Free Fortius T1940 - Los Angeles area

RodeTheAlpe, 2024-01-18:
Hey Wouter, I didn't see another way to contact you before posting this but I believe GitHub did away with that function.

I have a Fortius T1940 along with the steering frame and some software and would love to find someone (hopefully local to Los Angeles) who would like to have it for backup, parts or to use along with your software.  I've been using FortiusANT for the past 3+ years but I was recently gifted a new smart trainer and would hate to just drop this in the trash.

Feel free to email me red5five@yahoo.com if interested and if you feel this is not the place to post this please feel free to remove this post.

Thanks and thanks for all of your hard work that gave my Fortius a second life!
Jeff

WouterJD, 2024-01-18:
Hi Jeff

Good to hear you enjoyed FortiusAnt for all these years; if I recall well since early 2020!
I'm perfectly fine that trainers are exchanged through an issue and will not delete it!

Since you apparently Rode the Alps, do you have a link with Italy as well?

RodeTheAlpe, 2024-01-18:
Thanks for keeping it online.  If I don't get any hits in the next month or two I remove the link.
As far as Italy, I did a short tour years ago in northern Italy.  We rode from Trento around Largo di Garda, through Verona, Vicenza, and finished up on Lido across from Venice.  It was early October that year and the weather was perfect (and I enjoyed the food much more than in the French Alpes).  I'd love to get back to Europe to ride some more but I've got to get the kid off to college first, so maybe in a couple of years.  Let me know if you make it to California.

Thanks again, take care,Jeff

WouterJD, 2024-01-18:
Ha, great area! If you like, enjoy my video's: www.youtube.com/SaurisMTB.
Sauris is a small village in the Dolomites and from there I make most of my tours.
it's located north of Venice and was supplier of wood in the old days.

Visiting LA is as interesting as unlikely that it will happen shortly, but who knows and I'l give a ping if it might happen.
Cheers!

RodeTheAlpe, 2024-02-29:
Thanks Wouter for letting me post here.  Guess it's going to the recycling center (is cycling recycling a thing?!?).
Feel free to remove the post.  Take care!

WouterJD, 2024-11-29:
I assume it's found it's way :-)

closed 2024-11-29 by WouterJD

RodeTheAlpe, 2024-11-29:
Probably rusting away in a land fill unfortunately. Thanks for giving it a second life with Rouvy!Happy Holidays Jeff

WouterJD, 2024-12-01:
Sorry for this unfortunate 3rd life.
Hope you enjoy your new trainer!