WouterJD/FortiusANT #455: Rouvy issue

TomX6601, 2024-01-20:
from some reason the connection of FortiusANT on Rpi 3B+ to Rouvy on W11 notebook drops after few minutes of the workout.

TomX6601, 2024-01-20:
It seems, that onboard BLE chipset and as well as recommended BLE domgle Asus BT-400 has the same problem to connect reliably FortiusANT. Currently the only working setup is ZWIFT paired through IPhone. Pairing directly through BT does not work properly, although I can pair FortiusANT in Windows somehow.

TomX6601, 2024-01-21:
It finds the FortiusAnt, sends 3 BT requests for pairing to Rpi and then says "Device is not possible to connect as a smart trainer"

WouterJD, 2024-01-23:
It all seems a chipset / driver issue, which I cannot reproduce, not having your configuration.
The FortiusAnt logfiles will (most likely) noty be of any help, because the problem is likely at the lower levels.

WouterJD, 2024-01-23:
Same as #452

closed 2024-01-23 by WouterJD

TomX6601, 2024-02-10:
It suddenly works. No Idea how it happened, but it looks like some Hell in Windows11 :)

WouterJD, 2024-02-11:
Fortunately, now you can train👍