WouterJD/FortiusANT #458: Where I can find de Exe o Install on windows.

kuryaky, 2024-02-13:
Hello From Colombia, I have my e genius 12 years ago, I still rocking but always want to use it wit rouvy.
now I find you and o lord this its great, I already bought a dongle stick for the second ant+.
please let me know where I can find install and that. 
thank you and god bless you.

WouterJD, 2024-02-13:
Welcome to the FortiusANT community
- - -
So COlombia and Genius I know...

I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
You could also consider to visit [the sponsor page](https://github.com/sponsors/WouterJD)
- - -

A goiod place to start is https://github.com/WouterJD/FortiusANT/wiki with a link to the user-manual.

You have the choice to use the Python version or the **FortiusANT.exe** executable which can be found [here](https://github.com/WouterJD/FortiusANT/tree/master/WindowsExecutable/FortiusANT.exe)

The common way is to replicate the github to your local computer and then use whatever you choose. Read the manual for more info; read section **5.14 Detailed notes on Tacx Genius**.

Well, like to hear from you; cheerio.

WouterJD, 2024-02-17:
Hi Columbian @kuryaky how are you, any progress?

kuryaky, 2024-02-17:
Have a good one, great to know about you, I waiting my second ANT+ dongle.
That coming on Wednesday.
I will die hard needy until it make it works, May’s be some help and advice will be needed.
Thank you for your measenge.
Glad to hear about you.

Today real ride.

[image0.jpeg]

closed 2024-11-29 by WouterJD