WouterJD/FortiusANT #461: BLE error on OSX (MacBook Air)

pietercjanssen, 2024-03-03:
Dear Wouter and other FortiusANT users, 

Is there anyone who could help me with the following problem:

I used FortiusANT for a few years with two ANT+-dongles, but now after an update my CTP needs to use BLE.
My MacBook (model 2020-21 with M1 chip) is equipped with an internal BLE facility, so I thought to use that to send a BLE signal to the CTP on my Ipad or smartphone.
 
And although I succeed to run FortiusANT with the message in the GUI "Bluetooth interface open", I still get an error.
This is with bleno (-b). Essentially, I get the following error message in my Log-file:

`2024-03-03 21:20:23,639: [urllib3.connectionpool, DEBUG] Starting new HTTP connection (1): localhost:9999
2024-03-03 21:20:23,640: [FortiusAnt, INFO] ... requests.post() error HTTPConnectionPool(host='localhost', port=9999): Max retries exceeded with url: /ant (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x11da26b80>: Failed to establish a new connection: [Errno 61] Connection refused'))`

The full Logfile is attached
[FortiusAnt.2024-03-03 21-20-20.log](https://github.com/WouterJD/FortiusANT/files/14474487/FortiusAnt.2024-03-03.21-20-20.log)

I tried several reinstalls etc. but not successful so far.

When using bless (-bb) I even do not get a BLE connection at all. See the following logfile:
[FortiusAnt.2024-03-03 21-21-28.log](https://github.com/WouterJD/FortiusANT/files/14474496/FortiusAnt.2024-03-03.21-21-28.log)

I hope someone can help me out.
With kind regards

Pieter Janssen

WouterJD, 2024-03-04:
Just for my curiosity; what CTP does not support ANT anymore?

pietercjanssen, 2024-03-05:
That is Tacx (Garmin)
Already since last year...