WouterJD/FortiusANT #462: Meaning of command line values

TomX6601, 2024-03-08:
Hi,
what mean values shown in command line window (rpi)?
Power, Cadence, speed is clear, but what means r parameter? there is r=27 and when I move it changes to r=203. 
next is 100%. What it means? If I use parameter -p90, there is stil 100%? I am using i-Magic with Rouvy (-G50 as reccommended), but my Favero pedal meter always shows power higher for 10%, so I use -p90 parameter to synchronize it, but I do not know if the FortiusANT accepted it

WouterJD, 2024-03-10:
Ha, power meter!

I'll answer this q ASAP. OK?

WouterJD, 2024-03-10:
Target, Speed, Current Power, Cadence will be obvious
r=resistance, the resistance of the motor drive, which results in a target power at a given wheel speed.
The percentage is the virtual gearbox, which initially is 100%
The last parameter are the leds.

-p: Adjust target Power by multiplying by this factor for static calibration.
If the power, displayed by FortiusAnt is always 10% higher than your power-meter, specify -p90 to reduce resistance by 10%.

-c: Calibrate the rolling resistance for magnetic brake.

WouterJD, 2024-03-10:
To know the accepted values run FortiusAnt with -d1 to activate ddebugging and the parameters are displayed

TomX6601, 2024-03-10:
thnx

closed 2024-03-10 by WouterJD