WouterJD/FortiusANT #463: Power curve validation

TomX6601, 2024-03-09:
I have i-Magic 1902 + Rouvy. The trainer breaks more, then it is required. Is it enough to calibrate it by decreasing parameter -c or should I implement some function described in 6.3 section of Manual?

WouterJD, 2024-03-10:
Hi! Good that you look into this.

How do you know the power isincorrect; feeling or power meter? The latter is a good help.
Write down the differences. Is it a somewhat fixed %, then - c is a solution.

Please note that these legacy trainers should be compared to the modern tacxes. Yes we can connect to modern software, the trainer is "old design and capabilities".

Succes and let me know

TomX6601, 2024-03-10:
My setup is following:
i-Magic 1902 + FortiusAnt + Rouvy. I have Rouvy connected to TrainingPeaks where I choose the workout. In Rouvy I just connect any video to this workout. On my bike I use Favero power meter connected to Garmin Edge, independently from Rouvy. There was some static difference between Favero and FortiusAnt, so I solve it by -p parameter.It shows correct Power, but workout breaks the trainer too much. I have an idea to calibrate it by -c parameter. Is it correct?

WouterJD, 2024-03-11:
> workout breaks the trainer too much

Q1: please explain how you know that it's toomuch🤔

---

Rouvy transmits the real value for slope. 10% is received as 10%, whereas Zwift Shows 10% in it's own windows but transmits a lower value.

Real slope percentages cause noticeable resistances. 

You can adjust this in FortiusAnt using the - G parameter, see manual

TomX6601, 2024-03-14:
Independently I measure power through Favero pedals. Command -c does not work [image]

WouterJD, 2024-03-14:
I would say -c works, but you have to provide the correct value

TomX6601, 2024-03-14:
What am I doing wrong? I put there -c 14 or -c14 as decribed on page 42 of Manual and I receive error message "command line error"

WouterJD, 2024-03-14:
I modified https://github.com/WouterJD/FortiusANT/blob/master/pythoncode/FortiusAntCommand.py#L368
Please retry

Sorry for the inconvenience

decodeais, 2024-03-16:
Why did you  change FortiusAnt.sh?

WouterJD, 2024-03-16:
See comment

decodeais, 2024-03-16:
it looked for me you changed the FortiusAnt.sh script for debugging. it was only a hint. The changes for the -c parameter are ok.

TomX6601, 2024-03-19:
> I modified https://github.com/WouterJD/FortiusANT/blob/master/pythoncode/FortiusAntCommand.py#L368
> 
> Please retry
> 
> 
> 
> Sorry for the inconvenience
> 
> 
> 
> 

Thnx. It works somehow. It seems, that powercurve is slightly different in low power, like 110W, but still in tollerance I use

WouterJD, 2024-11-29:
Greate that the issue is sufficiently solved!

closed 2024-11-29 by WouterJD