WouterJD/FortiusANT #466: Can't make FortiusAnt to transmit info via BLE "Bluetooth interface n/a; Check exception."

DavidMedina092, 2024-09-07:
Hi there. I just recently heard about this amazing project. I have an old (but still working) i-flow magnetic tacx trainer. I'm running FortiusAnt on a MacBook air. I have already installed python, brew, and the libusb libraries. Also i ran the requirements.txt thing. 

My issue: when i run FortiusAnt i'm able to pair with my usb head unit (tacx t1932 fortius) and the ant message below says "Bluetooth interface aviable (bless)". But the moment i hit the start button, the message changes to "Bluetooth interface n/a; Check exception." and i cannot find my trainer on any training app. I believe that there is nothing being transmitted via bluetooth in the first place. 
I have basic knowledge of python and don't have hardcore programming abilities like you guys. But any help would be greatly appreciated. 
By the way. Add me to the map. I'm from Mexico City. Thanks!

WouterJD, 2024-09-09:
Welcome to the FortiusANT community
Please consider to visit [the sponsor page](https://github.com/sponsors/WouterJD)
- - -

Hi David; I'm happy to hear that you found FortiusAnt and that it's almost working.
I understand you have a USB-connected headunit, being the T1932 one.

Now BLE does not work; I would appreciate some more info; at lease the screen print of the console (black window) and the FortiusAnt main screen.
BLE is not the easiest interface, because drivers keep changing.

Let's see; perhaps I can help...

DavidMedina092, 2024-09-09:
Hello again. I don't know what actually happened. I tried to restart the graphic interface to copy the error message that i was getting and now it doesn't even pop. I'm just stuck with the following message: 

"FortiusANT is open source and can be used freely.
[.... .....]
or visit the sponsoring page https://github.com/sponsors/WouterJD
---------------------------------------------------------------
17:14:31,139: Dongle - No (free) ANT-dongle found
17:14:31,139: Find and initialise USB head unit
17:14:31,140: No Tacx trainer found
17:14:31,140: Tacx   - No Tacx trainer found
17:14:31,140: AntHRM - Heartrate expected from ANT+ HRM
Exit status: 0

Saving session...
...copying shared history...
...saving history...truncating history files...
...completed.

[Proceso completado]"

Maybe some more computer knowledge is needed to run FortiusAnt or maybe my computer is just old (as my trainer). Whatever it is i just don't seem to be able to run it properly. 
Yesterday y tried running it on an old windows10 laptop. I tried the windows executable and i get the same message as the beginning "Bluetooth interface n/a; Check exception". Thanks anyways for all the effort that you put into this project. I think i wont be able to run it.

WouterJD, 2024-09-10:
**USB-dongle:**
Do you have a USB-dongle in your system?
If yes, another application may have it in use. If there's no other application, unplug and insert to reset the interface.

**BLE interface**:
It seems you have no rights to write (El dispositivo no admite la característica de comando); rights must be granted in your operating system.

**General**:
Please summarize what configuration you want to use , refer to the [manual](https://github.com/WouterJD/FortiusANT/blob/master/doc/FortiusANTUserManual.pdf), please:

WouterJD, 2024-09-10:
Do not despair too soon; more than 1000 people world-wide use the application!
Since FortiusAnt runs on your Windows computer, it must be possible to get it running. I'm more familiar with Windows than Macintosh. Do not execute the executable by double-clicking but use the start-up script according to your parameterization!

Good luck

WouterJD, 2024-09-15:
Hi are you still trying to get things working?
If not, I close the issue until more info is provided.

DavidMedina092, 2024-09-16:
Hello! Yes. I'm still trying. I'm currently near a natural disaster zone and i've not been able to put much attention on the issue. There has been power outrages constantly which impeed to work on the problem. 

I ordered a new usb ble dongle to try and solve the issue (which has not been able to arrive to my destination due to roads blocked). Also, i quited on my attempt to run it on my macbook and i will only continue to work on my windows laptop.  
Hopefully i will update my progress by tuesday morning. 

Link to the news in spanish: 
https://www.eluniversal.com.mx/edomex/jilotzingo-sube-a-2-la-cifra-de-muertos-tras-deslave-rescatan-con-vida-a-otras-3-personas/

WouterJD, 2024-09-16:
Great! Lots of challenges, I see.
Curious on results of next test👍

DavidMedina092, 2024-09-18:
Ok. So u just bought a new usb ble dongle. Specifically the Asus BT400. I downloaded the drivers and installed them. Then, i opened the quickstart FortiusAnt app and i keep getting the same message: 

clsBleServer._Server(); start() exception [WinError -2147024580] "El dispositivo no admite la característica de comando" 
Bluetooth interface n/a; Check exception. 

I read about the problem on internet. I couldn't get specific information but i kinda understood that some permitions must be granted. I then followed the steps to add features to windows (no admite la característica de comando), but even after that i kept getting the very same error. I just don't know what else should i try. 

I tried to run it from the pythoncode or the StartUp files but i keep getting errors from faulty python installation apparently. For example i get "from bleak_winrt.windows.devices.bluetooth.genericattributeprofile import ( # type ignore # noqa: E501 ModuleNotFoundError: No module named 'Bleak_winrt'

WouterJD, 2024-09-20:
To start with the latter; try the command "pip install Bleak_winrt" and see what happens then.
It may be that dependencies are not installed.
I'm not a windows / python / BLE expert, so some trial-and-erro will be required.

WouterJD, 2024-09-20:
Of course, you have googled the error and found pages like https://support.microsoft.com/es-es/windows/solucionar-problemas-de-bluetooth-en-windows-723e092f-03fa-858b-5c80-131ec3fba75c I cannot tell from distance what the issue could be.

Did you read and follow manual 3.8?

WouterJD, 2024-09-20:
Please provide more info; not that there are many tacx types around.
Send some photo's.

As asked earlier; I understand you have a USB-connected trainer. Why not use ANT?

BLE/nodejs is a complex installation procedure
BLE/bless is a better setup, but the bless library tends to be less stable as wanted

If you have a USB-trainer; first get that side working and do some tests in test-mode with the -m flag.
That is a good step and provides some confidernce for you that it ever might work.

Then get along with the next step; connecting ZWIFT.

WouterJD, 2024-10-12:
Since there is no communication here, I assume can be closed.
If not happy to reopen

closed 2024-10-12 by WouterJD

firux88, 2024-12-04:
Hi Router,
I think I'm having the same issue. I was using -b (Bleno) last year, it was working fine with Rouvy but now it's not working well at all, so I've tried to upgrade to Bless, but no chance. All requirements is installed, I've also installed pip install Bleak_winrt as your propose.

12:59:09,365: Find and initialise USB head unit
12:59:09,382: Connected to Tacx Trainer T1932
12:59:09,685: Motor Brake Unit Firmware=0x1007 Serial= 4585 year=2010 type=T1949 Version2=1051 MotorBrake=True
12:59:09,786: FortiusAnt applies the MotorBrake power curve
Exception in thread Thread-2 (_OpenThread):
Traceback (most recent call last):
  File "C:\Python312\Lib\threading.py", line 1075, in _bootstrap_inner
    self.run()
  File "C:\Python312\Lib\threading.py", line 1012, in run
    self._target(*self._args, **self._kwargs)
  File "C:\github\FortiusANT-master\pythoncode\bleBlessClass.py", line 144, in _OpenThread
    self.loop.run_until_complete(self._Server())
  File "C:\Python312\Lib\asyncio\base_events.py", line 687, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "C:\github\FortiusANT-master\pythoncode\bleBlessClass.py", line 168, in _Server
    self.BlessServer = BlessServer(name=self.myServiceName, loop=self.loop)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: Can't instantiate abstract class BlessServerWinRT without an implementation for abstract method 'add_new_descriptor'
12:59:12,006: ---------- BLE-devices are activated ----------
Exception in thread Thread-1 (Tacx2DongleThread):
Traceback (most recent call last):
  File "C:\Python312\Lib\threading.py", line 1075, in _bootstrap_inner
    self.run()
  File "C:\Python312\Lib\threading.py", line 1012, in run
    self._target(*self._args, **self._kwargs)
  File "C:\github\FortiusANT-master\pythoncode\FortiusAnt.py", line 490, in Tacx2DongleThread
    rtn = Tacx2Dongle(self)
          ^^^^^^^^^^^^^^^^^
  File "C:\github\FortiusANT-master\pythoncode\FortiusAnt.py", line 171, in Tacx2Dongle
    rtn = FortiusAntBody.Tacx2Dongle(self)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\github\FortiusANT-master\pythoncode\FortiusAntBody.py", line 681, in Tacx2Dongle
    rtn = Tacx2DongleSub(FortiusAntGui, Restart)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\github\FortiusANT-master\pythoncode\FortiusAntBody.py", line 1407, in Tacx2DongleSub
    bleCTP.SetAthleteData(HeartRate)
  File "C:\github\FortiusANT-master\pythoncode\bleBless.py", line 358, in SetAthleteData
    self.BlessServer.get_characteristic(bc.cHeartRateMeasurementUUID).value = info
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'get_characteristic'

GPT told me to fix that line in bleBlessClass.py but no luck for now
class MyBlessServer(BlessServer):
    def __init__(self, name, loop):
        super().__init__(name=name, loop=loop)
        # Initialize additional properties if necessary

    def add_new_descriptor(self, descriptor):
        # Implement the method logic here
        print(f"Descriptor added: {descriptor}")
        # Add any logic needed to handle the descriptor