WouterJD/FortiusANT #468: USB ANT DEVICES NOT FOUND

EvoxRider365, 2024-09-26:
libusb-win32 driver for Tacx T1932 appears to be working.

FortiusANT unable to find Garmin and Tacx ANT Dongles

Under device manager there were exclamation marks on the two ANT Dongles and windows could not find drivers for them.

So out of desparation, I tried Zadig.  Now they are loading as USB Devices in device manager 
[image]
[image]
[image]
but FortiusANT is still not able to find them.

Any suggestions?

I followed the instructions in the manual.

Thanks

EvoxRider365, 2024-09-26:
I love it when I solve my own problems.  I uninstalled the Zadig drivers in the Device Manager for the two usb ANT sticks.

Then I downloaded drivers direct from Garmin/Tacx (likely an older version of libusb).

https://support.garmin.com/en-CA/?faq=lhcXYdqFRP1bbumefCOf37

and followed the Method Two (Manual) procedures.

Voila!  The correct libusb drivers for the Garmin/Tacx Ant sticks were installed!

FortiusANT is back and happy and I am happy and now I need to find an old ANT heart monitor...

Ride On!

WouterJD, 2024-09-26:
Happy to hear this.

Question "old ANT heart monitor"; any ANT-HRM will do.
What CTP do you use?

EvoxRider365, 2024-09-27:
Rouvy and Tacx and maybe bkool.  Also curious about Kinomap if I can get an ANT to Bluetooth converter to work (I have two somewhere) as Kinomap only works with Bluetooth and runs as an App on Apple TV hardware.I also was prompted to install a C++ package by Tacx in order to enable ANT and not just bluetooth.I had to edit the URL to what you see in the picture in order to get the download to work.Now i have some Fortius adjustments to make as the tire is slipping a bit.Sent from my iPhone

WouterJD, 2024-09-27:
FortiusAnt supports ble through -b or - bb, but some experimenting will be required.
-b using node.js is the complex path
-bb is less complex but the used libraries tend to change. Documentation is a starting point.

ANT HRM can be connected to CTP directly.

Tacx software does NOT need to be installed, and is usually more disturbing than helping

closed 2024-10-12 by WouterJD