WouterJD/FortiusANT #469: tacx 2021

albon4, 2024-10-12:
Hey I recently bought a genius tacx t2021 but ant+ but I don’t why it’s not wanting to connect I have updated drivers and everything pls help I’ve search every where nothing is working

albon4, 2024-10-12:
i dont understand how to setup the app

albon4, 2024-10-12:
i just need some guidance

WouterJD, 2024-10-12:
Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
You could also consider to visit [the sponsor page](https://github.com/sponsors/WouterJD)
- - -
I guess you have a genius.
FortiusAnt must be started with the -t genius flag as described in the [manual](https://github.com/WouterJD/FortiusANT/blob/master/doc/FortiusANTUserManual.pdf).

FortiusAnt uses one dongle; connecting to the genius and transmitting standard trainer protocol at the other side (converting the signals).
Zwift (using a second dongle / or on another system) receives the ANT signal and communicates through ForiusAnt with the genius.

Suces.

albon4, 2024-10-14:
> Welcome to the FortiusANT community
>
> I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used. Please tell me what bundle did you buy, and what [...]

albon4, 2024-10-14:
Hey mate what do I do if I don't have a second ant+ dongle

WouterJD, 2024-10-14:
Every application needs its own dongle.
If you do not have one, buy one.

Alternative: use another system with an ANT-interface, thus realizing the second dongle.

WouterJD, 2024-10-14:
You wanna have it closed?

albon4, 2024-10-14:
no

albon4, 2024-10-14:
Sorry i dont really understand github but i have a question do i need two dongles(ant+)

albon4, 2024-10-14:
[image]

albon4, 2024-10-14:
Does this work sir?

WouterJD, 2024-10-14:
Usually ant ANT dongle works, only the cycplus dongle are know for issues, as documented.

You can start with one dongle to validate the trainer and the interface, using the simulation and manual modes.

Note that I do not have a genius nor ever had, so some experimenting is to be done.

albon4, 2024-10-15:
[image]

WouterJD, 2024-10-15:
How did you start FortiusAnt?
Now use the settings wheel and sect genius as tacx type and retry.

FortiusAnt is not build to detect all possible tacx types, therefore you must tell what type to look for

albon4, 2024-10-15:
okay that worked but all i need now is a another ant+ right???

albon4, 2024-10-15:
or is there an alternative?

WouterJD, 2024-10-15:
> Every application needs its own dongle. If you do not have one, buy one.
> 
> Alternative: use another system with an ANT-interface, thus realizing the second dongle.

Read this

albon4, 2024-10-15:
Okay what do u want picture of, i'm guessing the ant+ i'm ordering?

WouterJD, 2024-10-15:
No, I believe that.

Your configuration + your setup, if you like.

albon4, 2024-10-15:
okay ill do that tmrw, great thanks for the help ur really saved me!

albon4, 2024-10-15:
i have another question u might of told me this but im not sure. But could i use my phone as a second ant+?

closed 2024-11-29 by WouterJD