WouterJD/FortiusANT #471: Install Instructions on Ubuntu

pitluko, 2024-11-04:
I can get Ubuntu up and Running on my Voyo V1 but im not able to get FortiusANT installed. The provided instructions in manual do not work. Can somebody do a noob friendly step by step guide to get it running?

decodeais, 2024-11-15:
May be this install skript [(https://github.com/decodeais/FortiusANT/blob/Wayland/raspberry/FortiusAntInstall.sh)](https://github.com/decodeais/FortiusANT/blob/Wayland/raspberry/FortiusAntInstall.sh) works for you. I used it with Linux Mint with success. ( issue #464).

decodeais, 2024-11-17:
Don't worry about the path and name of the script. It was supposed to be for the Raspberry, but since I kept it generic, I was able to quickly test it on the PC.