WouterJD/FortiusANT #472: MyWhoosh causes an error to be thrown

rcambrj, 2024-11-23:
I've been using FortiusANT with TrainerRoad and Zwift reliably, but when I connect MyWhoosh (a free alternative!), I get an error.

```
Nov 23 00:54:46 tacx python3[866]: 00:54:46,238: Target= 50W  0.0km/h Current=  0W Cad=  0 r=   0 100% ---b
Nov 23 00:54:47 tacx python3[866]: 00:54:47,239: Target= 50W  0.0km/h Current=  0W Cad=  0 r=   0 100% t---
Nov 23 00:54:48 tacx python3[866]: 00:54:48,241: Target= 50W  0.0km/h Current=  0W Cad=  0 r=   0 100% ---b
Nov 23 00:54:49 tacx python3[866]: 00:54:49,243: Target= 50W  0.0km/h Current=  0W Cad=  0 r=   0 100% t---
Nov 23 00:54:50 tacx python3[866]: ERROR:root:got unexpected error processing a message: read_request() missing 1 required positional argument: 'options'.
Nov 23 00:54:50 tacx python3[866]: Traceback (most recent call last):
Nov 23 00:54:50 tacx python3[866]:   File "/home/pi/.local/lib/python3.9/site-packages/dbus_next/message_bus.py", line 621, in _on_message
Nov 23 00:54:50 tacx python3[866]:     self._process_message(msg)
Nov 23 00:54:50 tacx python3[866]:   File "/home/pi/.local/lib/python3.9/site-packages/dbus_next/message_bus.py", line 712, in _process_message
Nov 23 00:54:50 tacx python3[866]:     handler(msg, send_reply)
Nov 23 00:54:50 tacx python3[866]:   File "/home/pi/.local/lib/python3.9/site-packages/dbus_next/message_bus.py", line 731, in handler
Nov 23 00:54:50 tacx python3[866]:     result = method.fn(interface, *args)
Nov 23 00:54:50 tacx python3[866]:   File "/home/pi/.local/lib/python3.9/site-packages/bless/backends/bluezdbus/dbus/characteristic.py", line 125, in ReadValue
Nov 23 00:54:50 tacx python3[866]:     return f(self)
Nov 23 00:54:50 tacx python3[866]:   File "/home/pi/.local/lib/python3.9/site-packages/bless/backends/bluezdbus/server.py", line 306, in read
Nov 23 00:54:50 tacx python3[866]:     return bytes(self.read_request(char.UUID))
Nov 23 00:54:50 tacx python3[866]: TypeError: read_request() missing 1 required positional argument: 'options'
Nov 23 00:54:50 tacx python3[866]: 00:54:50,244: Target= 50W  0.0km/h Current=  0W Cad=  0 r=   0 100% ---b
Nov 23 00:54:50 tacx python3[866]: ERROR:root:got unexpected error processing a message: read_request() missing 1 required positional argument: 'options'.
Nov 23 00:54:50 tacx python3[866]: Traceback (most recent call last):
Nov 23 00:54:50 tacx python3[866]:   File "/home/pi/.local/lib/python3.9/site-packages/dbus_next/message_bus.py", line 621, in _on_message
Nov 23 00:54:50 tacx python3[866]:     self._process_message(msg)
Nov 23 00:54:50 tacx python3[866]:   File "/home/pi/.local/lib/python3.9/site-packages/dbus_next/message_bus.py", line 712, in _process_message
Nov 23 00:54:50 tacx python3[866]:     handler(msg, send_reply)
Nov 23 00:54:50 tacx python3[866]:   File "/home/pi/.local/lib/python3.9/site-packages/dbus_next/message_bus.py", line 731, in handler
Nov 23 00:54:50 tacx python3[866]:     result = method.fn(interface, *args)
Nov 23 00:54:50 tacx python3[866]:   File "/home/pi/.local/lib/python3.9/site-packages/bless/backends/bluezdbus/dbus/characteristic.py", line 125, in ReadValue
Nov 23 00:54:50 tacx python3[866]:     return f(self)
Nov 23 00:54:50 tacx python3[866]:   File "/home/pi/.local/lib/python3.9/site-packages/bless/backends/bluezdbus/server.py", line 306, in read
Nov 23 00:54:50 tacx python3[866]:     return bytes(self.read_request(char.UUID))
Nov 23 00:54:50 tacx python3[866]: TypeError: read_request() missing 1 required positional argument: 'options'
Nov 23 00:54:51 tacx python3[866]: 00:54:51,246: Target= 50W  0.0km/h Current=  0W Cad=  0 r=   0 100% t---
Nov 23 00:54:52 tacx python3[866]: 00:54:52,247: Target= 50W  0.0km/h Current=  0W Cad=  0 r=   0 100% ---b
Nov 23 00:54:53 tacx python3[866]: 00:54:53,249: Target= 50W  0.0km/h Current=  0W Cad=  0 r=   0 100% t---
Nov 23 00:54:54 tacx python3[866]: 00:54:54,250: Target= 50W  0.0km/h Current=  0W Cad=  0 r=   0 100% ---b
```

then a little while later, this happens and FortiusANT doesn't recover.

```
Nov 23 00:57:56 tacx python3[866]: 00:57:56,538: Target= 50W  0.0km/h Current=  0W Cad=  0 r=   0 100% ---b
Nov 23 00:57:57 tacx python3[866]: 00:57:57,540: Target= 50W  0.0km/h Current=  0W Cad=  0 r=   0 100% t---
Nov 23 00:57:58 tacx python3[866]: 00:57:58,290: Read from USB trainer error: [Errno 19] No such device (it may have been disconnected)
Nov 23 00:57:58 tacx python3[866]: 00:57:58,391: Read from USB trainer error: [Errno 19] No such device (it may have been disconnected)
Nov 23 00:57:58 tacx python3[866]: 00:57:58,492: Read from USB trainer error: [Errno 19] No such device (it may have been disconnected)
Nov 23 00:57:58 tacx python3[866]: 00:57:58,593: Read from USB trainer error: [Errno 19] No such device (it may have been disconnected)
Nov 23 00:57:58 tacx python3[866]: 00:57:58,694: Read from USB trainer error: [Errno 19] No such device (it may have been disconnected)
Nov 23 00:57:58 tacx python3[866]: 00:57:58,695: Tacx head unit returns insufficient data, len=0
Nov 23 00:57:58 tacx python3[866]: 00:57:58,695: To resolve, check all (signal AND power) cabling for loose contacts.
Nov 23 00:57:58 tacx python3[866]: 00:57:58,696: Write to USB trainer error: [Errno 19] No such device (it may have been disconnected)
Nov 23 00:57:58 tacx python3[866]: 00:57:58,696: Target= 50W  0.0km/h Current=  0W Cad=  0 r=   0 100% ---b
Nov 23 00:57:58 tacx python3[866]: 00:57:58,706: Read from USB trainer error: [Errno 19] No such device (it may have been disconnected)
Nov 23 00:57:58 tacx python3[866]: 00:57:58,808: Read from USB trainer error: [Errno 19] No such device (it may have been disconnected)
Nov 23 00:57:58 tacx python3[866]: 00:57:58,909: Read from USB trainer error: [Errno 19] No such device (it may have been disconnected)
Nov 23 00:57:59 tacx python3[866]: 00:57:59,011: Read from USB trainer error: [Errno 19] No such device (it may have been disconnected)
Nov 23 00:57:59 tacx python3[866]: 00:57:59,112: Read from USB trainer error: [Errno 19] No such device (it may have been disconnected)
Nov 23 00:57:59 tacx python3[866]: 00:57:59,113: Tacx head unit returns insufficient data, len=0
Nov 23 00:57:59 tacx python3[866]: 00:57:59,114: To resolve, check all (signal AND power) cabling for loose contacts.
Nov 23 00:57:59 tacx python3[866]: 00:57:59,116: Write to USB trainer error: [Errno 19] No such device (it may have been disconnected)
Nov 23 00:57:59 tacx python3[866]: 00:57:59,125: Read from USB trainer error: [Errno 19] No such device (it may have been disconnected)
Nov 23 00:57:59 tacx python3[866]: 00:57:59,227: Read from USB trainer error: [Errno 19] No such device (it may have been disconnected)
Nov 23 00:57:59 tacx python3[866]: 00:57:59,328: Read from USB trainer error: [Errno 19] No such device (it may have been disconnected)
```

MyWhoosh reports a good connection, but never wattage or RPM

[image]

Here's the startup log, in case that helps:

```
Nov 23 00:52:48 tacx python3[866]: ---------------------------------------------------------------
Nov 23 00:52:48 tacx python3[866]: FortiusANT is open source and can be used freely.
Nov 23 00:52:48 tacx python3[866]: Just for the fun of knowing where you all are training,
Nov 23 00:52:48 tacx python3[866]: put yourself on the FortiusANT map by making yourself known
Nov 23 00:52:48 tacx python3[866]: by leaving a message with name/location/trainer on
Nov 23 00:52:48 tacx python3[866]: https://github.com/WouterJD/FortiusANT/issues/14
Nov 23 00:52:48 tacx python3[866]: or visit the sponsoring page https://github.com/sponsors/WouterJD
Nov 23 00:52:48 tacx python3[866]: ---------------------------------------------------------------
Nov 23 00:52:48 tacx python3[866]: 00:52:48,701: Dongle - No ANT, Bluetooth interface available (bless)
Nov 23 00:52:48 tacx python3[866]: 00:52:48,701: Find and initialise USB head unit
Nov 23 00:52:48 tacx python3[866]: 00:52:48,792: Connected to Tacx Trainer T1932
Nov 23 00:52:49 tacx python3[866]: 00:52:49,299: Tacx head unit returns incorrect header 0x21303 (expected: 0xc03)
Nov 23 00:52:49 tacx python3[866]: 00:52:49,502: Motor Brake Unit Firmware=0x1007 Serial=20366 year=2009 type=T1941 Version2=2069 MotorBrake=True
Nov 23 00:52:49 tacx python3[866]: 00:52:49,602: FortiusAnt applies the MotorBrake power curve
Nov 23 00:52:49 tacx python3[866]: 00:52:49,806: Tacx   - Connected to Tacx Trainer T1932
Nov 23 00:52:49 tacx python3[866]: 00:52:49,806: AntHRM - No heartrate monitor connected
```

Any idea what this might be?

rcambrj, 2024-11-24:
I thought I would add some further information:

* when using `FortiusAnt.py` with `-s`, MyWhoosh successfully pairs and retrieves the simulated wattage and cadence.

WouterJD, 2024-11-24:
OK, as you may understand, FortiusAnt has two "sides":
1. The tacx trainer
2. The CTP, like MyWhoosh

The -s flag allows to test side2; the interface to CTP which is apparently working.
This is consistent with teh error, which are reported on "Side 1", the tacx trainer.

`I've been using FortiusANT with TrainerRoad and Zwift reliably`
Since you have been working with Zwift, the chain (sid=trainer, FortiusANt, side2=CTP) **CAN** work.
It is not whoosh related.

The job now is to find out why FortiusANT cannot connect to the trainer through the USB cable.
1. Reboot system
2. Check all connections, not only the USB cable but all. Especially the mains connection can cause interference.
3. Retry

` TypeError: read_request() missing 1 required positional argument: 'options'`
I must say that this is an unexpected error (I will check later)

But after a restart, it should not occur again.

WouterJD, 2024-11-24:
Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
You could also consider to visit [the sponsor page](https://github.com/sponsors/WouterJD)
- - -

rcambrj, 2024-11-24:
Hey Wouter, thanks for your response. I have been reliably using Trainer Road. It works perfectly. Whenever I try to use MyWhoosh, FortiusANT dies. A reboot does not fix this. The USB cable is known to be working as it works great with Trainer Road, and I'm quite certain that interference is also not the problem on account of it working, still, with Trainer Road.

I have been using the 32bit version of Raspberry Pi OS and I've since switched to the 64bit version to see whether that would fix this problem. In doing so, the error you mention appears to have gone away, leaving only the read/write USB errors. This may be something to do with upgrading from python 3.9 to 3.10.

I suspect that MyWhoosh is doing something which doesn't get correctly translated to the trainer and some error pop up. I'd happily debug this further, but I don't get much more information by using a higher debug level.

I'm using a Fortius motor brake, and although my location is public, I would like to remain away from your data collection, if that's alright. I would, however, happily sponsor this investigation of this issue.

WouterJD, 2024-11-29:
Hey @rcambrj sorry for giving the impression I'm a facebook-like data gatherer, all I like is to put a star on the map - as you may have seen elsewhere in the git.

If MyWhoosh communicates correctly through the ANT protocol, you can investigate this using the tracing options and checking on the ANT-interface what differences occur between Trainer Road and MyWhoosh. 

I hope you manager to debug the stuff. Succes!

WouterJD, 2024-11-30:
No, never heard of MyWhoosh

WouterJD, 2024-12-09:
> Nov 23 00:54:50 tacx python3[866]:   File "/home/pi/.local/lib/python3.9/site-packages/bless/backends/bluezdbus/server.py", line 306, in read
> Nov 23 00:54:50 tacx python3[866]:     return bytes(self.read_request(char.UUID))

Any progress here?
I think it's not in FortiusAnt but in libraries being used.

rcambrj, 2024-12-09:
Hey @WouterJD!

As an exercise in dusting off my Go, I've been rewriting a portion of FortiusAnt https://github.com/rcambrj/bicipi

Although I'm doing this with the serial port for now (directly connected to the motorbrake), in doing this, I've been learning a lot about the protocol that the T1941 uses (which seems to be very similar to what's then forwarded over USB). So I hope to be in a good position to eventually a) get `bicipi` to work with MyWhoosh and b) to contribute back to FortiusAnt with the solution.

WouterJD, 2024-12-09:
Great stuff; I have relabelled the issue since I believe it's not a FortiusAnt issue. Please refer to #473

rcambrj, 2024-12-09:
I don't mind how you label the issue, but given that MyWhoosh works with FortiusAnt's simulator (using bless), I don't think that the problem is related to that bless issue or bluetooth at all.

WouterJD, 2025-01-11:
Since you rsimulator works, I assume can be closed.
If not happy to reopen

rcambrj, 2025-01-11:
@WouterJD there is currently a bug in FortiusAnt which prevents it from working with MyWhoosh. If you don't want to support MyWhoosh that's fine, but at least call it what it is.

My [bicipi](https://github.com/rcambrj/bicipi) project is coming along nicely, and it's working with MyWhoosh and TrainerRoad (Zwift untested).

WouterJD, 2025-01-11:
> If you don't want to support MyWhoosh that's fine

🤔 Who said that? 

Please provide more info when the issue occurs and I'm glad to assist in debugging/solving the issue.

Please note that FortiusAnt is provided for free😉

rcambrj, 2025-01-11:
Thanks for reopening the issue.

> Please note that FortiusAnt is provided for free😉

Allow me to remind you of a portion of an earlier message of mine:

> I would, however, happily sponsor this investigation of this issue.

Moving on...

> Please provide more info when the issue occurs and I'm glad to assist in debugging/solving the issue.

I will find the time to do this. Would you be so kind as to indicate the best CLI parameter(s) to use for logging this problem? In the meantime, reproduction steps are as follows:

1. Install and run FortiusAnt with BLE
2. Install and run MyWhoosh on a second machine with BLE
3. Connect MyWhoosh to FortiusAnt

Expected: wattage and cadence are reported to MyWhoosh
Observe: wattage and cadence are not reported to MyWhoosh. Errors in FortiusAnt log (see initial post)

WouterJD, 2025-01-12:
> happily sponsor this investigation of this issue

That's OK, my fee is € 150,= fixed price to get MyWhoosh working, sponsoring details on https://github.com/sponsors/WouterJD.

WouterJD, 2025-01-31:
Not reproducible. MyWhoosh is being used by FortiusAnt users on strava

closed 2025-01-31 by WouterJD