WouterJD/FortiusANT #474: Request: PWM output on Raspi

AfromD, 2025-01-06:
I would like to request a new feature for Raspberry Pi boards.

The objective is to automate the operation of a ventilation fan to cool the rider. I propose a command line option like -fp 200 or alternatively -fs 40. -fp 200 would cause the raspberry pi to output a PWM signal on GPIO 19 (lower pins are blocked by my screen), where 0 fortius power would mean no duty cycle, 200 would mean full duty cycle, and any value between that would linearly scale the duty cycle between 25 and 100%. -fs 40 would mean scale the duty cycle according to speed to reach 100% by 40 km/h.

On the hardware side, I would use this https://www.amazon.de/dp/B0C3QTKV5V/ which I think is a phase controlled triac with pwm input. This would control the 220v power to a regular fan.

AfromD, 2025-01-07:
I had a nights sleep over it. The first piece of hardware would be simpler to program, ask python to set PWM on GPIO 19 and the hardware does the rest. The second piece of hardware I found needs explicit and timing ciritical commands to chop up the AC power smoothly. 

Now the problem with the first hardware is that I found multiple reviews mentioning it going up in smoke. So that is not really something I want to use. It would be prudent to use the second hardware.

The example program I found is not that great as it puts the processor into wait loops, effectively monopolizing a processor core. Sufficient for a single purpose arduino, not so great on a proper computer like a pi. So what is needed is an efficient piece of interrupt coding that doesn't pause processing.

Given that I am a total noob at Python and not an experienced programmer in general (ChatGPT has advanced to my most productive IDE), I will need some studying up to do  that kind of coding (should I avoid floating point math, how to properly chop up the AC, getting the timing spot on). I have no experience with git. Also the code will need to take care of evaluating the AC frequency. And I will have to familiarize myself with the project and git. 

I have Ideas on a high level on how to do this, it will take some time though. Might not get done this winter.

AfromD, 2025-01-07:
Ok, I've been looking at this for some time now, but I think I do not need to reinvent the wheel. 

Integrating this real time stuff in FortiusANT doesn't seem ideal, and with my skills I would rather do it on a microcontroller, which would then interface with FortiusANT. That opens up the question about how to interface, and then there is a bit more effort in supplying the microcontroller with a power supply. I was thinking about making a standalone networked fan with an ESP8266. Networked would remove the need for pi hardware.

Researching the hardware side of this, I found there are smart home wifi enabled phase cutting dimmers available as a whole package for not a lot of money. A good example would be the sonoff ifan04. It allows control via http, and also supports Tasmota firmware. It simply connects inline in the AC supply, so no diy skills required. I found a german eBay supplier that [sells it pre-flashed with tasmota](https://www.ebay.de/itm/115725728957).

So all that would be needed for FortiusANT in this case is to fire off http requests to control the fan. 

So I will simply get a sonoff ifan module, and limit my FortiusANT efforts to getting it to fire off appropriate http requests. The big advantage of this is that it might even be platform-independent (like I said, Python noob), so all FortiusANT users can use it with a fan, not just Pi users.

WouterJD, 2025-01-07:
> I am a total noob at Python

The FortiusANT story started in finding antifier on github, concluding the idea was good but the implementation needed improvement.

So I learned python, git, vsc, oo-programming, tacx-, ant-and later also ble-bluetooth - as documented in the manual.

So let's the fear not overwhelm you and grab the challenge!

WouterJD, 2025-01-07:
By the way, I use a normal fan with a remote control that I have attached to the steering bar.
Perhaps not as advanced as a **Wahoo KICKR Headwind Ventilator** but for a fraction of the cost.

Drawback: no python, git, vsc, oo-programming, tacx-, ant- programming required😉

WouterJD, 2025-01-08:
The request library is way to do it, enjoy the programming.

I'm curious how you will decide the fanspeed; wheel-speed, power, heart rate, sweat rate (in ascending order of preferability) 😉

AfromD, 2025-01-08:
I plan to make it configurable. There will be command line options to specify speed or power control, with an optional parameter to determine the thresholds. If both options are specified, it will default to power control. Also an IP address has to be specified.
-fp defaults to 200, this will cause the fan to start at 1/3rd of 200W, stage 2 at 2/3rds of 200W, and stage 3 at 200W or more.
-fs defaults to 40, this will cause the fan to start at 3 km/h, stage 2 at 1/2 of 40 km/h, and stage 3 at 40 km/h or more.
The fan will stop automatically when falling below the thresholds again.
I might add some timer to not make it switch more often than every 5 seconds, and/or some individual hysteresis/moving average for speed and power so that it will need a significant change to switch fan speed.

WouterJD, 2025-01-08:
-fh for heart rate...

Please do not forget to implement the JSON an GUI parameterization.

AfromD, 2025-01-08:
I have already looked at those pieces of code actually. I am not so sure about heart rate, because we are not guaranteed to have it.

WouterJD, 2025-01-08:
If you have a ANT HR, it might be an option 😉

AfromD, 2025-01-08:
BTW the Sonoff iFan also has a light switch option. I might add that as well so that I can switch the light off whilst I am training since I use a beamer. Another option would be to administer an electric shock when the heart rate goes too low.

AfromD, 2025-01-10:
I have another idea for the light relay: Connect it via another capacitor (and a discharge resistor) to the fan, and change the other two capacitors. Then with a change in tasmota, capacitances can be switched additively to create extra speeds. Replace the 2.5uF and 3uF caps by 2.1 uF, 0.8uF and add 0.4uF on the light relay. In stead of the three settings 0uF, 3uF and 2.5 uF, one could then create 0uF, 3.3uF, 2.9uF, 2.5 uF and 2.1 uF. So five speed settings in stead of three.

But that is something for advanced users then. First step is to get this working as above.

closed 2025-01-31 by WouterJD